bdring/FluidNC #788: Problem: Pen plotter : uln2003 - unipolar 28byj-48 stepper motor config

Ammar56788, 2023-01-26:
### Controller Board

Pen plotter run with simple diy circuit such as Breadboard.!

### Machine Description

Pen plotter

28byj-48 as x axis motor.
28byj-48 as y axis motor.
sg90 as z axis motor (pen up and down)
Two simple limit switch circuit.
Esp32 devkit as main processor and wifi etc.

### Configuration file

```yaml
There is noting
```

### Startup Messages

```text
No
```

### User Interface Software

WebUI

### What happened?

I cant write new or edit other exist example config .yaml for my pen plotter
Because i dont know yaml language...

### Other Information

Excuse me for bad English

Ammar56788, 2023-01-26:
And wiki.fluidnc.com not responding

mani-fa, 2023-01-29:
> And wiki.fluidnc.com not responding
It works try VPN to access.

Ammar56788, 2023-01-30:
Thnaks and Damn to who boycott a country!
And is there any code for plotter with both uln2003 and esp32?

evgeny-smirnov, 2023-12-02:
To configure 28byj needs to use unipolar drive

here is example
```
  x:
    steps_per_mm: 80
    max_rate_mm_per_min: 500
    motor0:
      hard_limits: false
      unipolar:
        phase0_pin: gpio.13
        phase1_pin: gpio.12
        phase2_pin: gpio.14
        phase3_pin: gpio.27
        half_step: false
```

and try to use servo settings like in WallPlotter.yaml config example

raspiduino, 2024-11-11:
In case anyone wonder why there's no longer unipolar support in the latest version, then it's removed in commit https://github.com/bdring/FluidNC/commit/921f32bed06ad6bce24d5a49f3040b628a963aba.

You can still get it back in commit https://github.com/bdring/FluidNC/commit/c65c1d90d353825c24d64e11a85b16d0cd55cdf7.

TL;DR: If you want unipolar support, use [v3.8.3](https://github.com/bdring/FluidNC/releases/tag/v3.8.3)

oliof, 2024-11-11:
Alternately, convert the 28byj-48 into bipolar steppers by breaking one trace https://ardufocus.com/howto/28byj-48-bipolar-hw-mod/ and change to bipolar stepper config.

MitchBradley, 2024-11-11:
A4988 drivers cost less than a dollar.

raspiduino, 2024-11-12:
> If you use 3.8.3 you will have to accept the fact that it sometimes crashes in the middle of a job.

Can you link to the commit(s) that fix this bug?

MitchBradley, 2024-11-12:
Basically I decided not to support unipolar motors anymore because few people use them,  bipolar drivers have many advantages at very low cost, and supporting unipolar made it difficult to make some valuable improvements in the code.  Those improvements double the max stepping rate, fix the crash problem while saving precious RAM, eliminate step jitter, and  make it unnecessary to have separate I2S_STATIC and I2S_STREAM mode.

Ammar56788, 2024-11-12:
I am no longer using fluid NC and grbl esp32. The previous project was terminated. There is no need for a unipolar driver.

closed 2024-11-12 by MitchBradley

g1smo, 2025-04-15:
Maybe not for you. But there are others using unipolar motors for various reasons. They are cheap, quiet, stay cold and consume little power.

I have forked FluidNC 3.9.4 with unipolar motor support, you can find the repostitory (and a sample ULN2003 config) here:
https://git.kompot.si/g1smo/FluidNC/src/branch/uln2003

You can see it in action: https://www.youtube.com/watch?v=Sgox7lTbIbQ

MitchBradley, 2025-04-15:
> [Unipolar motors] are cheap, quiet, stay cold and consume little power.

For stepper motors of a given holding torque:

A bipolar motor needs less material since it does not need the center tap wires and connections.  Hence it is inherently cheaper.

Bipolar motors are more efficient, generating 1.4 times as much torque for the same amount of power dissipation.  Hence they are cooler and consume less power than unipolar motors at a given performance level.

Bipolar motors are quieter because the technology of bipolar drivers has advanced to provide better control of motor dynamics, while unipolar drivers are pretty much stuck in the "pair of transistors, on or off" era.

It could be that you have some cheap, quiet, cold, low power unipolar motors - but if so it is not because they are unipolar.  Cheap bipolar motors are also available, and will be quieter, cooler, and lower power for the same torque.

Unipolar drivers, though, are a bit cheaper.  A ULN2003A goes for about 10 cents, while A4988 costs 20 cents.

g1smo, 2025-04-15:
Not everything is a competition. Thank you for your work on FluidNC.
I just dropped the link for anyone who might have unipolar motors and driver and hand, wanting to drive them with an android phone.
I plan to update the fork every once in a while.

SRM-AI-Lab123, 2025-12-19:
Links:
Source code: [https://github.com/TGit-Tech/GRBL-28b...](https://www.youtube.com/redirect?event=video_description&redir_token=QUFFLUhqbmZnQUdIY1dKb3A4VTlLczRlbnV6aHdPOFZwZ3xBQ3Jtc0ttVDMwREtFR25XZlB3SWVRNTd1eHNaQndjZTA2bGJwanowdHR0WlVRWVZHNlhtcXBEaDlnY2EzNm55aWpWbjBlYkJSUElEWTFUSDBXcW51SW9zNElIYU42QkFRV2RONFlpN2VYdXg2RThYZnZkWlk3Zw&q=https%3A%2F%2Fgithub.com%2FTGit-Tech%2FGRBL-28byj-48&v=yjwiEk_ZV5E)
3d printed parts: [https://www.thingiverse.com/thing:460...](https://www.youtube.com/redirect?event=video_description&redir_token=QUFFLUhqbEo0dlN4Zk1oaW5QTTByUnlTN0NvOVdhelFLd3xBQ3Jtc0tsN1JBS2twYnBkUlhNWVo0aGZZcFJRUHJtZm10ZGd3MGZoUHlyaWU1U2dpcFFMM0ZYYTNtWENqLVJMVkZfVVVzcno5bUpackh0UWhXMDdjOGg0T2swT291UGVwb0c0MzV5NFc4QWFPSGJlYjV6b1dmNA&q=https%3A%2F%2Fwww.thingiverse.com%2Fthing%3A4607077&v=yjwiEk_ZV5E)
Inkscape: [https://inkscape.org/release/inkscape...](https://www.youtube.com/redirect?event=video_description&redir_token=QUFFLUhqa2ZjSmJQTUlINHVBTEMtWTRKa3p4MFRGNndWZ3xBQ3Jtc0trZXFKTTZDa3c2UFprQTlPTkQwQnNKQXZQOEpPVTF2VTlqRzF1M0FVSGdGOTBhYVViLS1FTVlCTVUyS2JxU05WS0RFQ21nYkppUmI1UWRrTlBjaFNQYjZkWVNjbVhqR2hyOHJGV19wMmhseWdhNjBzbw&q=https%3A%2F%2Finkscape.org%2Frelease%2Finkscape-1.0.1%2F&v=yjwiEk_ZV5E)
JSCUT tool: [https://jscut.org/](https://www.youtube.com/redirect?event=video_description&redir_token=QUFFLUhqbTNaY0NXd1E5bjJ2ZHVrQ0dLVlBrc21JY3NFUXxBQ3Jtc0ttalhCdDgtbWlrZUFpTWppYUpFVzN0MklxQ0lKdl9PR3c2T0U2UjJ5dHh1ZXV4Wkk0ZFpmZFZnUW5GbHA4WF8zbHNqY2g5RFFVV0doaExLTVI1Tl94SVNoTlp2Um1FTm52ZGpmOUtYQWZvY2NRellEOA&q=https%3A%2F%2Fjscut.org%2F&v=yjwiEk_ZV5E)
UGS Download link: [https://winder.github.io/ugs_website/...](https://www.youtube.com/redirect?event=video_description&redir_token=QUFFLUhqbWVYS2dic0xmZFNoelVxNndrVjQ5ZkpiMTlZd3xBQ3Jtc0tuWjRoazlMMEtPV2NWeUVOd09hWHZGMVVyOTVWWlVfeFh1R0pRTkllNzRlSjhSTFdCUWNIbkJmaWliUk1rekozcVFOdE9CVFdTNkhCSjIycXNqVGUwNk04NnhUZkxUMEZxdnppQ1BUVWZ6N3ItTGk3OA&q=https%3A%2F%2Fwinder.github.io%2Fugs_website%2Fdownload%2F&v=yjwiEk_ZV5E)