bdring/FluidNC #790: Comment activer air assist avec MKS DLC32

FMD14, 2023-01-27:
### Controller Board

MKS DLC32

### Machine Description

Decoupe Laser Core XY

Voici le montage
[image]

### Configuration file

```yaml
board: MKS-DLC32 V2.1
name: CoreXY Laser NJE 30W
meta: 02/11/2022 FMD14

kinematics:
  corexy:

stepping:
  engine: I2S_STATIC
  idle_ms: 0
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: I2SO.0
  x:
    steps_per_mm: 80
    max_rate_mm_per_min: 25000.000
    acceleration_mm_per_sec2: 3200.000
    max_travel_mm: 515.00
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 7000.000
      settle_ms: 525.00
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.35:low
      hard_limits: true
      pulloff_mm: 2.000
      stepstick:
        step_pin: I2SO.1
        direction_pin: I2SO.2:low

  y:
    steps_per_mm: 80
    max_rate_mm_per_min: 25000.000
    acceleration_mm_per_sec2: 3200.000
    max_travel_mm: 555.00
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 7000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.36:low
      hard_limits: true
      pulloff_mm: 2.000
      stepstick:
        step_pin: I2SO.5
        direction_pin: I2SO.6:low

  z:
    steps_per_mm: 157.750
    max_rate_mm_per_min: 1200.000
    acceleration_mm_per_sec2: 500.000
    max_travel_mm: 80.000
    soft_limits: true
    homing:
      cycle: 0
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 1000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.34:low
      hard_limits: false
      pulloff_mm: 1.000
      stepstick:
        step_pin: I2SO.3
        direction_pin: I2SO.4:low

i2so:
  bck_pin: gpio.16
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.12
  mosi_pin: gpio.13
  sck_pin: gpio.14

sdcard:
  cs_pin: gpio.15
  card_detect_pin: NO_PIN

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: gpio.33:low:pu
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

macros:
  startup_line0: 
  startup_line1: 
  macro0: $SD/Run=lasertest.gcode
  macro1: $SD/Run=home.gcode
  macro2:
  macro3:

coolant:
  flood_pin: NO_PIN
  mist_pin: NO_PIN
  delay_ms: 0

probe:
  pin: gpio.22
  check_mode_start: true

Laser:
  pwm_hz: 5000
  #L on Beeper / IN on TTL
  output_pin: gpio.32
  enable_pin: I2SO.7
  disable_with_s0: false
  s0_with_disable: false
  tool_num: 0
  speed_map: 0=0.000%  1000=100.000%
# 135=0mA 270=5mA 400=10mA 700=16mA

user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN

start:
  must_home: false

# 5,18,19,22,23,25,26,27,32,33,39,I2SO.7
# SDA 0 / SCL 4
```

### Startup Messages

```text
FluidNC v1.2.0 using COM3
Exit: Ctrl-C, Ctrl-Q or Ctrl-], Clear screen: CTRL-W
Upload: Ctrl-U, Reset ESP32: Ctrl-R, Send Override: Ctrl-O
FluidNC filename [fmd14_corexy.yaml]:
XModem Upload D:\FluidNC\configuration\fmd14_corexy.yaml fmd14_corexy.yaml
$Xmodem/Receive=fmd14_corexy.yaml
[MSG:INFO: Received 3048 bytes to file /spiffs/fmd14_corexy.yaml]
ok
$bye
[MSG:INFO: Restarting]
ok
[MSG:INFO: WiFi Disconnected]
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:12812
load:0x40080400,len:3032
entry 0x400805e4

[MSG:INFO: FluidNC v3.6.4-pre3]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:fmd14_corexy.yaml]
[MSG:INFO: Machine CoreXY Laser NJE 30W]
[MSG:INFO: Board MKS-DLC32 V2.1]
[MSG:INFO: I2SO BCK:gpio.16 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.14 MOSI:gpio.13 MISO:gpio.12]
[MSG:INFO: SD Card cs_pin:gpio.15 detect:NO_PIN]
[MSG:INFO: Stepping:I2S_static Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:0ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable I2SO.0]
[MSG:INFO: Axis X (0.000,515.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.1 Dir:I2SO.2:low Disable:NO_PIN]
[MSG:INFO:  X Neg Limit gpio.35:low]
[MSG:INFO: Axis Y (0.000,555.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.5 Dir:I2SO.6:low Disable:NO_PIN]
[MSG:INFO:  Y Neg Limit gpio.36:low]
[MSG:INFO: Axis Z (0.000,80.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.3 Dir:I2SO.4:low Disable:NO_PIN]
[MSG:INFO:  Z Neg Limit gpio.34:low]
[MSG:INFO: macro0_pin gpio.33:low:pu]
[MSG:INFO: Kinematic system: CoreXY]
[MSG:INFO: Laser Ena:I2SO.7 Out:gpio.32 Freq:5000Hz Period:8191]
[MSG:INFO: Using spindle Laser]
[MSG:INFO: Probe Pin: gpio.22]
[MSG:INFO: Connecting to STA SSID:Livebox-19B4]
[MSG:INFO: Connecting.]
[MSG:INFO: Connected - IP is 192.168.1.16]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.6 [FluidNC v3.6.4-pre3 (wifi) '$' for help]
```

### User Interface Software

LaseGRBL ,Lightburn

### What happened?

Le relais n'est pas exciter

Avataar120, 2023-01-27:
Il faut renseigner les sorties ici : 
[image]

Je ne connais pas cette carte et donc pas les valeurs à mettre, mais tu les trouveras sans doute facilement sur le net

FMD14, 2023-01-27:
Je ne trouve pas les valeurs pour la carte

FMD14, 2023-01-27:
Je viens de trouvé 
la valeur est 04 
les adresses sont là
https://github.com/makerbase-mks/MKS-DLC32/blob/main/hardware/MKS%20DLC32%20V2.1_003/MKS%20DLC32%20V2.1_003%20PIN.pdf

closed 2023-01-27 by FMD14