bdring/FluidNC #794: Panic: Cache disabled but cached memory region accessed in v3.6.6

kdomanski, 2023-01-29:
### Controller Board

[FYSETC E4](https://wiki.fysetc.com/E4/)

### Machine Description

Modded 3018

### Configuration file

```yaml
name: "3018 upgraded"
board: "FYSETC E4"
verbose_errors: true
use_line_numbers: true

start:
  must_home: false
  deactivate_parking: true
  check_limits: false

stepping:
  engine: RMT
  idle_ms: 255
  pulse_us: 2
  dir_delay_us: 1
  disable_delay_us: 1

axes:
  shared_stepper_disable_pin: gpio.25

  x:
    steps_per_mm: 800
    max_rate_mm_per_min: 5000
    acceleration_mm_per_sec2: 150
    max_travel_mm: 289
    soft_limits: false
    homing:
      cycle: 2
      allow_single_axis: true
      positive_direction: false
      mpos_mm: 1
      seek_mm_per_min: 1200

    motor0:
      hard_limits: false
      pulloff_mm: 4.000
      limit_all_pin: gpio.34
      tmc_2209:
        uart:
          txd_pin: gpio.22
          rxd_pin: gpio.21
          baud: 115200
          mode: 8N1
        addr: 1
        r_sense_ohms: 0.110
        run_amps: 1.0
        hold_amps: 0.250
        microsteps: 16
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: CoolStep
        homing_mode: CoolStep
        use_enable: false
        step_pin: gpio.27:high
        direction_pin: gpio.26:low

  y:
    steps_per_mm: 800
[... 159 lines cut ...]

Core  1 register dump:
PC      : 0x40112248  PS      : 0x00060035  A0      : 0x8008a724  A1      : 0x3ffc030c
A2      : 0x3ffb4ad4  A3      : 0x00000000  A4      : 0x3ffbed08  A5      : 0x3ffcd4ac
A6      : 0x3ffca344  A7      : 0x3ffb26fc  A8      : 0x80081e21  A9      : 0x00000002
A10     : 0x0000001a  A11     : 0x00000001  A12     : 0x3ffba630  A13     : 0x3ffba610
A14     : 0x00000004  A15     : 0x3ffcd0d4  SAR     : 0x00000000  EXCCAUSE: 0x00000007
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000

Backtrace:0x40112245:0x3ffc030c |<-CORRUPTED

ELF file SHA256: 0000000000000000

Rebooting...
ets Jul 29 2019 12:21:46

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:12812
load:0x40080400,len:3032
entry 0x400805e4
```

kdomanski, 2023-01-29:
The `<-CORRUPTED` not at the backtrace is surprising. Any advice how I might try to get a proper backtrace instead?

I looked up the PC address `0x40112248` in the v3.6.6 ELF, but that just points to `__digitalWrite`.

bdring, 2023-01-29:
We have some info on using the backtrace here.

http://wiki.fluidnc.com/en/installation#backtrace-decoding-windows

kdomanski, 2023-01-29:
Thanks, that helped a bit, but I still got nothing that makes sense.

`addr2line` points me to the framework code itself:
```
❯ addr2line -a 0x40112245:0x3ffc030c -e /Users/kdomanski/Downloads/wifi-firmware-fluidnc-v3.6.6.elf
0x40112245
/home/runner/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-gpio.c:108
```
---

My findings so far:
- [FluidNC v3.6.6 is built with espressif32 platform v4.3.0](https://github.com/bdring/FluidNC/blob/v3.6.6/platformio.ini#L46)
- [that in turn depends on arduinoespressif32 framework ~3.20003.0 ](https://github.com/platformio/platform-espressif32/blob/v4.3.0/platform.json#L71..L76)
- [which seems to translate to version 2.0.3 of the actual framework code](https://registry.platformio.org/tools/platformio/framework-arduinoespressif32?version=3.20003.220626)
- [in that version of the framework the suspect line simply sets a binary flag](https://github.com/espressif/arduino-esp32/blob/2.0.3/cores/esp32/esp32-hal-gpio.c#L108)

MitchBradley, 2023-01-29:
Please supply a GCode file that panics.  Also tell us what sender you are using - WebUI, UGS, or whatever.  If you are using a serial sender, is WebUI also connected?

How long does it take before the panic happens?  I know that it is unlikely to be deterministic, but is it a few minutes, or hours, or what?

kdomanski, 2023-01-29:
[1001.nc.zip](https://github.com/bdring/FluidNC/files/10530475/1001.nc.zip)

I've been streaming with UGS over serial, with the WebUI opened in background. The panic happens within 5 to 40 minutes.

I was unable to replicate the problem while streaming from SD (command issued in Fluidterm) without the browser UI opened.

MitchBradley, 2023-01-29:
Is WebUI using auto-reporting or polling?
[image]

MitchBradley, 2023-01-29:
Ouch.  __digitalWrite is in FLASH.  That is a huge problem because it is used by the stepper interrupt, and it could be displaced from icache and then refilled during an ISR.

MitchBradley, 2023-01-29:
Foiled again by the Arduino Framework:

framework-arduinoespressif32/cores/esp32/esp-hal-gpio.c:
 ```c
extern void ARDUINO_ISR_ATTR __digitalWrite(uint8_t pin, uint8_t val)
{
	gpio_set_level((gpio_num_t)pin, val);
}
```
framework-arduinoespressif32/cores/esp32/esp-hal.h:
```c
#if CONFIG_ARDUINO_ISR_IRAM
#define ARDUINO_ISR_ATTR IRAM_ATTR
#define ARDUINO_ISR_FLAG ESP_INTR_FLAG_IRAM
#else
#define ARDUINO_ISR_ATTR
#define ARDUINO_ISR_FLAG (0)
#endif
```
framework-arduinoespressif32/tools/sdk/esp32/sdkconfig:
```
# CONFIG_ARDUINO_ISR_IRAM is not set
```

kdomanski, 2023-01-29:
> Is WebUI using auto-reporting or polling?

Auto-reporting, 50ms.

---

Food for thought: while researching the issue, I came upon a post that suggested the following:
1. Turn the complex code that's currently called from ISR into a sleeping task.
2. The actual ISR only wakes/notifies that task (or posts data to a queue) and exits.
3. Keep `IRAM_ATTR` on the performance-critical code to avoid performance penalty.

Would that work?

MitchBradley, 2023-01-29:
I may have fixed it.  Re your suggestion: The step timing code n the GPIO case is too critical, by far, to do in a task.  In the I2SO case, it already works that way.

kdomanski, 2023-01-29:
Hey, I just saw the PR. Thanks for the quick response.

Tomorrow I'll try to reproduce with and without that patch.

kdomanski, 2023-02-02:
For the record, linking to results of the test: https://github.com/bdring/FluidNC/pull/796#issuecomment-1409529872

kdomanski, 2023-02-13:
Fixed in `Devt` by #796

commit 2488af11 (2023-02-10): Fix #794 - change Arduino __digital* to gpio_* in IRAM
commit a6d86bcf (2023-02-10): Merge pull request #796 from bdring/Fix794

Fix #794 - change Arduino __digital* to gpio_* in IRAM
closed 2023-02-13 by kdomanski