bdring/FluidNC #797: Problem: File taking long time for loading through webUI. after loading on clicking Start, machine does not take command and start. Sometimes it reset the Fluidnc.

Mamoonpk, 2023-01-30:
### Controller Board

6-pack controller

### Machine Description

External Driver DM556T  From Stepper online
Dual Y-Axis
End stops on all axes. 
Spindle is dewalt DWP611

### Configuration file

```yaml
name: 6 Pack External XYZ
board: 6-pack

stepping:
  engine: I2S_stream
  idle_ms: 255
  dir_delay_us: 6
  pulse_us: 4
  disable_delay_us: 5

axes:
  x:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 475.000
    soft_limits: false
    homing:
      cycle: 2
      mpos_mm: 0.000
      positive_direction: false
      settle_ms: 250.000
      seek_mm_per_min: 2000.000
      feed_mm_per_min: 200.000
      seek_scaler: 1.100
      feed_scaler: 5.000
    motor0:
      limit_all_pin: gpio.33
      hard_limits: true
      stepstick:
        direction_pin: i2so.1:low
        step_pin: i2so.2:low
        disable_pin: i2so.0
    
  y:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 475.000
    soft_limits: false
    homing:
      cycle: 3
      mpos_mm: 0.000
      positive_direction: false
      settle_ms: 250.000
      seek_mm_per_min: 2000.000
      feed_mm_per_min: 200.000
      seek_scaler: 1.100
      feed_scaler: 5.000
    motor0:
      limit_all_pin: gpio.35
      hard_limits: true
      pulloff_mm: 3.000
      stepstick:
        direction_pin: i2so.4:low
        step_pin: i2so.5:low
        disable_pin: i2so.7
    motor1:
      limit_all_pin: gpio.34
      hard_limits: true
      pulloff_mm: 3.000
      stepstick:
        direction_pin: i2so.12:low
        step_pin: i2so.13:low
        disable_pin: i2so.15
  z:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 70.000
    soft_limits: false
    homing:
      cycle: 1
      mpos_mm: 0.000
      positive_direction: true
      settle_ms: 250.000
      seek_mm_per_min: 2000.000
      feed_mm_per_min: 200.000
      seek_scaler: 1.100
      feed_scaler: 5.000
    motor0:
      limit_all_pin: gpio.32
      hard_limits: true
      stepstick:
        direction_pin: i2so.9:low
        step_pin: i2so.10:low
        disable_pin: i2so.8
    

i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5

control:
  reset_pin: gpio.2:pu
  
coolant:
  delay_ms: 1000.000

probe:
  pin: gpio.25:pu
  check_mode_start: false

macros:
  startup_line0: 
  startup_line1: 
  macro0: 
  macro1: 
  macro2: 
  macro3: 

start:
  must_home: true
  check_limits: true
  deactivate_parking: false

user_outputs:
arc_tolerance_mm: 0.002
junction_deviation_mm: 0.010
verbose_errors: false
report_inches: false
enable_parking_override_control: false
use_line_numbers: false
```

### Startup Messages

```text
after loading the program there is no message. the start button just enables
on clicking it does not proceed.
```

### User Interface Software

WebUI - Tablet mode using via WIFI

### What happened?

I was trying to make a 3d relief. 
I ran a 3 hrs rough pass successfully. 
Now i was trying to run the smooth pass. the g-code file size is 17.2MB and it is created using Easel pro. it took some time to load in WebUI and after that it enabled the start button. on pressing the start button it does not respond. couple of time it restarted as well.

I have a attached the gcode file.

[G21.docx](https://github.com/bdring/FluidNC/files/10532318/G21.docx)

Mamoonpk, 2023-01-30:
I am using Fluidnc V3.6.6 win64

MitchBradley, 2023-01-30:
Since you are using tablet mode, the delay is probably due to the visualizer.  To display the visualization, the software on the browser end has to ask the FluidNC controller to send it the GCode file from the SD card.  That can take a long time.  Then, once it gets the file, it has to process it twice - once to determine the overall size in order to scale the drawing to the viewport, and a second time to render the vectors into the viewport.  For large files that can take a long time. The visualizer has never been optimized for very large files.  Handling them well would probably require some sort of preview/thumbnail feature. I would like to add such a feature, but it is so far down the list that I might never get to it.  I might, however, add an option to disable the visualizer, or automatically disable it for large files.  Does your file work from the main UI?

Mamoonpk, 2023-01-30:
@MitchBradley 
Yes I checked and it worked on main UI.
I think disabling visualizer in webui will be a great help. It is not needed on all jobs. Overall I rate webui better than main UI.

Secondly I recently updated to 3.6.6. I noticed that if my tablet sleeps, my machine stops as well which was not the case earlier. Is it a feature added recently.

mani-fa, 2023-01-31:
> @MitchBradley Yes I checked and it worked on main UI. I think disabling visualizer in webui will be a great help. It is not needed on all jobs. Overall I rate webui better than main UI.
>
> Secondly I [...]

sleep causes stop working reported here 
https://github.com/bdring/FluidNC/issues/782
your browser is open during sleep but your wifi may turn off and machine can not continue to work.
(to use this version until new release , close browser before tablet sleep, nothing happen to working machine 

disable virtualizer helps a lot for big size.

commit 0f1b5a09 (2023-02-01): Merge pull request #800 from bdring/Fix797

Fix #797 - WebUI tablet hangs on huge GCode files
commit 7402ff03 (2023-02-01): Fix #797 - WebUI tablet hangs on huge GCode files
closed 2023-03-01 by bdring