bdring/FluidNC #801: Problem: Motion stop suddenly

lagnat, 2023-02-01:
### Controller Board

6-pack, BQ TMC5160s

### Machine Description

Gantry, XXYZ, Makita trim router spindle

### Configuration file

```yaml
board: 6 Pack
name: 6 Pack TMC5160 XXYZ
meta: 
stepping: 
  engine: I2S_static
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0
  segments: 12

axes: 
  shared_stepper_disable_pin: NO_PIN
  shared_stepper_reset_pin: NO_PIN
  x: 
    steps_per_mm: 160.000000
    max_rate_mm_per_min: 4000.000000
    acceleration_mm_per_sec2: 100.000000
    max_travel_mm: 800.000000
    soft_limits: false
    homing: 
      cycle: 1
      allow_single_axis: true
      positive_direction: false
      mpos_mm: 0.000000
      feed_mm_per_min: 200.000000
      seek_mm_per_min: 3000.000000
      settle_ms: 250
      seek_scaler: 1.000000
      feed_scaler: 1.500000

    motor0: 
      limit_neg_pin: gpio.25:low
      limit_pos_pin: gpio.2:low
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 4.400000
      tmc_5160: 
        step_pin: I2SO.2
        direction_pin: I2SO.1:low
        disable_pin: I2SO.0
        r_sense_ohms: 0.075000
        run_amps: 2.500000
        hold_amps: 1.250000
        microsteps: 8
        toff_disable: 0
        toff_stealthchop: 5
        use_enable: false
        cs_pin: I2SO.3
        spi_index: -1
        run_mode: CoolStep
        homing_mode: CoolStep
        stallguard: 16
        stallguard_debug: false
        toff_coolstep: 3
        tpfd: 4

    motor1: 
      limit_neg_pin: gpio.39:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 5.000000
      tmc_5160: 
        step_pin: I2SO.5
        direction_pin: I2SO.4:low
        disable_pin: I2SO.7
        r_sense_ohms: 0.075000
        run_amps: 2.500000
        hold_amps: 1.250000
        microsteps: 8
        toff_disable: 0
        toff_stealthchop: 5
        use_enable: false
        cs_pin: I2SO.6
        spi_index: -1
        run_mode: CoolStep
        homing_mode: CoolStep
        stallguard: 16
        stallguard_debug: false
        toff_coolstep: 3
        tpfd: 4

  y: 
    steps_per_mm: 160.000000
    max_rate_mm_per_min: 4000.000000
    acceleration_mm_per_sec2: 100.000000
    max_travel_mm: 550.000000
    soft_limits: false
    homing: 
      cycle: 1
      allow_single_axis: true
      positive_direction: false
      mpos_mm: 0.000000
      feed_mm_per_min: 200.000000
      seek_mm_per_min: 3000.000000
      settle_ms: 250
      seek_scaler: 1.000000
      feed_scaler: 1.500000

    motor0: 
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.26:low
      hard_limits: true
      pulloff_mm: 5.000000
      tmc_5160: 
        step_pin: I2SO.18
        direction_pin: I2SO.17:low
        disable_pin: I2SO.16
        r_sense_ohms: 0.075000
        run_amps: 2.500000
        hold_amps: 1.250000
        microsteps: 8
        toff_disable: 0
        toff_stealthchop: 5
        use_enable: false
        cs_pin: I2SO.19
        spi_index: -1
        run_mode: CoolStep
        homing_mode: CoolStep
        stallguard: 0
        stallguard_debug: false
        toff_coolstep: 3
        tpfd: 4

  z: 
    steps_per_mm: 400.000000
    max_rate_mm_per_min: 1500.000000
    acceleration_mm_per_sec2: 100.000000
    max_travel_mm: 300.000000
    soft_limits: false
    homing: 
      cycle: 1
      allow_single_axis: true
      positive_direction: true
      mpos_mm: 0.000000
      feed_mm_per_min: 200.000000
      seek_mm_per_min: 1000.000000
      settle_ms: 250
      seek_scaler: 1.000000
      feed_scaler: 1.500000

    motor0: 
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.16:low
      hard_limits: true
      pulloff_mm: 1.000000
      tmc_5160: 
        step_pin: I2SO.21
        direction_pin: I2SO.20:low
        disable_pin: I2SO.23
        r_sense_ohms: 0.075000
        run_amps: 2.500000
        hold_amps: 1.250000
        microsteps: 8
        toff_disable: 0
        toff_stealthchop: 5
        use_enable: false
        cs_pin: I2SO.22
        spi_index: -1
        run_mode: CoolStep
        homing_mode: CoolStep
        stallguard: 0
        stallguard_debug: false
        toff_coolstep: 3
        tpfd: 4

kinematics: 
  Cartesian: 

i2so: 
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

spi: 
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard: 
  cs_pin: gpio.5
  card_detect_pin: NO_PIN
  frequency_hz: 0

control: 
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

coolant: 
  flood_pin: NO_PIN
  mist_pin: NO_PIN
  delay_ms: 0

probe: 
  pin: gpio.27:low
  check_mode_start: true

macros: 
  startup_line0: 
  startup_line1: 
  macro0: 
  macro1: 
  macro2: 
  macro3: 

start: 
  must_home: false
  deactivate_parking: false
  check_limits: true

parking: 
  enable: false
  axis: Z
  target_mpos_mm: -5.000000
  rate_mm_per_min: 800.000000
  pullout_distance_mm: 5.000000
  pullout_rate_mm_per_min: 250.000000

user_outputs: 
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN

arc_tolerance_mm: 0.002000
junction_deviation_mm: 0.010000
verbose_errors: false
report_inches: false
enable_parking_override_control: false
use_line_numbers: false
planner_blocks: 16
Relay: 
  direction_pin: NO_PIN
  output_pin: NO_PIN
  enable_pin: I2SO.27
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 2000
  spindown_ms: 1000
  tool_num: 0
  speed_map: 0=0.000000% 0=100.000000% 1=100.000000%
  off_on_alarm: true

NoSpindle:
```

### Startup Messages

```text
INFO: FluidNC v3.6.6
INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e
INFO: Local filesystem type is spiffs
INFO: Configuration file:config.yaml
INFO: Machine 6 Pack TMC5160 XXYZ
INFO: Board 6 Pack
INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21
INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19
INFO: SD Card cs_pin:gpio.5 detect:NO_PIN
INFO: Stepping:I2S_static Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms
INFO: Axis count 3
INFO: Axis X (0.000,800.000)
INFO:   Motor0
INFO:     tmc_5160 Step:I2SO.2 Dir:I2SO.1:low CS:I2SO.3 Disable:I2SO.0 Index:-1 R:0.075
INFO:  X Neg Limit gpio.25:low
INFO:  X Pos Limit gpio.2:low
INFO:   Motor1
INFO:     tmc_5160 Step:I2SO.5 Dir:I2SO.4:low CS:I2SO.6 Disable:I2SO.7 Index:-1 R:0.075
INFO:  X2 Neg Limit gpio.39:low
INFO: Axis Y (0.000,550.000)
INFO:   Motor0
INFO:     tmc_5160 Step:I2SO.18 Dir:I2SO.17:low CS:I2SO.19 Disable:I2SO.16 Index:-1 R:0.075
INFO:  Y All Limit gpio.26:low
INFO: Axis Z (-300.000,0.000)
INFO:   Motor0
INFO:     tmc_5160 Step:I2SO.21 Dir:I2SO.20:low CS:I2SO.22 Disable:I2SO.23 Index:-1 R:0.075
INFO:  Z All Limit gpio.16:low
INFO: X Axis driver test passed
INFO: X2 Axis driver test passed
INFO: Y Axis driver test passed
INFO: Z Axis driver test passed
INFO: Kinematic system: Cartesian
INFO: Relay Spindle Ena:I2SO.27 Out:NO_PIN Dir:NO_PIN
INFO: Using spindle Relay
INFO: Probe Pin: gpio.27:low
INFO: STA SSID is not set
INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1
INFO: AP started
INFO: WiFi on
INFO: Captive Portal Started
INFO: HTTP started on port 80
INFO: Telnet started on port 23
```

### User Interface Software

UGS

### What happened?

I've been running the same gcode a few dozen times and most of the time, and after different lengths of time, the motion of all 3 axes will just stop.  I had 1 occasion where y stopped but x kept going for another second and then x stopped, but that only ever happened once.

When this happens, I have to disconnect UGS and reconnect, as it is not responsive.  The location is not lost.  I can just send the gcode again and it will eventually get caught up to where it stopped and then go further.  Sometimes it will stop again and sometimes it will go to completion.

I've tried running the gcode with the spindle in the air and there's no issue.  Given how easily reproducible the problem is, I would expect it to fail while just in air but I ran that test 3 of 4 times without issue.  I'm going to run this test a few more times.

My last observation is that motion stops at the end of a motion command and never in the middle of one.  It's as if UGS is just waiting for the OK and the OK doesn't come.  I don't have an easy way to manually send a command at this point to see if fluidnc is responsive.

As suggested I tried fiddling with `tpfd` but anything other than `4` and the steppers would not behave.

I also tried lowering the current to `1.0` but that didn't help either.

Please see the problem here: https://www.youtube.com/watch?v=AYaaO5UpeZQ

robercy, 2023-02-03:
Hello @lagnat

           Just for information, I had the same problem, then I switch to other computer, and everything works flawless. After compare both computers, I found that the "Put the computer to sleep" had a timeout set. I set to "Never", and solved the problem. 
  
  I hope this help you !

lagnat, 2023-02-04:
This issue ends up being fairly simple.  For a reason that I do not yet know, the serial connection is dropping.  UGS has a flaw where by it doesn't detect when the serial connection is lost so it appeared as if fluidnc was having an issue.  When fluid had finished generating steps for whatever was already buffered, motion stopped.

closed 2023-02-04 by lagnat