bdring/FluidNC #804: Losing step douring a job

Prelo96, 2023-02-05:
### Controller Board

custom
optoisolated input pd 
npn output 5v to the optocoupler.

### Machine Description

gantry router servo drives controlled in step/dir

### Configuration file

```yaml
name: "ESP32 Dev Controller V4"
board: "ESP32 Dev Controller V4"

stepping:
  engine: RMT
  idle_ms: 255
  dir_delay_us: 10
  pulse_us: 10
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: gpio.13:low
  
  x:
    steps_per_mm: 400
    max_rate_mm_per_min: 3000
    acceleration_mm_per_sec2:80
    max_travel_mm: 600
    homing:
      cycle: 2
      allow_single_axis: true
      positive_direction: false
      mpos_mm: -15
      seek_mm_per_min: 1000
      feed_mm_per_min: 20
      seek_scaler: 1.0
      feed_scaler: 1.0
    
    motor0:
      limit_all_pin: gpio.17:high:pd
      standard_stepper:
        direction_pin: gpio.14:low
        step_pin: gpio.12
    motor1:
      null_motor:

  y:
    steps_per_mm: 400
    max_rate_mm_per_min: 3000
    acceleration_mm_per_sec2: 80
    max_travel_mm: 400
    homing:
      cycle: 2
      allow_single_axis: true
      positive_direction: false
      mpos_mm: -15
      seek_mm_per_min: 1000
      feed_mm_per_min: 20
      seek_scaler: 1.0
      feed_scaler: 1.0

    motor0:
      limit_all_pin: gpio.4:high:pd
      standard_stepper:
        direction_pin: gpio.15:low
        step_pin: gpio.26
    motor1:
      null_motor:

  z:
    steps_per_mm: 400
    max_rate_mm_per_min: 3000
    acceleration_mm_per_sec2: 80
    max_travel_mm: 135
    homing:
      cycle: 1
      allow_single_axis: true
      positive_direction: true
      mpos_mm: 15
      seek_mm_per_min: 1000
      feed_mm_per_min: 20
      seek_scaler: 1.0
      feed_scaler: 1.0

    motor0:
      limit_all_pin: gpio.16:high:pd
      standard_stepper:
        direction_pin: gpio.33:low
        step_pin: gpio.27
    motor1:
      null_motor:

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN

coolant:
  flood_pin: gpio.22
  mist_pin: NO_PIN

        
probe:
  pin: gpio.32:high:pd
  check_mode_start: true

PWM:
  pwm_hz: 5000
  output_pin: gpio.25
  enable_pin: gpio.21:low
  direction_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 0
  speed_map: 0=0% 10000=100%
```

### Startup Messages

```text
[MSG:INFO: FluidNC v3.6.8-pre1]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine ESP32 Dev Controller V4]
[MSG:INFO: Board ESP32 Dev Controller V4]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN]
[MSG:INFO: Stepping:RMT Pulse:10us Dsbl Delay:0us Dir Delay:10us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable gpio.13:low]
[MSG:INFO: Axis X (-15.000,585.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.12 Dir:gpio.14:low Disable:NO_PIN]
[MSG:INFO:  X All Limit gpio.17:pd]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Y (-15.000,385.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.26 Dir:gpio.15:low Disable:NO_PIN]
[MSG:INFO:  Y All Limit gpio.4:pd]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Z (-120.000,15.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.27 Dir:gpio.33:low Disable:NO_PIN]
[MSG:INFO:  Z All Limit gpio.16:pd]
[MSG:INFO:   Motor1]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: PWM Spindle Ena:gpio.21:low Out:gpio.25 Dir:NO_PIN Freq:5000Hz Period:8191]
[MSG:INFO: Using spindle PWM]
[MSG:INFO: Flood coolant gpio.22]
[MSG:INFO: Probe Pin: gpio.32:pd]
[MSG:INFO: Connecting to STA SSID:TP-LINK_973A3E]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connecting...]
[MSG:INFO: Connected - IP is 192.168.0.100]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
```

### User Interface Software

webui

### What happened?

during a job got a series of missing step, drivers count still 0 when i came back to 0in the machine so no electric problem or the count of servo must be !0

the problem seems to start at line 14167
(parallelo 1mm)
(T4  D=1 CR=0.5 - ZMIN=-0.053 - fresa sferica)
G90 G94
G17
G21
(When using Fusion 360 for Personal Use, the feedrate of)
(rapid moves is reduced to match the feedrate of cutting)
(moves, which can increase machining time. Unrestricted rapid)
(moves are available with a Fusion 360 Subscription.)
G28 G91 Z0
G90

(1mmParallelo2)
T4
S18000 M3
G54
G0 X109.006 Y12.878
Z12
G1 Z3.302 F5000
Z0.096 F333
X109.001 Y12.873 Z0.058 F5000
X108.986 Y12.858 Z0.026
....
.....
X98.268 Y23.671 Z0.15
Z6.078
X103.619 Y29.314
Z5.643
Z0.122 F333
X103.613 Y29.308 Z0.084 F5000
X103.598 Y29.293 Z0.052
etc
....
....
X23.317 Y87.586 Z0.082
Z12
G28 G91 Z0
G90
G28 G91 X0 Y0
G90
M5
M30
end
or maybe in this portion 

X104.121 Y34.934
G3 X104.121 Y35.133 I-0.099 J0.099
X103.922 Y35.133 I-0.099 J-0.099
G1 X103.907 Y35.117 Z0.055
X103.891 Y35.101 Z0.05
X102.734 Y33.94

wrong gcode interpretation? 

repeating the gcode generate the same error again
at the end with the g28 the home is misalligned like loosing step but checked and the count of the driver is always 0, 
like a stepping procesor bug 

### Other Information

i will add some photo too
https://drive.google.com/drive/folders/1fB7DPPuY2qw04L8eZZJZwzCYJLDn5Kso?usp=share_link

MitchBradley, 2023-02-06:
Try downloading the GCode file from the SD Card to the host computer and comparing to the local copy, to see if the version on the SD Card got corrupted somehow.

MitchBradley, 2023-02-06:
> Checked again, when the tool retract and come back to the work position loose the position in negative and goes under the work position,

If you mean that the physical Z position is wrong, sometimes that can be caused by a shaft coupler that is loose.  My router was losing Y position recently and I found that the spring-type shaft coupler was actually broken, so motion worked in the direction where the broken ends were pushing against each other, but was intermittent in the other direction where the pieces could slip.

Prelo96, 2023-02-13:
Finally solved, I broke a coupler that slip a little little bit when rapid
movements occurs.
Thank you!

closed 2023-02-13 by Prelo96