bdring/FluidNC #805: Usb interface Problem:

Geekineer, 2023-02-08:
### Controller Board

6-pack

### Machine Description

Xyyz gantry

### Input Circuits

```yaml
NA
```

### Configuration file

```yaml
NA
```

### Startup Messages

```text
Na
```

### User Interface Software

USB,   uGS

### What happened?

Is my usb gui interface limited to 3rd party g codecsenders. UGS, oor can fluidnc web ui work without any wifi

### Other Information

Na

MitchBradley, 2023-02-08:
If you do not have WiFi in the location where you want to use FluidNC, you could use an old router as an access point without a backhaul to the internet.  Any router would do.  It does not have to be a modern fast one.  I have about 5 old ones gathering dust.  You, or a friend, probably does too.  If not, used routers cost about $15 on ebay.

Geekineer, 2023-02-08:
Great idea ---I do travel with this machine to the 'boonies; (no internet) so that works.
THANKYOU

closed 2023-02-08 by MitchBradley