bdring/FluidNC #807: gpio32 and 33 not working, how to connect to 4 port io board and 4.5v on spindle output always.

jappiemike, 2023-02-13:
### Controller Board

6-pack external drives 10v spindle board and 4 port io board

### Machine Description

Home made gantry router with  proximity sensors, probe and feed hold switches and hyuang VFD.

### Input Circuits

```yaml
All the inputs are through 4n25 optocouplers with 12V on the input side. the output side is 5V on the #5 pin and a 10K resistor to ground (shared with 6 pack board) and a wire going to the respective inputs 32-39.Using the $limits program #32 and #33 are not working. I am also having trouble getting the correct pin configuration for the probe and feed hold as I cannot determine which terminal of the 4 to connect to i.e. is gpio12 terminal 1 and gpio15 terminal 2?. I also have a constant 4.5 volts on the 10V spindle board regardless of the M3 or M5 commands. Today I noticed that the second led on drive 0 is not working. I'm yet to test the motors as I cannot get my homing sensors working. I've spent the last 3 weeks getting nowhere and am tempted to go back to my arduino uno after trying grblhal and a reprap board and now your board all to no avail. I've made so many changes to my config file I've accidently lost the original and would be most grateful if you could send me another copy.
Cheers
Mike
```

### Configuration file

```yaml
board: 6 Pack
name: 6 Pack External XYZ 10V
meta: 2022-08-20 B. Dring
start:
  must_home: false
  deactivate_parking: false
  check_limits: true
stepping:
  engine: I2S_STATIC
  idle_ms: 225
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0
axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 119.9
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 545.0
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100
    motor0:
      limit_neg_pin: gpio.34:low
      limit_pos_pin: gpio.35:low
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 5.000
      # Terminal Block 1
      standard_stepper:
        step_pin: I2SO.2
        direction_pin: I2SO.1: high
        disable_pin: I2SO.0
  y:
    steps_per_mm: 119.2
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 560.0
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100
    motor0:
      limit_neg_pin: gpio.36:low
      limit_pos_pin: gpio.39:low
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 5.000
      # Terminal Block 2
      standard_stepper:
        step_pin: I2SO.5
        direction_pin: I2SO.4:low
        disable_pin: I2SO.7
  z:
    steps_per_mm: 319.000
    max_rate_mm_per_min: 1500.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 60.000
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 50.000
      seek_mm_per_min: 50.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100
    motor0:
      limit_neg_pin: gpio.32:low
      limit_pos_pin: gpio.33:low
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 5.000
      # Terminal Block 3
      standard_stepper:
        step_pin: I2SO.10
        direction_pin: I2SO.9: low
        disable_pin: I2SO.8
probe:
  pin: gpio.26:high 
  check_mode_start: true
control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: gpio.15:high
  cycle_start_pin: gpio.13:high
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN
i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17
spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18
sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5
10V:
  forward_pin: NO_PIN
  reverse_pin: NO_PIN
  pwm_hz: 3000
  output_pin: gpio.14
  enable_pin: NO_PIN
  direction_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 5000
  spindown_ms: 5000
  tool_num: 0
  speed_map: 0=0.000% 1000=0.000% 24000=100.000%
  off_on_alarm: true
```

### Startup Messages

```text
FluidNC v1.2.0 using COM5
Exit: Ctrl-C, Ctrl-Q or Ctrl-], Clear screen: CTRL-W
Upload: Ctrl-U, Reset ESP32: Ctrl-R, Send Override: Ctrl-O
Resetting MCU

[MSG:INFO: FluidNC v3.6.5]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:6_Pext_XYZ_10V.yaml]
[MSG:INFO: Machine 6 Pack External XYZ 10V]
[MSG:INFO: Board 6 Pack]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN]
[MSG:INFO: Stepping:I2S_static Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:225ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (-545.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.2 Dir:I2SO.1 Disable:I2SO.0]
[MSG:INFO:  X Neg Limit gpio.34:low]
[MSG:INFO:  X Pos Limit gpio.35:low]
[MSG:INFO: Axis Y (-560.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.5 Dir:I2SO.4:low Disable:I2SO.7]
[MSG:INFO:  Y Neg Limit gpio.36:low]
[MSG:INFO:  Y Pos Limit gpio.39:low]
[MSG:INFO: Axis Z (-60.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.10 Dir:I2SO.9:low Disable:I2SO.8]
[MSG:INFO:  Z Neg Limit gpio.32:low]
[MSG:INFO:  Z Pos Limit gpio.33:low]
[MSG:INFO: feed_hold_pin gpio.15]
[MSG:INFO: cycle_start_pin gpio.13]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: 10V Spindle Ena:NO_PIN Out:gpio.14 Dir:NO_PIN Fwd:NO_PIN Rev:NO_PIN Freq:3000Hz Period:16383]
[MSG:INFO: Using spindle 10V]
[MSG:INFO: Probe Pin: gpio.26]
[MSG:INFO: BT Started with FluidNC]

Grbl 3.6 [FluidNC v3.6.5 (bt) '$' for help]
[MSG:INFO: Check limits]
[MSG:INFO: '$H'|'$X' to unlock]
```

### User Interface Software

fluidterm

### What happened?

I cannot get my gpio pins 32 and 33 to work, I even swap my X min and max from 34 and 35 which do work into 32 and 33 and they then stop working. I haven't any further than this, no motor testing or homing.

sjonholle, 2023-02-13:
https://esp32.com/viewtopic.php?t=9151 IO32 IO33 are connected to the crystal. Docs say you can't use these 2x pins as GPIO unless you desolder some stuff on the module.

Input-only pins: IO34, IO35, Sensor_VP, Sensor_VN, are labeled in docs as GPI (note the lack of O), which may mean they're input-only pins.

bdring, 2023-02-13:
You must use the pullup attribute on those pins. See this wiki page.

http://wiki.fluidnc.com/en/hardware/official/6_Pack_External#inputs

MitchBradley, 2023-02-13:
Please provide a schematic drawing of how your optocouplers are wired.  I am not exactly sure what you mean from the description.

ESP32 inputs are 3.3V, not 5V, so there could be a problem in that respect - but a drawing would help a lot.

jappiemike, 2023-02-14:
Hi, thanks for all the replies. As far as I can tell according to the picture on the external wiki page I'm dealing with GPIO 32 and 33 on the 6 pack board not the ESP pins. If my proximity sensors are NC as per best practice then surely having the pull ups on is counter intuitive as when the circuit opens the pull ups will want to negate the open circuit? I thought the reason for using your board was to provide a level of protection by using optocouplers to isolate the 3.3V environment from the outside, as it is we still need to provide our own level of isolation if we are using proximity switches most of which use greater than 5V as their power supply. If you look at https://www.hownottoengineer.com/projects/cnc/images/4n25.png this is how I've wired up all of my inputs to your board.
Thanks for all the inputs to some of my problems.
Cheers
Mike

bdring, 2023-02-14:
Here is the input circuit. The signals to the CPU pins need a pull-up.

[image]

Several pins have pull-ups. 32 and 33 need to use the internal ones on the CPU

[image]

bdring, 2023-02-15:
Please show the complete schematic of your input circuit including the switches and where the power and ground is coming from.

jappiemike, 2023-02-17:
> 
> 
> Please show the complete schematic of your input circuit including the switches and where the power and ground is coming from.
Hello, here is my schematic, It’s my first use of kicad so there might be errors but this is the gist of it.
￼

bdring, 2023-02-17:
I don't see anything. Paste a screenshot. Use the github interface, not an email reply.

jappiemike, 2023-02-18:
OK, lets try this again.
[Inputboard.pdf](https://github.com/bdring/FluidNC/files/10773465/Inputboard.pdf)

bdring, 2023-02-18:
The schematic does not tell the whole story. Power, etc.

1. You must use the the internal pullups on the 6 pack external regardless of what your circuitry is.

2. You must use a common ground with your circuitry and the 6 pack.

3. The 6 pack isolators are used to help with noise and spikes, but do not provide true galvanic isolation. To save costs, a 2nd isolated power supply was not used.

4. Adding another set of isolators in series with the 6 pack isolators complicates the situation.

jappiemike, 2023-02-19:
Hi, yes my schematic drawing skill are pretty dodgy. There is a common ground from the power supplies and the output side of the optocouplers to the ground on the board. I've managed to get all the limit  switches working now having used the low:pu modifier on pins 32 and 33, thanks for the help with that. I'm now trying to get the probe, feed hold and cyclestart pins to work. I've got them set to GPIO pins 12,15 and 13 on the header and I assumed they would connect to terminals 1,2 and 3 on the 4xinput schmitt but this appears not to be the case. I've got them all set as high:pd . I'm using the $limits utility to check the probe only at this time.
Cheers
Mike

jappiemike, 2023-02-19:
OK' I've now got the probe working by using GPIO.26 which I didn't think was connected to terminal 1 of the io board, but there you go. Only 2 more buttons to sort out and then the PWM signal to my VFD as well.
Cheers
Mike

jappiemike, 2023-02-25:
Hello, OK, I appear to have everything working now. Thank you for all your help and guidance.
Mike

closed 2023-02-25 by MitchBradley