bdring/FluidNC #811: Problem: Board crashes / stalls / disconnects from wifi when trying to upload a file to the SD card after there are about 150 or so programs already on the SD card.

brakthehun, 2023-02-19:
### Controller Board

MKS DLC32 V2.1

### Machine Description

Polar pen plotter

### Input Circuits

```yaml
Has 2 buttons - pen offset and park
It has homing of X / Y / Z
```

### Configuration file

```yaml
board: MKS-DLC32 V2.1
name: Big Polar CNC
meta: 9/22/2022 brakthehun

kinematics:
  Cartesian:

stepping:
  engine: I2S_STATIC
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0
axes:
  shared_stepper_disable_pin: I2SO.0
  x:
    steps_per_mm: 200
    max_rate_mm_per_min: 12000.000
    acceleration_mm_per_sec2: 2400.000
    max_travel_mm: 350
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 5000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.36:low
      hard_limits: false
      pulloff_mm: 3.000
      stepstick:
        step_pin: I2SO.1
        direction_pin: I2SO.2

  y:
    steps_per_mm: 106.666
    max_rate_mm_per_min: 4000.000
    acceleration_mm_per_sec2: 900.000
    max_travel_mm: 360
    soft_limits: false
    homing:
      cycle: 3
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 5000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.35:high
      hard_limits: false
      pulloff_mm: 8.000
      stepstick:
        step_pin: I2SO.5
        direction_pin: I2SO.6:high

  z:
    steps_per_mm: 200
    max_rate_mm_per_min: 6000
    acceleration_mm_per_sec2: 1200
    max_travel_mm: 200.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 1000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.34:low
      hard_limits: false
      pulloff_mm: 3.000
      stepstick:
        step_pin: I2SO.3
        direction_pin: I2SO.4

i2so:
  bck_pin: gpio.16
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.12
  mosi_pin: gpio.13
  sck_pin: gpio.14

sdcard:
  cs_pin: gpio.15
  card_detect_pin: NO_PIN

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: gpio.4:low:pu
  macro1_pin: gpio.0:low:pu
  macro2_pin: No_PIN
  macro3_pin: No_PIN

macros:
  startup_line0:G53X-297Y-69.25Z-100
  startup_line1:G92X0Y0Z0
  macro0: G53X0Z-100&G38.3Z-40F1000&G92Z-18&G0Z0&G0X0Y0
  macro1: G53X-297Y-69.25Z-10F5000
  macro2: G0X0Y0Z0
  macro3: G53Z-10

coolant:
  flood_pin: NO_PIN
  mist_pin: NO_PIN
  delay_ms: 0

probe:
  pin: gpio.22:low:PU
  check_mode_start: true

Laser:
  pwm_hz: 5000
  #L on Beeper / IN on TTL
  output_pin: gpio.32
  enable_pin: I2SO.7
  disable_with_s0: false
  s0_with_disable: false
  tool_num: 0
  speed_map: 0=0.000% 0=12.500% 1700=100.000%
# 135=0mA 270=5mA 400=10mA 700=16mA
user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN

start:
  must_home: true

# 5,18,19,22,23,25,26,27,32,33,39,I2SO.7
# SDA 0 / SCL 4
```

### Startup Messages

```text
It boots just fine,, for a while
No MSG:ERR or MSG:WARN
```

### User Interface Software

WebUI 

### What happened?

I tried uploading another file to the SD card and it hangs when it's almost finished uploading then then it stalls the board.
It also occasionally crashes when I'm running a file in a folder with 100s of other files.

### Other Information

I took all the .gcode files and placed them in a folder on the SD card and it boots fine and much faster and it's much more responsive when I run from an SD file. ( It used to take a couple seconds to start)

I was also unable to run it from the USB port using lightburn or LaserGRBL. It would crash soon after homing.

I was able to run it (jittery) over Telnet with LaserGRBL.

flavioprattico, 2023-02-22:
Try uploading this yaml file

https://github.com/makerbase-mks/FluidNC/blob/main/example_configs/MKS_DLC32_BASE.yaml

I had a similar problem few days ago.

brakthehun, 2023-02-22:
Line 1: it is the DLC32 2.1
Line 5: stream,, huh?? I thought the MKS wasn't supposed to be streamed. I'll try it!!
Line 6: I need the idle set to 255 because the Zaxis is belt driven and will fall (crash) if the motor lets go.
Line 102 and 103 are swapped. Should I change that also??
All the other stepping and directions are wrong too. I'll put all my stuff in and try it.

From what I see about I2S stream,, is it's preferred unless you're using a laser which this one is not,, so,, I think I'll just change that.

flavioprattico, 2023-02-22:
I tested it without changing anything and worked fine for MKS DLC32 V2.1. I tested it on router CNC machine (no laser). Let me know if it works.

MitchBradley, 2023-02-22:
It is possible that SD cards with a lot of files on them are consuming too much system memory.

closed 2025-01-03 by bdring