bdring/FluidNC #812: Problem: System Crashes during run

snakescb, 2023-02-23:
### Controller Board

Custmo Controller, based on 6-Pack schematic, no SD Card Slot

### Machine Description

DIY machine

### Configuration file

```yaml
board: NCMate
name: CNC PitschBitsch

start:
  must_home: true
  
stepping:
  engine: I2S_STATIC
  idle_ms: 255
  pulse_us: 5
  dir_delay_us: 0
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  
  x:
    steps_per_mm: 640.000
    max_rate_mm_per_min: 4000.000
    acceleration_mm_per_sec2: 150.000
    max_travel_mm: 420.000
    soft_limits: true
    
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: -10.000
      feed_mm_per_min:  800.000
      seek_mm_per_min: 2000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.2:low:pu
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 7.750

      standard_stepper:
        step_pin: i2so.6
        direction_pin: i2so.5:low
        disable_pin: i2so.4:low

  y:
    steps_per_mm: 640.000
    max_rate_mm_per_min: 4000.000
    acceleration_mm_per_sec2: 150.000
    max_travel_mm: 510.000
    soft_limits: true
    
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 505.000
      feed_mm_per_min:  100.000
      seek_mm_per_min: 2000.000
      settle_ms: 500
[... 214 lines cut ...]
[MSG:INFO: Axis X (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Axis Y (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Axis Z (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: Connecting to STA SSID:Einstein]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 10.64.20.155]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
Grbl 3.6 [FluidNC v3.6.8-pre1 (wifi) '$' for help]
[MSG:INFO: '$H'|'$X' to unlock]

[1001.txt](https://github.com/bdring/FluidNC/files/10811907/1001.txt)

Thank you for having a look, really hope I can help to get this fixed.

Best regards
Christian

MitchBradley, 2023-02-23:
This is going to be excruciating to debug.  The first crash has almost no information because the backtrace got corrupted.  The only information is that the program counter is somewhere deep in the Espressif network stack.  The secondary crash has more information - the memory allocator heap is messed up.  It is happening in response to a '?' status report request, when the C++ standard ostringstream library is creating an ostringstream object, specifically when it is determining the local-specific numeric punctuation character.  It tries to free some allocated memory and finds that the heap data structure is corrupted.  Stuff like this, where some key system data structure has gone bad, can take forever to track down.  Sigh.

MitchBradley, 2023-02-23:
Since you are using CNCjs, you could try turning off WiFi with $wifi/mode=off

snakescb, 2023-02-23:
Thanks Bradley. This may be very very well related to the network stack, almost reproducable. I have several reboots sometimes, always as soon as the WIFI is connected. See below the full reboot sequence that happens SOMETIMES. As soon as I disable WIFI, not issues dooring boot anymore. Will try to run my program today keeping WIFI disabled and let you know the result.

> $bye
[MSG:INFO: Restarting]
ets Jun  8 2016 00:22:57
rst:0xc (SW_CPU_RESET),boot:0x3b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5856
entry 0x400806a8
[MSG:INFO: FluidNC v3.6.8-pre1]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine CNC PitschBitsch]
[MSG:INFO: Board NCMate]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: Stepping:I2S_static Pulse:5us Dsbl Delay:0us Dir Delay:0us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (-10.000,410.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.6 Dir:I2SO.5:low Disable:I2SO.4:low]
[MSG:INFO:  X Neg Limit gpio.2:low:pu]
[MSG:INFO: Axis Y (-5.000,505.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.10 Dir:I2SO.9 Disable:I2SO.8:low]
[MSG:INFO:  Y Pos Limit gpio.15:low:pu]
[MSG:INFO:   Motor1]
[MSG:INFO:     standard_stepper Step:I2SO.2 Dir:I2SO.1 Disable:I2SO.0:low]
[MSG:INFO:  Y2 Pos Limit gpio.13:low:pu]
[MSG:INFO: Axis Z (0.000,120.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.14 Dir:I2SO.13:low Disable:I2SO.12:low]
[MSG:INFO:  Z Pos Limit gpio.12:low:pu]
[MSG:INFO: safety_door_pin gpio.25:low]
[MSG:INFO: cycle_start_pin gpio.33:low]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: 10V Spindle Ena:NO_PIN Out:gpio.16 Dir:NO_PIN Fwd:I2SO.16 Rev:I2SO.17 Freq:5000Hz Period:8191]
[MSG:INFO: Using spindle 10V]
[MSG:INFO: Flood coolant I2SO.23]
[MSG:INFO: Mist coolant I2SO.22]
[MSG:INFO: Probe Pin: gpio.4:low]
[MSG:INFO: Connecting to STA SSID:Einstein]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 10.64.20.155]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
Grbl 3.6 [FluidNC v3.6.8-pre1 (wifi) '$' for help]
[MSG:INFO: '$H'|'$X' to unlock]
assert failed: heap_caps_free heap_caps.c:339 (heap != NULL && "free() target pointer is outside heap areas")
Backtrace:0x400859b1:0x3ffe81600x4008fadd:0x3ffe8180 0x40095b85:0x3ffe81a0 0x40085e6a:0x3ffe82d0 0x40095bb5:0x3ffe82f0 0x401a3c59:0x3ffe8310 0x401ad0c0:0x3ffe8330 0x401b37a5:0x3ffe8350 0x401b3a67:0x3ffe8370 0x401b3ee6:0x3ffe83a0 0x401b4254:0x3ffe83e0 0x401b4282:0x3ffe8400 0x401b4292:0x3ffe8420 0x401a6ae5:0x3ffe8450 0x401a6a31:0x3ffe8470 0x401b3353:0x3ffe8490 0x400ee2d6:0x3ffe84b0 0x400ef37e:0x3ffe85b0 0x400f008b:0x3ffe8630 0x400d5725:0x3ffe8660 0x400f67c7:0x3ffe8690 0x400efbb2:0x3ffe86b0 0x400efcbb:0x3ffe86e0 0x400ed0c3:0x3ffe8700
ELF file SHA256: 0000000000000000
Rebooting...
ets Jun  8 2016 00:22:57
rst:0xc (SW_CPU_RESET),boot:0x3b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5856
entry 0x400806a8
[... 172 lines cut ...]
[MSG:INFO: Using default configuration]
[MSG:INFO: Axes: using defaults]
[MSG:INFO: Machine Default (Test Drive)]
[MSG:INFO: Board None]
[MSG:INFO: Stepping:RMT Pulse:4us Dsbl Delay:0us Dir Delay:0us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Axis Y (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Axis Z (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: Connecting to STA SSID:Einstein]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 10.64.20.155]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
Grbl 3.6 [FluidNC v3.6.8-pre1 (wifi) '$' for help]
[MSG:INFO: '$H'|'$X' to unlock]
>

MitchBradley, 2023-02-23:
Try adding 100 uF to the 5V rail and the 3.3V rail.  I have seen lots of reboots when the wifi turns on and the voltage dips.  It is usually not a problem with DevKitC modules from Espressif, but I have some  esp-mini modules that are particularly bad about it.  I always add extra capacitance with them.  50uF is probably enough, so long as the capacitor is good quality.  I use 100uF just for extra safety margin.

snakescb, 2023-02-25:
ok, disabling WIFI was not doing the trick, but the error message changed, and is not corrupted this time:

Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.
Core  0 register dump:
PC      : 0x400d94c4  PS      : 0x00060330  A0      : 0x801a5e94  A1      : 0x3ffda920
A2      : 0x3ffda970  A3      : 0x00000000  A4      : 0x0000000f  A5      : 0xc0184c04
A6      : 0x0000000f  A7      : 0x3ffded70  A8      : 0x800d94e9  A9      : 0x3ffda900
A10     : 0x3ffded70  A11     : 0x3ffdac34  A12     : 0x0000000f  A13     : 0x3ffdac44
A14     : 0x00000020  A15     : 0x00000001  SAR     : 0x0000001f  EXCCAUSE: 0x0000001d
EXCVADDR: 0xc0184c04  LBEG    : 0x4008bac4  LEND    : 0x4008bada  LCOUNT  : 0xffffffff
Backtrace:0x400d94c1:0x3ffda9200x401a5e91:0x3ffda950 0x401b3102:0x3ffda970 0x401b5086:0x3ffda9b0 0x401df329:0x3ffda9d0 0x401ae229:0x3ffda9f0 0x401af22f:0x3ffdaa30 0x401af255:0x3ffdaae0 0x401ded79:0x3ffdab30 0x401a68b6:0x3ffdab80 0x401a695d:0x3ffdabd0 0x400ee339:0x3ffdabf0 0x400ef37e:0x3ffdacf0 0x400f008b:0x3ffdad70 0x400d5725:0x3ffdada0 0x400f67c7:0x3ffdadd0 0x400efb81:0x3ffdadf0 0x400efcbb:0x3ffdae20 0x400ed0c3:0x3ffdae40
ELF file SHA256: 0000000000000000
Rebooting...
ets Jun  8 2016 00:22:57
rst:0xc (SW_CPU_RESET),boot:0x3f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:12812
load:0x40080400,len:3032
entry 0x400805e4
[MSG:INFO: FluidNC v3.6.8-pre1]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:ERR: Skipping configuration file due to panic]
[MSG:INFO: Using default configuration]
[MSG:INFO: Axes: using defaults]
[MSG:INFO: Machine Default (Test Drive)]
[MSG:INFO: Board None]
[MSG:INFO: Stepping:RMT Pulse:4us Dsbl Delay:0us Dir Delay:0us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Axis Y (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Axis Z (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: WiFi is disabled]
Grbl 3.6 [FluidNC v3.6.8-pre1 (wifi) '$' for help]
[MSG

snakescb, 2023-02-25:
I am digging into it. For now, it looks more HW / Controller related, maybe power supply or anything else. Will elminiate all these concerns now before anybody putting more time in it. Thank you!

browntyler, 2023-06-29:
I'm having the same issue, running FluidNC Wifi version 3.7.1 on an XPro V5, connected to a Huanyang VFD. Often when booting up it will crash. Sending '$Bye' will sometimes reboot it into the correct setup, but most often it will boot into a default configuration and nothing will work until I'm able to boot into my proper configuration.

Here is the console output:
`load:0x3fff0030,len:1184
load:0x40078000,len:12812
load:0x40080400,len:3032
entry 0x400805e4
[MSG:INFO: FluidNC v3.7.1]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:OpenBuilds-Lead1515.yaml]
[MSG:WARN: Ignored key digital0_pi]
[MSG:WARN: Ignored key spinup_ms]
[MSG:WARN: Ignored key spindown_ms]
[MSG:WARN: Ignored key tool_num]
[MSG:WARN: Ignored key speed_map]
[MSG:INFO: Machine CNC_xPRO_V5]
[MSG:INFO: Board xPRO_v5]
[MSG:INFO: UART1 Tx:gpio.4 Rx:gpio.25 RTS:NO_PIN Baud:9600]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:8000000]
[MSG:INFO: Stepping:RMT Pulse:4us Dsbl Delay:0us Dir Delay:0us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (1.000,1171.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_5160 Step:gpio.12 Dir:gpio.14 CS:gpio.17 Disable:NO_PIN Index:1 R:0.050]
[MSG:INFO:  X Neg Limit gpio.35]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Y (1.000,301.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_5160 Step:gpio.27 Dir:gpio.26 CS:NO_PIN Disable:NO_PIN Index:2 R:0.050]
[MSG:INFO:  Y Neg Limit gpio.34:low]
[MSG:INFO:   Motor1]
[MSG:INFO:     tmc_5160 Step:gpio.33 Dir:gpio.32 CS:NO_PIN Disable:NO_PIN Index:3 R:0.050]
[MSG:INFO: Axis Z (-299.000,1.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_5160 Step:gpio.15 Dir:gpio.2 CS:NO_PIN Disable:NO_PIN Index:4 R:0.050]
[MSG:INFO:  Z Neg Limit gpio.39:low]
[MSG:INFO:   Motor1]
[MSG:ERR: X Axis TMC driver not detected - expected 0x0x30 got 0x0x0]
[MSG:ERR: Y Axis TMC driver not detected - expected 0x0x30 got 0x0x0]
[MSG:ERR: Y2 Axis TMC driver not detected - expected 0x0x30 got 0x0x0]
[MSG:ERR: Z Axis TMC driver not detected - expected 0x0x30 got 0x0x0]
[MSG:INFO: safety_door_pin gpio.16:low:pu]
[MSG:INFO: macro0_pin gpio.13:low:pu]
[MSG:INFO: macro1_pin gpio.0:low:pu]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Huanyang Spindle  Tx:gpio.4 Rx:gpio.25 RTS:NO_PIN Baud:9600]
[MSG:INFO: Using spindle Huanyang]
[MSG:INFO: Mist coolant gpio.21]
[MSG:INFO: Probe Pin: gpio.22:low:pu]
[MSG:INFO: STA SSID is not set]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
Grbl 3.7 [FluidNC v3.7.1 (wifi) '$' for help]
client $$
[MSG:INFO: '$H'|'$X' to unlock]
$10=1 (Status report options, mask)
ok
assert failed: heap_caps_free heap_caps.c:360 (heap != NULL && "free() target pointer is outside heap areas")
Backtrace: 0x4008553d:0x3ffe8e90 0x4008fa05:0x3ffe8eb0 0x400965a1:0x3ffe8ed0 0x40085a7a:0x3ffe9000 0x400965d1:0x3ffe9020 0x401afb25:0x3ffe9040 0x401b4b58:0x3ffe9060 0x401b0e29:0x3ffe9080 0x401b10eb:0x3ffe90a0 0x401b1476:0x3ffe90d0 0x401b19a8:0x3ffe9110 0x401b19d6:0x3ffe9130 0x401b19e6:0x3ffe9150 0x401c6329:0x3ffe9180 0x401ba8b5:0x3ffe91a0 0x401b40cb:0x3ffe91c0 0x400f19aa:0x3ffe91e0 0x400f2a76:0x3ffe92e0 0x400f37a3:0x3ffe9360 0x400d5881:0x3ffe9390 0x400fa337:0x3ffe93c0 0x400f32c6:0x3ffe93e0 0x400f33cf:0x3ffe9410 0x400f06fb:0x3ffe9430
ELF file SHA256: c0232e7857604a85
Rebooting...`

MitchBradley, 2023-06-29:
It's not the same issue at all.  There are many things that can result in a backtrace.  In your case, one problem - but perhaps not the only one - is that the config file is malformed, based on the WARN messages.

Start a new issue and provide all of the information that the issue form asks for.

closed 2025-01-21 by bdring