bdring/FluidNC #813: System failures, does not work for more than 5 minutes.

us6iae, 2023-02-23:
### Controller Board

Custom controller, tested on two controllers.

### Machine Description

Gantry router with external DM542 stepper drivers.

### Configuration file

```yaml
name: "ESP32_V1"
board: "ESP32"
meta: "1_ver"

stepping:
  engine: RMT
  idle_ms: 250
  dir_delay_us: 2
  pulse_us: 4
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN

  x:
    steps_per_mm: 640
    max_rate_mm_per_min: 2500
    acceleration_mm_per_sec2: 250
    max_travel_mm: 500
    homing:
      cycle: 2
      mpos_mm: 10
      feed_mm_per_min: 50
      seek_mm_per_min: 200
      settle_ms: 500
      positive_direction: false
    
    motor0:
      limit_all_pin: gpio.34:low
      hard_limits: false
      pulloff_mm: 1
      stepstick:
        direction_pin: gpio.4
        step_pin: gpio.15

  y:
    steps_per_mm: 640
    max_rate_mm_per_min: 2500
    acceleration_mm_per_sec2: 250
    max_travel_mm: 500
    homing:
      cycle: 2
      mpos_mm: 10
      feed_mm_per_min: 50
      seek_mm_per_min: 200
      settle_ms: 500
      positive_direction: false
    
    motor0:
      limit_all_pin: gpio.39:low
      hard_limits: false
      pulloff_mm: 1
      stepstick:
        direction_pin: gpio.27
        step_pin: gpio.25

  z:
    steps_per_mm: 640
    max_rate_mm_per_min: 2500
    acceleration_mm_per_sec2: 200
    max_travel_mm: 120
    homing:
      cycle: 1
      mpos_mm: 10
      feed_mm_per_min: 50
      seek_mm_per_min: 200
      settle_ms: 500
      positive_direction: true
    
    motor0:
      limit_all_pin: gpio.36:low
      hard_limits: false
      pulloff_mm: 3
      stepstick:
        direction_pin: gpio.26
        step_pin: gpio.12

  a:
    steps_per_mm: 93
    max_rate_mm_per_min: 200
    acceleration_mm_per_sec2: 25
    max_travel_mm: 360
    
    motor0:
      stepstick:
        direction_pin: gpio.32
        step_pin: gpio.33

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN
  frequency_hz: 10000000

coolant:
  flood_pin: NO_PIN

probe:
  pin: gpio.16:low:pu

user_outputs:
  digital0_pin: NO_PIN

PWM:
  pwm_hz: 5000
  output_pin: gpio.2
  enable_pin: gpio.17
  disable_with_s0: false
  s0_with_disable: true
  speed_map: 0=0% 20000=100%
```

### Startup Messages

```text
$SS
[MSG:INFO: FluidNC v3.6.8-pre1]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:4axis_v1.yaml]
[MSG:INFO: Machine ESP32_V1]
[MSG:INFO: Board ESP32]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN]
[MSG:INFO: Stepping:RMT Pulse:4us Dsbl Delay:0us Dir Delay:2us Idle Delay:250ms]
[MSG:INFO: Axis count 4]
[MSG:INFO: Axis X (10.000,510.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.15 Dir:gpio.4 Disable:NO_PIN]
[MSG:INFO:  X All Limit gpio.34:low]
[MSG:INFO: Axis Y (10.000,510.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.25 Dir:gpio.27 Disable:NO_PIN]
[MSG:INFO:  Y All Limit gpio.39:low]
[MSG:INFO: Axis Z (-110.000,10.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.12 Dir:gpio.26 Disable:NO_PIN]
[MSG:INFO:  Z All Limit gpio.36:low]
[MSG:INFO: Axis A (-360.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.33 Dir:gpio.32 Disable:NO_PIN]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: PWM Spindle Ena:gpio.17 Out:gpio.2 Dir:NO_PIN Freq:5000Hz Period:8191]
[MSG:INFO: Using spindle PWM]
[MSG:INFO: Probe Pin: gpio.16:low:pu]
[MSG:INFO: STA SSID is not set]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
```

### User Interface Software

WebUi

### What happened?

The system stops after 1-5 minutes and is silent. After 1-2 minutes, he takes a few steps and is silent again. So several times. Then hangs forever. Changed code files, changed maps, Changed controllers. Nothing helped.
What else can you try?

bdring, 2023-02-23:
You are not giving much about what happens or anything you have tried to debug the problem. It could be that the stepper drivers are not working. Overheating, power problems, signal issues, etc.

 - Send a photo your controller and wiring.
 - How are you sending the commands? Are you using a gcode sender, the WebUI, SD card, etc.
 - Does FluidNC lockup or do only the motors just stop moving.
 - Give as many details as possible.

us6iae, 2023-02-23:
Homemade controller, drivers DM542.
[image]
The codes file is sent from the card via the web interface.
The tablet is also not working.
Fluidnc is blocked.

us6iae, 2023-02-23:
I assembled a new assembly of the controller processor+card. The firmware does not work on it either.
[image]

bdring, 2023-02-23:
It is likely due to the DIY SD. See this page on the Wiki.

http://wiki.fluidnc.com/en/config/sd_card#help-troubleshooting

We do not provide support for DIY SD card applications. You can search the Github issues for other people having trouble with SD adapters and some possible solutions.

us6iae, 2023-02-23:
Noticed:
AutorePort Interval: 50 works very unstable. freezes after 2-3min.
AutorePort Interval: 500 is already better. freezes after 5-10min.
AutorePort Interval: 1000 is even better. Hangs after 15-20min.

us6iae, 2023-02-23:
I have no problems with SD. Everything is done there correctly and works well.

us6iae, 2023-02-23:
This is an option for tests.

MitchBradley, 2023-02-23:
I will search for memory leaks in the auto report code.

us6iae, 2023-02-24:
Addition:
When you run the file with the "macro0: $SD/Run=kub58mm.nc" macro, everything works if the web is not enabled.
Sincerely, Alexander.

us6iae, 2023-03-05:
Addition:
I changed the ESP 32F processor, installed a new version 3.6.8, the display, the result is the same - it stops after a while. After stopping, the display says IDLE instead of RUN. When continuing, writes RUN. But it became better to work, if earlier it failed by 40-45%, now by 80-90%.
[image]

MitchBradley, 2023-03-05:
We do not support using AP mode for production runs.  The AP code in the Espressif libraries appears to leak memory, causing problems after long periods of operation.  We support AP mode for initial setup only.  Use STA mode for long term production use.  If you do not have WiFi in your shop, you can get a dedicated WiFi access point/router to create a private network.  The router can be an old/slow one, perhaps one that was previously taken out of service.

us6iae, 2023-03-08:
Thank you, this has improved the performance a lot. Failures have become less frequent and shorter, but they still exist. I will test on a real machine.

closed 2023-03-08 by us6iae

MitchBradley, 2023-03-09:
Failures should not happen when using STA mode