bdring/FluidNC #819: WEB-UI --> FLUIDNC Setting --> Config Items --> are not saved (DEVT 01.03.2023)

PatrikRindlisbacher, 2023-03-02:
### Controller Board

6-pack with 2x4 Input's & MosFet

### Machine Description

Test Station 3 Axis

### Input Circuits

```yaml
Default Setting / and User setting's
```

### Configuration file

```yaml
Default Setting / and User setting's
```

### Startup Messages

```text
$G
$Report/Interval=50
[MSG:DBG: WebSocket 0 from 192.168.5.4 uri /]
```

### User Interface Software

Web UI

### What happened?

WEB-UI --> FLUIDNC Setting --> Config Items --> are not saved (DEVT 01.03.2023)

### Other Information

Tested many parameters, it doesn't seem to be saved.

(DEVT 01.03.2023)

bdring, 2023-03-02:
You are mixing terms. **$Report/Interval** is not a setting or config item. It is a command.

***Commands*** are things like **$H** or **$Startup/Show**. They do something, but do not save any state.

***Settings*** are things like **$Wifi/Mode**. They are stored in the **eeprom** section of the memory partition scheme. They are automatically saved.

***Config*** items like **$start/must_home** can be changed on the fly, but are not saved. You must send **$CD=<filename>** to save them.

PatrikRindlisbacher, 2023-03-02:
[image]

bdring, 2023-03-02:
Words?

PatrikRindlisbacher, 2023-03-02:
I made the process as a picture

Avataar120, 2023-03-02:
Did you try : 
$start/must_home=true
$CD=config.yaml

?

PatrikRindlisbacher, 2023-03-02:
I'm 2-3 hours away now, I'll test later

MitchBradley, 2023-03-03:
This is a known problem with WebUI

PatrikRindlisbacher, 2023-03-03:
Didn't find an #entry in the booklet,
it's already known in which region the error is.

MitchBradley, 2023-03-04:
We have not yet implemented a WebUI button to save the results of config item changes.  I prefer to edit the config file on a host computer, so there is a record of the current state - and possibly history if source control is used.

liwajaya, 2024-09-13:
hello, is this issue with editing via web browser resolved?