bdring/FluidNC #820: Physical joystick 5pin arduino with Z-axis stepper motor control with smooth speed control

regin92, 2023-03-04:
### Machine Context

Physical joystick 5pin arduino with Z-axis stepper motor control with smooth speed control

Hello, I need your help.
I want to embed a 5 pin physical joystick into a GRBL ESP32 system.
I have a laser machine, ESP32 TINYBEE board, ESP32-GRBL, 5 PIN ARDUINO joystick, sketch with code for joystick and stepper motor.
Everything individually works very well.
esp32 grbl and tinybee work well.
sketch and tinybee work well.

[image]

Here is my sketch.

#define step_pin 21   // Pin 21 connected to Steps pin on EasyDriver 
#define dir_pin 15   // Pin 15 connected to Direction pin           
#define en_pin 23       // Pin 23 connected to en pin             
#define A_pin 35    // Pin 35 connected to joystick x axis       

int A = 0;
int T = 1;

void Step() {
  pinMode(23, OUTPUT);
  pinMode(21, OUTPUT);
   digitalWrite(21, 1);
  delayMicroseconds(T);
  pinMode(23, OUTPUT);
  pinMode(21, OUTPUT);
   digitalWrite(21, 0);
  delayMicroseconds(T);
}

void setup() {

}

void loop() {
    A = analogRead(35);
    Serial.println(A);
    if (A > 2800) {
      digitalWrite(23, LOW);
      T = map(A, 2800, 4095, 10000, 700);
      pinMode(15, OUTPUT);
       digitalWrite(15, 1);
      Step();
    }
    if (A < 1590) {
      digitalWrite(23, LOW);
      T = map(A, 1590, 0, 10000, 700);
      pinMode(15, OUTPUT);
       digitalWrite(15, 0);
      Step();
    }
    if (A > 1590 & A < 2800) {
      digitalWrite(23, HIGH);
      pinMode(15, OUTPUT);
       digitalWrite(15, 0);
      pinMode(21, OUTPUT);
       digitalWrite(21, 0);
    }

}

I need to create two files djoy.h and djoy.cpp in the \Grbl_Esp32\src folders and transfer my code from the sketch into them.
Help me please, rewrite my code in these files djoy.h and djoy.cpp and I will learn from this example to write another code for ESP32-GRBL.
I will test everything and show the results, maybe this will be useful to many.

after looking at the work of the code, I want to continue to insert new functions into my machine, for example, to touch the screen, etc. and share with other participants.
i need 2 files to make djoy h and djoy. cpp, I will then compile them myself into esp32_grbl

### Feature Description

..... 

### Other Approaches

.. 

### How I Can Help

..

regin92, 2023-03-04:
` #define step_pin 21   // Pin 21 connected to Steps pin on EasyDriver 
#define dir_pin 15   // Pin 15 connected to Direction pin           
#define en_pin 23       // Pin 23 connected to en pin             
#define A_pin 35    // Pin 35 connected to joystick x axis       

int A = 0;
int T = 1;

void Step() {
  pinMode(23, OUTPUT);
  pinMode(21, OUTPUT);
   digitalWrite(21, 1);
  delayMicroseconds(T);
  pinMode(23, OUTPUT);
  pinMode(21, OUTPUT);
   digitalWrite(21, 0);
  delayMicroseconds(T);
}

void setup() {

}

void loop() {
    A = analogRead(35);
    Serial.println(A);
    if (A > 2800) {
      digitalWrite(23, LOW);
      T = map(A, 2800, 4095, 10000, 700);
      pinMode(15, OUTPUT);
       digitalWrite(15, 1);
      Step();
    }
    if (A < 1590) {
      digitalWrite(23, LOW);
      T = map(A, 1590, 0, 10000, 700);
      pinMode(15, OUTPUT);
       digitalWrite(15, 0);
      Step();
    }
    if (A > 1590 & A < 2800) {
      digitalWrite(23, HIGH);
      pinMode(15, OUTPUT);
       digitalWrite(15, 0);
      pinMode(21, OUTPUT);
       digitalWrite(21, 0);
    }

}
   `

PatrikRindlisbacher, 2023-03-04:
Hi
It's not going to be that easy.
it is easier to transfer the jog commands via UART. But even there it needs a bit more code.
greeting Patrik

MitchBradley, 2023-03-04:
We are not supporting Grbl_Esp32 any more.  All work going forward is focused on FluidNC

MitchBradley, 2023-03-04:
MKS Tinybee does not pulse the steppers via GPIO pins.  It uses shift registers that are driven from an I2S serial bus.  The code to control that is very complicated.  There is no simple way to integrate you sketch into either Grbl_Esp32 or FluidNC.  It would be like trying to integrate a go-kart into a bus.

regin92, 2023-03-04:
I have a FluidNC system and there is no difference in file organization. I will compile to
FluidNC
I only need your help to transfer the code to these two files, and then I will try, change, etc. It won't be difficult for you, and without you I can't begin to understand how to build this code.
I have a separate drive for the Z axis and I use only Gpio for them
[image]

regin92, 2023-03-04:
[image]

MitchBradley, 2023-03-04:
[You must use platformio to compile](http://wiki.fluidnc.com/en/installation#compiling-reference-only-you-do-not-need-to-compile). Any .cpp file that is in the src/ directory will be compiled and added to the code.  You will need to replace the Arduino loop() routine with a FreeRTOS task, and initialize that task from init() in Main.cpp.

Since your Z axis does not interact with the rest of the FluidNC code, it would be better to leave it as-is, driven from a separate Arduino.

closed 2023-03-04 by MitchBradley