bdring/FluidNC #821: Feature:

regin92, 2023-03-04:
### Machine Context

Physical joystick 5pin arduino with Z-axis stepper motor control with smooth speed control

Hello, I need your help.
I want to embed a 5 pin physical joystick into a GRBL ESP32 system.
I have a laser machine, ESP32 TINYBEE board, ESP32-GRBL, 5 PIN ARDUINO joystick, sketch with code for joystick and stepper motor.
Everything individually works very well.
esp32 grbl and tinybee work well.
sketch and tinybee work well.

[image]

Here is my sketch.

#define step_pin 21   // Pin 21 connected to Steps pin on EasyDriver 
#define dir_pin 15   // Pin 15 connected to Direction pin           
#define en_pin 23       // Pin 23 connected to en pin             
#define A_pin 35    // Pin 35 connected to joystick x axis       

int A = 0;
int T = 1;

void Step() {
  pinMode(23, OUTPUT);
  pinMode(21, OUTPUT);
   digitalWrite(21, 1);
  delayMicroseconds(T);
  pinMode(23, OUTPUT);
  pinMode(21, OUTPUT);
   digitalWrite(21, 0);
  delayMicroseconds(T);
}

void setup() {

}

void loop() {
    A = analogRead(35);
    Serial.println(A);
    if (A > 2800) {
      digitalWrite(23, LOW);
      T = map(A, 2800, 4095, 10000, 700);
      pinMode(15, OUTPUT);
       digitalWrite(15, 1);
      Step();
    }
    if (A < 1590) {
      digitalWrite(23, LOW);
      T = map(A, 1590, 0, 10000, 700);
      pinMode(15, OUTPUT);
       digitalWrite(15, 0);
      Step();
    }
    if (A > 1590 & A < 2800) {
      digitalWrite(23, HIGH);
      pinMode(15, OUTPUT);
       digitalWrite(15, 0);
      pinMode(21, OUTPUT);
       digitalWrite(21, 0);
    }
[Grbl_Esp32 djoystick.ino.zip](https://github.com/bdring/FluidNC/files/10888467/Grbl_Esp32.djoystick.ino.zip)

}

I need to create two files djoy.h and djoy.cpp in the \Grbl_Esp32\src folders and transfer my code from the sketch into them.
Help me please, rewrite my code in these files djoy.h and djoy.cpp and I will learn from this example to write another code for ESP32-GRBL.
I will test everything and show the results, maybe this will be useful to many.

after looking at the work of the code, I want to continue to insert new functions into my machine, for example, to touch the screen, etc. and share with other participants.
i need 2 files to make djoy h and djoy. cpp, I will then compile them myself into esp32_grbl

### Feature Description

..... 

### Other Approaches

.. 

### How I Can Help

..

macmimy, 2023-05-29:
not sure on the technicals and what not but I think i2c should be fine but idk I'm a dumby  BUT what about QWIIC over i2c ? https://www.sparkfun.com/qwiic

MitchBradley, 2023-05-29:
This is the FluidNC repo, not Grbl_Esp32.  Grbl_Esp32 is no longer being maintained.  For FluidNC we recommend that user interface gadgets like this should be implemented with separate microcontrollers that talk to FluidNC over one of the channel interfaces like serial or websockets.  Adding UI code to the FluidNC code base creates maintenance problems and possible issues with the performance of the realtime motion control.

closed 2023-05-29 by MitchBradley