bdring/FluidNC #822: Problem: Display

us6iae, 2023-03-05:
### Controller Board

4-pack

### Machine Description

Gantry router with external DM542 stepper drivers.

### Configuration file

```yaml
name: "ESP32_V1"
board: "ESP32"
meta: "1_ver"

stepping:
  engine: RMT
  idle_ms: 250
  dir_delay_us: 2
  pulse_us: 4
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN

  x:
    steps_per_mm: 640
    max_rate_mm_per_min: 2500
    acceleration_mm_per_sec2: 250
    max_travel_mm: 500
    homing:
      cycle: -1
      mpos_mm: 10
      feed_mm_per_min: 50
      seek_mm_per_min: 200
      settle_ms: 500
      positive_direction: false
    
    motor0:
      limit_all_pin: gpio.34:low
      hard_limits: false
      pulloff_mm: 1
      stepstick:
        direction_pin: gpio.4
        step_pin: gpio.22

  y:
    steps_per_mm: 640
    max_rate_mm_per_min: 2500
    acceleration_mm_per_sec2: 250
    max_travel_mm: 500
    homing:
      cycle: -1
      mpos_mm: 10
      feed_mm_per_min: 50
      seek_mm_per_min: 200
      settle_ms: 500
      positive_direction: false
    
    motor0:
      limit_all_pin: gpio.39:low
      hard_limits: false
      pulloff_mm: 1
      stepstick:
        direction_pin: gpio.27
        step_pin: gpio.25

  z:
    steps_per_mm: 640
    max_rate_mm_per_min: 2500
    acceleration_mm_per_sec2: 200
    max_travel_mm: 120
    homing:
      cycle: -1
      mpos_mm: 10
      feed_mm_per_min: 50
      seek_mm_per_min: 200
      settle_ms: 500
      positive_direction: true
    
    motor0:
      limit_all_pin: gpio.36:low
      hard_limits: false
      pulloff_mm: 3
      stepstick:
        direction_pin: gpio.26
        step_pin: gpio.12

  a:
    steps_per_mm: 93
    max_rate_mm_per_min: 200
    acceleration_mm_per_sec2: 25
    max_travel_mm: 360
    
    motor0:
      stepstick:
        direction_pin: gpio.32
        step_pin: gpio.33

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN
#  frequency_hz: 10000000

i2c0:
   sda_pin: gpio.15
   scl_pin: gpio.14

oled:
   i2c_num: 0
   i2c_address: 60
   width: 128
   height: 64
   radio_delay_ms: 1000

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
#  macro0_pin: gpio.35:low
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

macros:
  startup_line0:
  startup_line1:
#  macro0: $X&$SD/Run=kub58mm.nc
  macro1:
  macro2:
  macro3:

coolant:
  flood_pin: NO_PIN

probe:
  pin: gpio.16:low:pu

user_outputs:
  digital0_pin: NO_PIN

PWM:
  pwm_hz: 5000
  output_pin: gpio.2
  enable_pin: gpio.17
  disable_with_s0: false
  s0_with_disable: true
  speed_map: 0=0% 20000=100%
```

### Startup Messages

```text
$ss
[MSG:INFO: FluidNC v3.6.8]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:4axis_v2.yaml]
[MSG:INFO: Machine ESP32_V1]
[MSG:INFO: Board ESP32]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:0]
[MSG:INFO: I2C SDA: gpio.15, SCL: gpio.14, Freq: 100000, Bus #: 0]
[MSG:INFO: OLED I2C address:0x3c width: 128 height: 64]
[MSG:INFO: Stepping:RMT Pulse:4us Dsbl Delay:0us Dir Delay:2us Idle Delay:250ms]
[MSG:INFO: Axis count 4]
[MSG:INFO: Axis X (10.000,510.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.22 Dir:gpio.4 Disable:NO_PIN]
[MSG:INFO:  X All Limit gpio.34:low]
[MSG:INFO: Axis Y (10.000,510.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.25 Dir:gpio.27 Disable:NO_PIN]
[MSG:INFO:  Y All Limit gpio.39:low]
[MSG:INFO: Axis Z (-110.000,10.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.12 Dir:gpio.26 Disable:NO_PIN]
[MSG:INFO:  Z All Limit gpio.36:low]
[MSG:INFO: Axis A (-360.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.33 Dir:gpio.32 Disable:NO_PIN]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: PWM Spindle Ena:gpio.17 Out:gpio.2 Dir:NO_PIN Freq:5000Hz Period:8191]
[MSG:INFO: Using spindle PWM]
[MSG:INFO: Probe Pin: gpio.16:low:pu]
[MSG:INFO: STA SSID is not set]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
```

### What happened?

Version 3.6.8
When executing the Jog Z10-100 command, the display shows:
0.000
1.XXX
2.XXX
3.XXX
4.XXX
5.XXX
6.XXX
0.000 (All axes show 0.000)
7.XXX
8.XXX
9.XXX
10.000
WEB shows correctly. The error is not critical, but you need to look.

MitchBradley, 2023-03-05:
What exactly is "the Jog Z10-100 command"?  Did you click a jog button on WebUI or issue a command via a serial port, or what?

us6iae, 2023-03-05:
G1Z0.0F100
G1Z10F100
[image]
[image]

MitchBradley, 2023-03-05:
I think I solved the problem.  Your description confused me because I did not understand what you meant by "XXX".  I now think that you meant that "XXX" is the correct data, but periodically the display showed "0" instead of the correct data.  In the future, please try to be more clear in your descriptions.

us6iae, 2023-03-06:
Sorry, this is a language barrier. If it was possible to attach a video it would be easier. You got it right.

qrstu, 2023-03-06:
I noticed when I pushed one of the homing buttons x,y,z all Axis displayed 0,0,0 on OLED
In some cases when I pushed the z homing switch it came back with the x=-231 y=-451 z=-451 (Value by power on)
I don't know where this values are comming from ? Can't find them in config.yaml
Display of Icon X,Y,Z Button pushed is working correct.

qrstu, 2023-03-06:
x=-231 y=-451 z=-451 --> Could you replace this values with "not homed" ?
Because when i move it shows correct values.
But it shouldn't "0" all Axis when only push one homing button. (ex. x,y,z)
Not homed axis can be "0" on first Axis move.

MitchBradley, 2023-03-06:
Somebody else will have to do this.  I an unwilling to spent any more time fine tuning this.  I initially resisted doing it at all because I knew that people would want an endless list of tweaks.  I should have stuck with that.

qrstu, 2023-03-06:
Sorry for my request. I tested OLED funcionality yesterday. I recognized that it's not as expected.
If you can give me some hints where to implement. I will see whqat I can do. I'm not famillary with the code and
I'm not a really programmer

PatrikRindlisbacher, 2023-03-06:
Short input
I am working on a solution for these requests. (extern via UART)
I hope to deliver first results in a few days / weeks.
I'm also still open to input

Yours sincerely, Patrik Rindlibacher

closed 2023-04-27 by bdring