bdring/FluidNC #823: Feature: warning soft limit Before start the job.

RDS356, 2023-03-06:
### Machine Context

All.

### Feature Description

Sometimes It happens to me. I belive that happens with other users.
Lost the job because the machine reached the soft limits when this job is almost in the final.
Imagine if FluidNC can report before you start the job about the possible soft limits. Is it possible ?

### Other Approaches

.

### How I Can Help

.

SnowBeaver, 2023-03-06:
I took the habit to execute an Outline command before Starting the Job. Both Lightburn and gSender offer this functionality and I imagine it is a common feature amongst gcode senders. I do not have soft limits enable on my 6-pack, but I believe that if the job  crosses soft limits, executing Outline will show you where it will hit one.

MitchBradley, 2023-03-06:
Try sending $c before running the jog. That will run it in check mode without actual motion

closed 2024-02-03 by bdring