bdring/FluidNC #824: Feature: Serial Input (74hct165)

Niko98407, 2023-03-07:
### Machine Context

I am currently building an CNC Laser with ganged X-Axis, Y Axis and Optional Z- and A-Axis -> 5Axis so far. In an attempt to futureproof my custom PCB, I have added an optional 6th axis output. I am controlling all Stepper Motors with the Stepstick interface -> STEP/DIR Mode (Pins are conected via I2S (74HCT595)), since this gives me the option to use external stepper drivers.
The Problem is, when using the input features (Macros, Pause, Resume, Endstops,etc.) and analog outputs for a VFD-Controlled suction, I am running out of pins on the ESP32.

### Feature Description

The Feature would be to use 74HCT165 Parallel to Serial Registers. As most schematics already use I2S Outputs, I guess it would even be possible to use the existing interface (Clock and Load) maybe with the ability to have a second I2S Bus to seperate inputs and outputs. Since it uses a 16MHz clock, I also guess thats even fast enough to get accurate endstop positions at reaonable speeds.

### Other Approaches

It would give the enduser the option to use as many other features as possible, and save the valuable ESP32 GPIO Pins for various bus systems or other GPIO funcions that are only available at the ESP32 GPIO's

### How I Can Help

I would be happy to test this feature. Sadly my C++ skills are not quite good enough to implement this feature myself. At least for now.

MitchBradley, 2023-03-07:
We have considered this before - the suggestion has come up many times - and rejected it for various technical reasons that I do not have time to reiterate.

The approaches that we are considering for input expansion include I2C - for which some implementations are already in the PR phase - and UART.  One approach that I particularly like is to use a separate low-end ($1) MCU that tracks input states and sends a short report over a UART line when something changes.

Niko98407, 2023-03-08:
Thanks for the fast reply!
Alright, good to know. The I2C approach sounds interesting, I guess your talking about something like the PCF8574.
Just throwing another MCU on the board seems a bit unnecessary in my opinion. Just another MCU to be programmed, the hardware programming interface has to be implemented in the pcb-design...

MitchBradley, 2023-03-08:
An MCU often costs less than a PCF8574.  With a dumb I2C chip, you either have to poll it frequently which can be a real-time headache, or else burn a third pin for an interrupt, which adds yet another level of programming headache on the ESP32 side. 
Once the MCU programming environment is set up and you have the basic code structure in place, you have a lot of options for improvement.