bdring/FluidNC #830: Embroidery mode

Ams-Laser, 2023-03-10:
### Machine Context

Im building a Embroidery machine, i currently use GRBL on it and the sender takes care of sending the G1 move lines one by one only on a joystick button input

### Feature Description

Fluid nc Need to check if next line in buffer is a G1 move (or can be GE new code?) and if it is do not send the line untill a i/o pin is pulsed (stitch completed) this way we move the fabric only when we have completed a stitch and the needle is in the up position

### Other Approaches

Im currently usin a custom sender app (wich i can provide) this app only sends the move after a joystick button is pressed, however this puts a limit on the stitches per minute becouse the i/o and code overhead, integrating into fluidnc will permit much faster operation, plus the use of the sd card for local jobs instead of having the gui connected all times

### How I Can Help

i can provide the current code i have (mostly python) help test and develop, provide video feedback schematics and ideas.

Thanks in advance

MitchBradley, 2023-03-10:
Can you estimate the possible improvement amount?  If you send a gcode move in short form - without the G1, and possibly using G91 mode so you can send short offsets - the line is often about 5 bytes long.  That takes about half a millisecond to send.  Parsing takes almost no time.  Let's say that the I/O plus parsing overhead is 1 millisecond for the sake of discussion.  Can the machine run fast enough that 1 ms overhead matters?

Ams-Laser, 2023-03-10:
> Exciting task please send a photo. possibly if you like in the FluidNC Discord Chat

i will, where can i find the discord invite ?

> Can you estimate the possible improvement amount? If you send a gcode move in short form - without the G1, and possibly using G91 mode so you can send short offsets - the line is often about 5 bytes [...]

it might be my sender and not nesesarly the overhead bit i have seen machines reacing 800+stitches per minute while im way behind

Ams-Laser, 2023-03-10:
https://user-images.githubusercontent.com/87819514/224433512-54e61fc6-b747-4dd0-b2ed-71137404c088.mp4

here is a video of the prototype machine, i know much greater speed can be acheived
thanks

MitchBradley, 2023-03-10:
The answer to many questions, including the discord one, can be found by search in the fluidnc wiki.

800/min is 13/sec or 1 stitch per 75 milliseconds.  That rate should be achievable easily if the sender works well.

Ams-Laser, 2023-03-10:
> The answer to many questions, including the discord one, can be found by search in the fluidnc wiki.
> 
> 800/min is 13/sec or 1 stitch per 75 milliseconds. That rate should be achievable easily if the sender works well.

Mitch, 
the most valuable assets for me to try to get a "embroidery mode" are

1 not depending on a lptop or a computer being connected to the machine
2 not having extra hardware such as a joystick interface 
3 being able to use aof the shelf fluidnc boards so anyone can make its own machine
4 being able to load files to sd and run from there

if the answer is we wont add features that can be handled by sender i will understand and look elswhere for help, but i still think it would be a good enhancement in functionality as for cnc sewing machines.

thanks.
Ed.

MitchBradley, 2023-03-10:
The value of this feature to me personally is rather small.  It seems like a modification that benefits only a very small number of potential users - given that you are the first person to mention cnc embroidery - while costing us time to implement and support.  If you are looking to create a commercial product that you support, and are willing to pay us to implement some custom code, the equation changes.  If it is just "hey, will you work for me for free", then, well, you know ...

Ams-Laser, 2023-03-10:
how much would somethink like what im asking costs ?
i pesonaly build laser cuttes, and i use light burn as sender and GRBL on atmega for them, but hey if its something that wont break the bank or cost more than alternate solutions for embroidery i might give it a shot

MitchBradley, 2023-03-10:
You really need to look at it in the context of the total system cost.  You already need some sort of controller hardware to run FluidNC.  FluidNC can speak to multiple UI agents (senders) via channels like UART, bluetooth, wifi websockets, etc.  You can buy another ESP32-S2 for about $2 and program it to be a "smart sender" that knows what to do about the embroidery application.  It is hard to beat the economics of that.

Ams-Laser, 2023-03-10:
how much would you charge for programing a smart sender replacing my sender ?
i can send you what i currently have so you can see how it works
is there another more direct way to contact you ?

MitchBradley, 2023-03-10:
I am so far behind on my other projects right now that I would not be able to give you good service.

Have a look at issue #759 .  The people who are participating in that discussion are thinking about pendants, which is very related.  They might be able to help.  You can post to that issue and ask them to contact you, or get onto our Discord server (search for "discord" in the FluidNC wiki) and try to message them.

PatrikRindlisbacher, 2023-03-10:
Hi Mitch am already studying :-)

bdring, 2023-03-11:
Can you control the needle motor via gcode? I think it would be easy to coordinate the motion that way without changes. 

That would also allow you to move the fabric and needle at the same time.

MitchBradley, 2023-03-11:
Good idea about gcode controlling the needle. The needle could be treated as a spindle and controlled with S words, or as a digital out controlled with M62/M63.  Either way it would be synchronized with motion - the next move would not start until the "spindle" has been activated.

Another synchronization approach would be to add explicit delays between motion lines via the standard GCode "dwell" command - G4 Pn.  Dwell waits for motion to complete then pauses for n seconds.  You could set n to be the time it takes to lift the needle, say 0.050 if the needle goes up in 50 milliseconds

PatrikRindlisbacher, 2023-03-11:
Poor guy has been in prison in Mexico for 11.5 years. Another 6 months and then he'll be free again. He is now building machines for his colleagues in prison to earn some money. Got lots of photos

PatrikRindlisbacher, 2023-03-11:
The needle must be the boss over a sensor. Marlin knows wait for pin M226

PatrikRindlisbacher, 2023-03-11:
[image]

bdring, 2024-12-12:
Closing inactive issue

closed 2024-12-12 by bdring