bdring/FluidNC #831: Problem: Enabling i2so  bck_pin on gpio.22 breaks WiFi.

colin-campbell, 2023-03-10:
### Controller Board

"ESP32 6 axis breakout board" (listed in wiki - existing hardware) 

### Machine Description

Gantry router with external DM542 stepper drivers.

### Configuration file

```yaml
board: 6axis CNC controller
name: FluidTest
meta: "2023-03-08 steenhive"
stepping:
  engine: I2S_STREAM
  idle_ms: 250
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 1500.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.2
        direction_pin: I2SO.1
        disable_pin: I2SO.0

  y:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 1500.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 1500.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.32
      hard_limits: true
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.5
        direction_pin: I2SO.4
        disable_pin: I2SO.7

    motor1:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.13
        direction_pin: I2SO.12
        disable_pin: I2SO.15

  z:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 500.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 500.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.35
      hard_limits: true
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.10
        direction_pin: I2SO.9
        disable_pin: I2SO.8

i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

# spi:
#   miso_pin: NO_PIN
#   mosi_pin: NO_PIN
#   sck_pin: NO_PIN

# sdcard:
#   card_detect_pin: NO_PIN
#   cs_pin: NO_PIN

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: gpio.36
  cycle_start_pin: NO_PIN
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

coolant:
  flood_pin: NO_PIN
  mist_pin: gpio.4
  delay_ms: 0

probe:
  pin: gpio.34
  check_mode_start: true

macros:
  startup_line0:
  startup_line1:
  macro0:
  macro1:
  macro2:
  macro3:

start:
  must_home: false

PWM:
  pwm_hz: 5000
  output_pin: gpio.27
  enable_pin: NO_PIN
  direction_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 1000
  spindown_ms: 1000
  tool_num: 0
  speed_map: 0=0.000% 1000=100.000%
```

### Startup Messages

```text
[MSG:INFO: FluidNC v3.6.8]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine FluidTest]
[MSG:INFO: Board 6axis CNC controller]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: Stepping:I2S_stream Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:250ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (150.000,1650.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.2 Dir:I2SO.1 Disable:I2SO.0]
[MSG:INFO: Axis Y (0.000,1500.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.5 Dir:I2SO.4 Disable:I2SO.7]
[MSG:INFO:  Y All Limit gpio.32]
[MSG:INFO:   Motor1]
[MSG:INFO:     standard_stepper Step:I2SO.13 Dir:I2SO.12 Disable:I2SO.15]
[MSG:INFO: Axis Z (0.000,500.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.10 Dir:I2SO.9 Disable:I2SO.8]
[MSG:INFO:  Z All Limit gpio.35]
[MSG:INFO: feed_hold_pin gpio.36]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: PWM Spindle Ena:NO_PIN Out:gpio.27 Dir:NO_PIN Freq:5000Hz Period:8191]
[MSG:INFO: Using spindle PWM]
[MSG:INFO: Mist coolant gpio.4]
[MSG:INFO: Probe Pin: gpio.34]
[MSG:INFO: Connecting to STA SSID:yahozna2]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: No SSID]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
```

### User Interface Software

minicom

### What happened?

Booting with the default config connects to WiFi in station mode without a problem, but adding a config.yaml with a correct i2so key such as that listed here: http://wiki.fluidnc.com/en/hardware/3rd-party/ESP32_6_axis_breakout_board prevents the WiFi connecting. The board actually runs fine, just without WiFi

If I place the i2so bck_pin on an unused gpio, then the WiFi works again, but of course the i2s is now non-functional.   
Note: Although the messages indicate the fallback AP has started, there is no SSID broadcast. 

### Other Information

Messages booting with default config:
[MSG:INFO: FluidNC v3.6.8]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:WARN: Cannot open configuration file:config.yaml]
[MSG:INFO: Using default configuration]
[MSG:INFO: Axes: using defaults]
[MSG:INFO: Machine Default (Test Drive)]
[MSG:INFO: Board None]
[MSG:INFO: Stepping:RMT Pulse:4us Dsbl Delay:0us Dir Delay:0us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Axis Y (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Axis Z (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: Connecting to STA SSID:yahozna2]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connecting...]
[MSG:INFO: Connected - IP is 10.0.0.168]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://bawbag-cnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.6 [FluidNC v3.6.8 (wifi) '$' for help]
[MSG:INFO: '$H'|'$X' to unlock]

MitchBradley, 2023-03-10:
You might also try to find contact information for the person who created the wiki page for that hardware.  The edit history says that it is "Myles Newton".  The FluidNC development team does not have an instances of that hardware for testing.

colin-campbell, 2023-03-10:
Thanks. I'll try to follow up with the wiki page creator. The board is Chinesium, but nothing funny looking in the layout - WROOM-32 driving 74AHC595D. I was wondering if you might know anything that could cause a conflict, before I get a scope onto the pins.  Otherwise, it'll be fine without WiFi.

MitchBradley, 2023-03-10:
The only thing that comes to mind: Maybe when you enable I2S, the steppers turn on and that causes electrical interference that interferes with WiFi.

MitchBradley, 2023-03-10:
One more thing - when the wifi radio turns on, there is surge in the power that the chip draws.  I have seen modules with insufficient filter capacitance on either 3v3 or 5v or both, that brownout when wifi comes on.  Perhaps the board is right on the edge so when stepping is enabled, it doesn't quite have enough to run the wifi.  Bad filter capacitor?

colin-campbell, 2023-03-10:
I have the board disconnected, and hooked up to my lab supply with plenty of oomph - I observe the same behaviour. But I shall check the filter cap and buck converter, thank you.

colin-campbell, 2023-04-04:
Well I fixed it by cutting the trace to gpio.22 and wiring i2so bck_pin manually to the unused gpio.5. 
I wonder if it has anything to do with gpio.22 being U0RTS. Whatever, close this, since the board is listed now as unsupported.

closed 2023-04-04 by bdring