bdring/FluidNC #833: ProblemWebUI lags after some time, restarts.:

BlueOrangeLive, 2023-03-13:
### Controller Board

Custom 

### Machine Description

Router with external stepper driver xyz motors, endstops on all axes. Spindle 800Wats switsch by hand

### Configuration file

```yaml
name: "CoCo_CNC_Relay"
board: "ESP32 Dev Controller HJH"

stepping:
  engine: RMT
  idle_ms: 255
  dir_delay_us: 1
  pulse_us: 5
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: gpio.13:low

  x:
    steps_per_mm: 1280.000
    max_rate_mm_per_min: 1600
    acceleration_mm_per_sec2: 200
    max_travel_mm: 243
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0
      feed_mm_per_min: 200.000
      seek_mm_per_min: 700.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.17:low:pu
      hard_limits: false
      pulloff_mm: 3.000
      stepstick:
        direction_pin: gpio.14
        step_pin: gpio.12

  y:
    steps_per_mm: 1280.000
    max_rate_mm_per_min: 1600
    acceleration_mm_per_sec2: 200
    max_travel_mm: 332
    soft_limits: true
    homing:
      cycle: 3
      positive_direction: false
      mpos_mm: 0
      feed_mm_per_min: 200.000
      seek_mm_per_min: 700.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.4:low:pu
      hard_limits: false
      pulloff_mm: 3.000
      stepstick:
        direction_pin: gpio.15:low
        step_pin: gpio.26

  z:
    steps_per_mm: 1280.000
    max_rate_mm_per_min: 700
    acceleration_mm_per_sec2: 200
    max_travel_mm: 102
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 0
      feed_mm_per_min: 200.000
      seek_mm_per_min: 500.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.16:low:pu
      hard_limits: false
      pulloff_mm: 3.000
      stepstick:
        direction_pin: gpio.33
        step_pin: gpio.27

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN
  frequency_hz: 10000000

i2c0:
  sda_pin: gpio.21
  scl_pin: gpio.22
  frequency: 400000

oled:
  i2c_num: 0
  i2c_address: 61
  width: 128
  height: 64
  radio_delay_ms: 1000

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: gpio.36:low
  cycle_start_pin: gpio.39:low
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

macros:
  startup_line0:
  startup_line1:
  macro0:
  macro1:
  macro2:
  macro3:

coolant:
  mist_pin:  gpio.25

probe:
  pin: gpio.32:low:pu
  check_mode_start: false

relay:
  output_pin: gpio.2
  enable_pin: NO_PIN
  direction_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 6000
  spindown_ms: 600
  tool_num: 0
  speed_map: 0=0.000% 0=100.000% 1=100%
  off_on_alarm: true

start:
  must_home: false
  deactivate_parking: false
  check_limits: true
```

### Startup Messages

```text
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:12812
load:0x40080400,len:3032
entry 0x400805e4

[MSG:INFO: FluidNC v3.6.8]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:CoCoCNC_Relay_OLED_61.yaml]
[MSG:INFO: Machine CoCo_CNC_Relay]
[MSG:INFO: Board ESP32 Dev Controller HJH]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:10000000]
hz = 10000000
[MSG:INFO: I2C SDA: gpio.21, SCL: gpio.22, Freq: 400000, Bus #: 0]
[MSG:INFO: OLED I2C address:0x3d width: 128 height: 64]
[MSG:INFO: Stepping:RMT Pulse:5us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable gpio.13:low]
[MSG:INFO: Axis X (0.000,243.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.12 Dir:gpio.14 Disable:NO_PIN]
[MSG:INFO:  X All Limit gpio.17:low:pu]
[MSG:INFO: Axis Y (0.000,332.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.26 Dir:gpio.15:low Disable:NO_PIN]
[MSG:INFO:  Y All Limit gpio.4:low:pu]
[MSG:INFO: Axis Z (-102.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.27 Dir:gpio.33 Disable:NO_PIN]
[MSG:INFO:  Z All Limit gpio.16:low:pu]
[MSG:INFO: feed_hold_pin gpio.36:low]
[MSG:INFO: cycle_start_pin gpio.39:low]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Relay Spindle Ena:NO_PIN Out:gpio.2 Dir:NO_PIN]
[MSG:INFO: Using spindle Relay]
[MSG:INFO: Mist coolant gpio.25]
[MSG:INFO: Probe Pin: gpio.32:low:pu]
[MSG:INFO: Connecting to STA SSID:Negruj2G]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
E (6270) wifi:Association refused temporarily, comeback time 1024 mSec
[MSG:INFO: Connecting...]
[MSG:INFO: Connecting....]
[MSG:INFO: Connected - IP is 192.168.188.54]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.6 [FluidNC v3.6.8 (wifi) '$' for help]
```

### User Interface Software

WebUI

### What happened?

During milling, the fourth part was finished, but the movements were still visible on the tablet.
After that I did it on my test unit, with and without OLed display. From the third time it starts slowly. 
The fourth time the delay was 35 seconds but sometimes 2 3 minutes. After that, the WebUI restarted.
 I tested it on two different tablets and my PC. Browser is Firefox and Chrome.
I held it tight in the video. To be seen on Youtube.
https://www.youtube.com/watch?v=-oGQCcNOR6w

### Other Information

Begin Y_StirnProfiel_V4.nc on android tablets.
Serial output on the PC.

[MSG:INFO: Tool No: 4]
[MSG:INFO: Tool No: 4]
[MSG:INFO: Tool No: 4]
[MSG:INFO: Tool No: 4]
[MSG: /sd/Y_StirnProfiel_V4.nc file job succeeded]
[MSG:INFO: Tool No: 4]
[MSG:INFO: Tool No: 4]
[MSG:INFO: Tool No: 4]
[MSG:INFO: Tool No: 4]
[MSG: /sd/Y_StirnProfiel_V4.nc file job succeeded]
[MSG:INFO: Tool No: 4]
[MSG:INFO: Tool No: 4]
[MSG:INFO: Tool No: 4]
[MSG:INFO: Tool No: 4]
[MSG: /sd/Y_StirnProfiel_V4.nc file job succeeded]
[MSG:INFO: Tool No: 4]
[MSG:INFO: Tool No: 4]
[MSG:INFO: Tool No: 4]
[MSG:INFO: Tool No: 4]
[MSG: /sd/Y_StirnProfiel_V4.nc file job succeeded]  (about 15 seconds longer on the tablet)
[MSG:INFO: Tool No: 4]
[MSG:INFO: Tool No: 4]
[MSG:INFO: Tool No: 4]
[MSG:INFO: Tool No: 4]
[MSG: /sd/Y_StirnProfiel_V4.nc file job succeeded]  (about 35 seconds longer on the tablet)
[MSG:INFO: Channel auto report interval set to 50 ms]  (WebUI is restarted about 10 seconds after the job)
 
[Y_StirnProfiel_V4.nc.txt](https://github.com/bdring/FluidNC/files/10955835/Y_StirnProfiel_V4.nc.txt)

BlueOrangeLive, 2023-05-02:
This morning I tested it with the Code Branch   "CacheWebUI". It also happens here after 2 - 3 times in a row use of the file SchluesselAutoX.nc.
All so the ESP32 is finished with the file but the webui still needs up to 30 seconds. While this time you can't use anything on the WebUI.

MitchBradley, 2023-05-02:
My guess is that the TCP connection to the android tablet is getting jammed, maybe because of interference from the spindle.  The rapid (50 ms) auto-updates are trying to push a lot of data through the connection.  Perhaps a lot of packets are getting dropped and TCP is having to do a lot of retries to get the data through.  Try increasing the interval to, say, 1000 ms, or switch to polled reporting.

MitchBradley, 2023-05-02:
This message in the startup log:
E (6270) wifi:Association refused temporarily, comeback time 1024 mSec
also suggests that the WiFi might be overloaded, or has bad signal quality, or some other degradation.

MitchBradley, 2023-05-02:
I ran SchluesselAutoX.nc 6 times on my setup with no lag.

BlueOrangeLive, 2023-05-02:
On the PC with LAN cable to the router, I can also use the file more than 6 times without LAG. When used with the tablet, this LAGs (Auto Report 3000 ms) occur after two three times.
Ok, it works with Poll 3 sec more than 10 times. OK.

Note: (Auto Report 3000ms embedded under Preferences) When the webui starts, you display under Commands [MSG: Info: Channel Auto Report Interval Set to 50 MS]. Set None and then again to Auto Report. Then the correct value is displayed.
Maybe the Auto Report value is not set properly.

MitchBradley, 2023-05-02:
The preferences code in WebUI is really clumsy.  It needs to be rewritten from the ground up.

BlueOrangeLive, 2023-05-02:
Thanks. Then I put Poll 3sec every time when I start the webui. For the time being, I get along with that. Thanks Mitch.

closed 2023-05-02 by BlueOrangeLive