bdring/FluidNC #835: Problem: parking of a house of one X axis

ser159, 2023-03-15:
### Controller Board

MKS DLC32

### Machine Description

cartesian kinematics, stepper motor driver stepstick S109 TB67S109.

### Input Circuits

```yaml
4pin KY-024 Linear Magnetic Hall Switches Speed Counting Sensor Module, home min on all X Y Z axes.
```

### Configuration file

```yaml
board: MKS DLC32 2.0
name: MKS DLC32 XYZ
meta: (12/2/2021) B. Dring for diruuu
stepping:
  engine: I2S_STREAM
  idle_ms: 250
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: i2so.0
  x:
    steps_per_mm: 400.000
    max_rate_mm_per_min: 2000.000
    acceleration_mm_per_sec2: 50.000
    max_travel_mm: 290.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 1000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.36
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm:1.000
      stepstick:
        step_pin:  i2so.1
        direction_pin: I2SO.2

  y:
    steps_per_mm: 400.000
    max_rate_mm_per_min: 2000.000
    acceleration_mm_per_sec2: 50.000
    max_travel_mm: 205.000
    soft_limits: false
    homing:
      cycle: 3
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 1000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.35
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm:1.000
      stepstick:
        step_pin: I2SO.5
        direction_pin: I2SO.6

  z:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 2000.000
    acceleration_mm_per_sec2: 50.000
    max_travel_mm: 45.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 500.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.34
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm:1.000
      stepstick:
        step_pin: I2SO.3
        direction_pin: I2SO.4

i2so:
  bck_pin: gpio.16
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.12
  mosi_pin: gpio.13
  sck_pin: gpio.14

sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.15

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

macros:
  startup_line0:
  startup_line1:
  macro0: $SD/Run=lasertest.gcode
  macro1: $SD/Run=home.gcode
  macro2:
  macro3:

coolant:
  flood_pin: NO_PIN
  mist_pin: NO_PIN
  delay_ms: 0

probe:
  pin: gpio.22:low:pu
  check_mode_start: true

start:
  must_home: false

PWM:
  pwm_hz: 5000
  output_pin: gpio.32
  direction_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 0
  speed_map: 0=0.000% 8000=100.000%
```

### Startup Messages

```text
$ ss
[MSG:INFO: FluidNC v3.6.8]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:MKS_DLC32_2_ABC.yaml]
[MSG:INFO: Machine MKS DLC32 XYZ]
[MSG:INFO: Board MKS DLC32 2.0]
[MSG:INFO: I2SO BCK:gpio.16 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.14 MOSI:gpio.13 MISO:gpio.12]
[MSG:INFO: SD Card cs_pin:gpio.15 detect:NO_PIN freq:0]
[MSG:INFO: Stepping:I2S_stream Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:250ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable I2SO.0]
[MSG:INFO: Axis X (0.000,290.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.1 Dir:I2SO.2 Disable:NO_PIN]
[MSG:INFO:  X Neg Limit gpio.36]
[MSG:INFO: Axis Y (-205.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.5 Dir:I2SO.6 Disable:NO_PIN]
[MSG:INFO:  Y Neg Limit gpio.35]
[MSG:INFO: Axis Z (-45.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.3 Dir:I2SO.4 Disable:NO_PIN]
[MSG:INFO:  Z Neg Limit gpio.34]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: PWM Spindle Ena:NO_PIN Out:gpio.32 Dir:NO_PIN Freq:5000Hz Period:8191]
[MSG:INFO: Using spindle PWM]
[MSG:INFO: Probe Pin: gpio.22:low:pu]
[MSG:INFO: Connecting to STA SSID:MikroTik-F741C0]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.0.135]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 8080]
ok
```

### User Interface Software

UI: github.com/MitchBradley/ESP3D-WEBUI@a90d79b / FW: fluidnc v3.6.8

### What happened?

When parking in the house separately YZ, works well. If you send X home, when the end cap is triggered for the first time without a second rollback, the axis becomes on hold and the machine does not react to anything. If all axes send
home: the first Z is good, the second problem X, reaches the first trigger of the end cap and the engines are on hold on all axles. Then only reset.
Works fine only on FW: fluidnc v3.4.4. On the above versions, which I put (not all checked) hangs in the parking lot in the X-axis house.

MitchBradley, 2023-03-15:
If the limit switches are on the negative end, I think you need `positive_direction: false` .

ser159, 2023-03-15:
> If the limit switches are on the negative end, I think you need `positive_direction: false` .

I have X min, Y Z max. The Z axis is mandatory, the spindle raises as much as possible up.

bdring, 2023-03-15:
Send ***$Message/Level=Debug*** to output debug messages.  Home each axes individually with commands like ***$HX*** from FluidTerm and paste the results here.

http://wiki.fluidnc.com/en/features/commands_and_settings#message_level

ser159, 2023-03-16:
> Send _**$Message/Level=Debug**_ to output debug messages. Home each axes individually with commands like _**$HX**_ from FluidTerm and paste the results here.
> 
> http://wiki.fluidnc.com/en/features/commands_and_settings#message_level

Here's the version where everything works fine
UI: 2.1b72 / FW: fluidnc v3.6.4-pre1

$Message/Level=Debug
[MSG:DBG: Homing Cycle Z]
[MSG:DBG: Homing nextPhase FastApproach]
[MSG:DBG: Starting from 10.000,-10.000,-10.000]
[MSG:DBG: Planned move to 10.000,-10.000,39.500 @ 500.000]
[MSG:DBG:  Z Neg Limit 1]
[MSG:DBG: Homing limited Z]
[MSG:DBG: Homing nextPhase Pulloff0]
[MSG:DBG: Starting from 10.000,-10.000,0.961]
[MSG:DBG: Planned move to 10.000,-10.000,-0.039 @ 100.000]
[MSG:DBG:  Z Neg Limit 0]
[MSG:DBG: CycleStop Pulloff0]
[MSG:DBG: Homing nextPhase SlowApproach]
[MSG:DBG: Starting from 10.000,-10.000,-0.039]
[MSG:DBG: Planned move to 10.000,-10.000,1.061 @ 100.000]
[MSG:DBG:  Z Neg Limit 1]
[MSG:DBG: Homing limited Z]
[MSG:DBG: Homing nextPhase Pulloff1]
[MSG:DBG: Starting from 10.000,-10.000,0.994]
[MSG:DBG: Planned move to 10.000,-10.000,-0.006 @ 100.000]
[MSG:DBG:  Z Neg Limit 0]
[MSG:DBG:  Z Neg Limit 1]
[MSG:DBG:  Z Neg Limit 0]
[MSG:DBG:  Z Neg Limit 1]
[MSG:DBG:  Z Neg Limit 0]
[MSG:DBG:  Z Neg Limit 1]
[MSG:DBG:  Z Neg Limit 0]
[MSG:DBG: CycleStop Pulloff1]
[MSG:DBG: Homing nextPhase Pulloff2]
[MSG:DBG: mpos was 10.000,-10.000,-0.006]
[MSG:DBG: mpos becomes 10.000,-10.000,0.000]
[MSG:DBG: mpos transformed 10.000,-10.000,0.000]
[MSG:DBG: Homing done]
[MSG:DBG: Homing Cycle Y]
[MSG:DBG: Homing nextPhase FastApproach]
[MSG:DBG: Starting from 10.000,-10.000,0.000]
[MSG:DBG: Planned move to 10.000,215.500,0.000 @ 1000.000]
[MSG:DBG:  Y Neg Limit 1]
[MSG:DBG: Homing limited Y]
[MSG:DBG: Homing nextPhase Pulloff0]
[MSG:DBG: Starting from 10.000,1.015,0.000]
[MSG:DBG: Planned move to 10.000,0.015,0.000 @ 100.000]
[MSG:DBG:  Y Neg Limit 0]
[MSG:DBG: CycleStop Pulloff0]
[MSG:DBG: Homing nextPhase SlowApproach]
[MSG:DBG: Starting from 10.000,0.015,0.000]
[MSG:DBG: Planned move to 10.000,1.115,0.000 @ 100.000]
[MSG:DBG:  Y Neg Limit 1]
[MSG:DBG: Homing limited Y]
[MSG:DBG: Homing nextPhase Pulloff1]
[MSG:DBG: Starting from 10.000,1.003,0.000]
[MSG:DBG: Planned move to 10.000,0.003,0.000 @ 100.000]
[MSG:DBG:  Y Neg Limit 0]
[MSG:DBG:  Y Neg Limit 1]
[MSG:DBG:  Y Neg Limit 0]
[MSG:DBG: CycleStop Pulloff1]
[MSG:DBG: Homing nextPhase Pulloff2]
[MSG:DBG: mpos was 10.000,0.002,0.000]
[MSG:DBG: mpos becomes 10.000,0.000,0.000]
[MSG:DBG: mpos transformed 10.000,0.000,0.000]
[MSG:DBG: Homing done]
[MSG:DBG: Homing Cycle X]
[MSG:DBG: Homing nextPhase FastApproach]
[... 98 lines cut ...]
[MSG:DBG: Homing limited Y]
[MSG:DBG: Homing nextPhase Pulloff1]
[MSG:DBG: Starting from 0.000,10.993,0.000]
[MSG:DBG: Planned move to 0.000,9.993,0.000 @ 100.000]
[MSG:DBG:  Y Neg Limit 0]
[MSG:DBG: CycleStop Pulloff1]
[MSG:DBG: Homing nextPhase Pulloff2]
[MSG:DBG: mpos was 0.000,9.993,0.000]
[MSG:DBG: mpos becomes 0.000,0.000,0.000]
[MSG:DBG: mpos transformed 0.000,0.000,0.000]
[MSG:DBG: Homing done]

Here the problems begin, the interface freezes. Took screenshots. By the way, such problems have appeared with other axes sometimes.

[image]

[image]

[image]

After clicking reset the firmware

[image]

Problems with the switch at home?

bdring, 2023-03-16:
Use FluidTerm for testing this problem. See my last post.

MitchBradley, 2023-03-16:
Reverting to previous working versions will not help up debug the problem.  We made a lot of changes in limit switches and homing.  We need to debug the current version.

MitchBradley, 2023-03-16:
It appears that the limit switches are bouncing around, with a lot of transitions from 0 to 1 and back.  If they are hall sensors they could be affected by stray magnetic fields.  Show us a photo.

ser159, 2023-03-16:
> Send _**$Message/Level=Debug**_ to output debug messages. Home each axes individually with commands like _**$HX**_ from FluidTerm and paste the results here.
> 
> http://wiki.fluidnc.com/en/features/commands_and_settings#message_level

launched from under FluidTerm

FluidNC v1.2.0 using COM3
Exit: Ctrl-C, Ctrl-Q or Ctrl-], Clear screen: CTRL-W
Upload: Ctrl-U, Reset ESP32: Ctrl-R, Send Override: Ctrl-O
 $Message/Level=Debug
ok
$HZ
[MSG:DBG: Homing Cycle Z]
[MSG:DBG: Homing nextPhase FastApproach]
[MSG:DBG: Starting from 0.000,0.000,0.000]
[MSG:DBG: Planned move to 0.000,0.000,49.500 @ 500.000]
[MSG:DBG:  Z Neg Limit 1]
[MSG:DBG: Homing limited Z]
[MSG:DBG:  Z Neg Limit 0]
[MSG:DBG:  Z Neg Limit 1]
[MSG:DBG:  Z Neg Limit 0]
[MSG:DBG:  Z Neg Limit 1]
[MSG:DBG:  Z Neg Limit 0]
[MSG:DBG:  Z Neg Limit 1]
[MSG:DBG:  Z Neg Limit 0]
[MSG:DBG:  Z Neg Limit 1]
[MSG:DBG:  Z Neg Limit 0]
[MSG:DBG:  Z Neg Limit 1]
[MSG:DBG:  Z Neg Limit 0]
[MSG:DBG:  Z Neg Limit 1]
[MSG:DBG:  Z Neg Limit 0]
[MSG:DBG:  Z Neg Limit 1]
[MSG:DBG: Homing nextPhase Pulloff0]
[MSG:DBG: Starting from 0.000,0.000,9.994]
[MSG:DBG: Planned move to 0.000,0.000,8.994 @ 100.000]
[MSG:DBG:  Z Neg Limit 0]
[MSG:DBG:  Z Neg Limit 1]
[MSG:DBG:  Z Neg Limit 0]
[MSG:DBG:  Z Neg Limit 1]
[MSG:DBG:  Z Neg Limit 0]
[MSG:DBG: CycleStop Pulloff0]
[MSG:DBG: Homing nextPhase SlowApproach]
[MSG:DBG: Starting from 0.000,0.000,8.994]
[MSG:DBG: Planned move to 0.000,0.000,10.094 @ 100.000]
[MSG:DBG:  Z Neg Limit 1]
[MSG:DBG: Homing limited Z]
[MSG:DBG:  Z Neg Limit 0]
[MSG:DBG: Homing nextPhase Pulloff1]
[MSG:DBG: Starting from 0.000,0.000,9.990]
[MSG:DBG: Planned move to 0.000,0.000,8.990 @ 100.000]
[MSG:DBG:  Z Neg Limit 1]
[MSG:DBG:  Z Neg Limit 0]
[MSG:DBG: CycleStop Pulloff1]
[MSG:DBG: Homing nextPhase Pulloff2]
[MSG:DBG: mpos was 0.000,0.000,8.990]
[MSG:DBG: mpos becomes 0.000,0.000,0.000]
[MSG:DBG: mpos transformed 0.000,0.000,0.000]
[MSG:DBG: Homing done]
ok
$HY
[MSG:DBG: Homing Cycle Y]
[MSG:DBG: Homing nextPhase FastApproach]
[MSG:DBG: Starting from 0.000,0.000,0.000]
[MSG:DBG: Planned move to 0.000,225.500,0.000 @ 1000.000]
[MSG:DBG:  Y Neg Limit 1]
[MSG:DBG: Homing limited Y]
[MSG:DBG: Homing nextPhase Pulloff0]
[MSG:DBG: Starting from 0.000,1.090,0.000]
[MSG:DBG: Planned move to 0.000,0.090,0.000 @ 100.000]
[MSG:DBG:  Y Neg Limit 0]
[... 949 lines cut ...]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.5 Dir:I2SO.6 Disable:NO_PIN]
[MSG:INFO:  Y Neg Limit gpio.35]
[MSG:DBG: gpio conf 0 35]
[MSG:DBG:  Y Neg Limit 0]
[MSG:INFO: Axis Z (-45.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.3 Dir:I2SO.4 Disable:NO_PIN]
[MSG:INFO:  Z Neg Limit gpio.34]
[MSG:DBG: gpio conf 0 34]
[MSG:DBG:  Z Neg Limit 0]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: PWM Spindle Ena:NO_PIN Out:gpio.32 Dir:NO_PIN Freq:5000Hz Period:8191]
[MSG:INFO: Using spindle PWM]
[MSG:DBG: gpio conf 1 22]
[MSG:INFO: Probe Pin: gpio.22:low:pu]
[MSG:INFO: Connecting to STA SSID:MikroTik-F741C0]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.0.135]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 8080]

bdring, 2023-03-16:
BBBBBBBBBad X switch?

ser159, 2023-03-16:
Yes X axis.sometimes there are the same problems with the z axis

ser159, 2023-03-16:
Tomorrow I will try to reduce the sensitivity of the hall sensor and report the results.

MitchBradley, 2023-03-16:
Those hall effect sensors are very sensitive to noise.  When the magnet approaches the sensor slowly, there is a time where the magnetic field is just barely enough to make the device begin to switch on.  At that place, any stray magnetic fields from outside sources like motors might be enough to push it over and under the switching threshold, so the output cycles on and off.

One possible solution is to add hysteresis to the circuit.  The integrated circuit on those KY-024 boards is usually an LM393.  You can add hysteresis by putting a resistor between pins 1 and 3 of that IC.  I would start with 100K ohms, but it might need to be larger, say 220K.

ser159, 2023-03-17:
Thank you for your time

sjonholle, 2023-03-18:
I have also a chinese spindle. it is not brushless (they say it is). I had to put the control-unit more than one metre away from my esp32 and limit my rpm's to 8300 or else i get lots of errors. All wires are shielded and my limitswitches have optocouplers and that didn't help either. it's chinese.......

ser159, 2023-03-18:
> I have also a chinese spindle. it is not brushless (they say it is). I had to put the control-unit more than one metre away from my esp32 and limit my rpm's to 8300 or else i get lots of errors. All [...]

Yes, I understand that. But older firmware 3.4.4 and 3.6.4, work fine on this hardware.

MitchBradley, 2023-03-18:
Older versions of firmware used a slow form of polling to test limit switches.  Current versions use a very fast polling method, which can respond to changes very rapidly, so it can "see" noisy switches that older versions miss.  The disadvantage of the slow polling is slow response to switch changes so homing accuracy is not as good and limit violations are not detected as rapidly.

If you choose to stick with older firmware versions, please do not file any future problem reports when you run into issues with them.  We have fixed a lot of bugs since 3.6.4 and cannot support non-current versions.

The many repetitions of
```
[MSG:DBG: X Neg Limit 0]
[MSG:DBG: X Neg Limit 1]
```
is a clear indication that the limit switch is "sitting right at the threshold".  The old slow code probably missed the initial transition and did not stop the machine quickly, so the machine kept moving for a few steps and the switch sensing code did not notice the noisy area.  The real solution is to fix the noise, preferably by adding hysteresis to the hall effect sensors.  A fallback approach is to use mechanical limit switches.

closed 2023-03-18 by MitchBradley