bdring/FluidNC #836: Rc servo as Z axis

dmagyar, 2023-03-16:
### Controller Board

Simple ESP32 Devkit v4 wired to an SG90 servo 5V, GPIO25, GND. An SD card socket is connected to gpio.23/21/22/19 (not used for this test).

### Machine Description

This is a test setup with a dev board and a single servo.

### Configuration file

```yaml
name: "ESP32 Dev Controller V4"
board: "ESP32 Dev Controller V4"

stepping:
  engine: RMT
  idle_ms: 250
  dir_delay_us: 1
  pulse_us: 2
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: gpio.18
  
  x:
    steps_per_mm: 800
    max_rate_mm_per_min: 2000
    acceleration_mm_per_sec2: 25
    max_travel_mm: 1000
    homing:
      cycle: 2
      mpos_mm: 10
      positive_direction: false
    
    motor0:
      stepstick:
        direction_pin: gpio.16
        step_pin: gpio.2
    motor1:
      null_motor:

  y:
    steps_per_mm: 800
    max_rate_mm_per_min: 2000
    acceleration_mm_per_sec2: 25
    max_travel_mm: 1000
    homing:
      cycle: 2
      mpos_mm: 10
      positive_direction: false

    motor0:
      stepstick:
        direction_pin: gpio.17
        step_pin: gpio.0
    motor1:
      null_motor:

  z:
    steps_per_mm: 100
    max_rate_mm_per_min: 5000
    acceleration_mm_per_sec2: 100
    max_travel_mm: 5
    homing:
      cycle: 1
      mpos_mm: 0
      positive_direction: true

    motor0:
      rc_servo:
        pwm_hz: 50
        output_pin: gpio.25
        min_pulse_us: 1000
        max_pulse_us: 2000

    motor1:
      null_motor:

spi:
  miso_pin: gpio.23
  mosi_pin: gpio.21
  sck_pin: gpio.22

sdcard:
  cs_pin: gpio.19
  card_detect_pin: NO_PIN
       
probe:
  pin: gpio.32:low:pu
```

### Startup Messages

```text
[MSG:INFO: FluidNC v3.6.8]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine ESP32 Dev Controller V4]
[MSG:INFO: Board ESP32 Dev Controller V4]
[MSG:INFO: SPI SCK:gpio.22 MOSI:gpio.21 MISO:gpio.23]
[MSG:INFO: SD Card cs_pin:gpio.19 detect:NO_PIN freq:0]
[MSG:INFO: Stepping:RMT Pulse:2us Dsbl Delay:0us Dir Delay:1us Idle Delay:250ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable gpio.18]
[MSG:INFO: Axis X (10.000,1010.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.2 Dir:gpio.16 Disable:NO_PIN]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Y (10.000,1010.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.0 Dir:gpio.17 Disable:NO_PIN]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Z (-5.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     rc_servo Pin:gpio.25 Pulse Len(1000,2000 period:1048575)]
[MSG:INFO:   Motor1]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: Probe Pin: gpio.32:low:pu]
[MSG:INFO: Connecting to STA SSID:POISONSHQ]
[MSG:INFO: Connecting.]
[MSG:INF[MSG:INFO: FluidNC v3.6.8]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine ESP32 Dev Controller V4]
[MSG:INFO: Board ESP32 Dev Controller V4]
[MSG:INFO: SPI SCK:gpio.22 MOSI:gpio.21 MISO:gpio.23]
[MSG:INFO: SD Card cs_pin:gpio.19 detect:NO_PIN freq:0]
[MSG:INFO: Stepping:RMT Pulse:2us Dsbl Delay:0us Dir Delay:1us Idle Delay:250ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable gpio.18]
[MSG:INFO: Axis X (10.000,1010.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.2 Dir:gpio.16 Disable:NO_PIN]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Y (10.000,1010.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.0 Dir:gpio.17 Disable:NO_PIN]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Z (-5.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     rc_servo Pin:gpio.25 Pulse Len(1000,2000 period:1048575)]
[MSG:INFO:   Motor1]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: Probe Pin: gpio.32:low:pu]
[MSG:INFO: Connecting to STA SSID:POISONSHQ]
[MSG:INFO: Connecting.]
[MSG:INF[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
```

### User Interface Software

WebUI

### What happened?

Servo reacts to the first move command but FluidNC never returns with an "ok". Any subsequent commands result in an "Error" until I restart FluidNC.

```
$G
$Report/Interval=50
[MSG:INFO: Channel auto report interval set to 50 ms]
ok
<Alarm|MPos:0.000,0.000,0.000|FS:0,0|WCO:0.000,0.000,-1.000>
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
$X
[MSG:INFO: Caution: Unlocked]
ok
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Ov:100,100,100>
G53 G0 Z5
<<servo moves, but no response from FluidNC>>
G53 G0 Z0
Error
```

bdring, 2023-03-16:
It works for me with your config file.

 - The FluidTerm console only echos the characters you send in [advanced editing mode](http://wiki.fluidnc.com/en/fluidterm/fluidterm_usage#character-echo). Clicking the TAB key is an easy way to enter that mode. 
 - The error does not look right. All error responses should be lower case and give an error number. Did you edit that out? 
 
Here is an example of sending a bad gcode command
```
g51 Z5
error:20
$e=20
20: Unsupported GCode command
ok
```

You have some weird formatting in the startup messages that could indicate a bad serial connection. See the second line. Can you send a photo of your controller?

```
[MSG:INFO: Connecting.]
[MSG:INF[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
```

Here is my session.

```
[MSG:INFO: FluidNC v3.6.8]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:issue_836.yaml]
[MSG:INFO: Machine ESP32 Dev Controller V4]
[MSG:INFO: Board ESP32 Dev Controller V4]
[MSG:INFO: SPI SCK:gpio.22 MOSI:gpio.21 MISO:gpio.23]
[MSG:INFO: SD Card cs_pin:gpio.19 detect:NO_PIN freq:0]
[MSG:INFO: Stepping:RMT Pulse:2us Dsbl Delay:0us Dir Delay:1us Idle Delay:250ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable gpio.18]
[MSG:INFO: Axis X (10.000,1010.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.2 Dir:gpio.16 Disable:NO_PIN]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Y (10.000,1010.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.0 Dir:gpio.17 Disable:NO_PIN]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Z (-5.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     rc_servo Pin:gpio.25 Pulse Len(1000,2000 period:1048575)]
[MSG:INFO:   Motor1]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: Probe Pin: gpio.32:low:pu]
[MSG:INFO: Connecting to STA SSID:Barts-WLAN]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connecting...]
[MSG:INFO: Connected - IP is 192.168.1.47]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.6 [FluidNC v3.6.8 (wifi) '$' for help]
[MSG:INFO: '$H'|'$X' to unlock]
$G
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
ok
$Report/Interval=50
[MSG:INFO: Channel auto report interval set to 50 ms]
ok
<Alarm|WPos:0.000,0.000,0.000|Bf:15,128|FS:0,0|WCO:0.000,0.000,0.000>
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
$X
[MSG:INFO: Caution: Unlocked]
ok
<Idle|WPos:0.000,0.000,0.000|Bf:15,128|FS:0,0|Ov:100,100,100>
G53 G0 Z5
ok
<Run|WPos:0.000,0.000,0.000|Bf:14,128|FS:720,0>
<Run|WPos:0.000,0.000,0.120|Bf:14,128|FS:960,0>
<Run|WPos:0.000,0.000,0.490|Bf:14,128|FS:1260,0>
<Run|WPos:0.000,0.000,1.110|Bf:14,128|FS:1123,0>
<Run|WPos:0.000,0.000,1.990|Bf:14,128|FS:823,0>
<Run|WPos:0.000,0.000,3.040|Bf:14,128|FS:523,0>
<Run|WPos:0.000,0.000,3.900|Bf:14,128|FS:223,0>
<Run|WPos:0.000,0.000,4.520|Bf:15,128|FS:0,0>
<Run|WPos:0.000,0.000,4.880|Bf:15,128|FS:0,0|WCO:0.000,0.000,0.000>
<Idle|WPos:0.000,0.000,5.000|Bf:15,128|FS:0,0|Ov:100,100,100>
G53G0Z0
ok
<Run|WPos:0.000,0.000,5.000|Bf:14,128|FS:720,0>
<Run|WPos:0.000,0.000,4.880|Bf:14,128|FS:960,0>
<Run|WPos:0.000,0.000,4.500|Bf:14,128|FS:1260,0>
<Run|WPos:0.000,0.000,3.880|Bf:14,128|FS:1123,0>
<Run|WPos:0.000,0.000,3.000|Bf:14,128|FS:823,0>
<Run|WPos:0.000,0.000,1.950|Bf:14,128|FS:523,0>
<Run|WPos:0.000,0.000,1.090|Bf:14,128|FS:223,0>
<Run|WPos:0.000,0.000,0.480|Bf:15,128|FS:0,0>
<Run|WPos:0.000,0.000,0.110|Bf:15,128|FS:0,0|WCO:0.000,0.000,0.000>
<Idle|WPos:0.000,0.000,0.000|Bf:15,128|FS:0,0|Ov:100,100,100>
```

dmagyar, 2023-03-16:
The logs were copied from the WebUI - now I tried with FluidTerm and it seems when the servo moves the MCU disconnects... 
The controller is connected to my PC right now via USB - I'll try to add a cap on the 5v line and try again.

```
ok
$X
[MSG:INFO: Caution: Unlocked]
ok
G53 G0 Z5
ok
Serial port disconnected - waiting for reconnect
Type Ctrl-] to quit
Serial port reconnected
Resetting MCU
```

bdring, 2023-03-16:
Do not power the servo from the USB. It will drop the voltage and reset the ESP32