bdring/FluidNC #841: Motor Twitches

Oleg-Perevyshin, 2023-03-22:
### Controller Board

MKS TinyBee v1.0
Standard connection

### Machine Description

Driver - TLC2209
Motor - Nema17
Standard connection

### Input Circuits

```yaml
Standard connection
```

### Configuration file

```yaml
board: MKS TinyBee V1.0
name: OpenPnP
meta: By POE

kinematics:
  corexy:

stepping:
  engine: I2S_STATIC
  idle_ms: 0
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  x:
    # X
    steps_per_mm: 80.00
    max_rate_mm_per_min: 25000.000
    acceleration_mm_per_sec2: 2500.000
    max_travel_mm: 400.000
    soft_limits: false
    homing:
      cycle: 0
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 500.000
      seek_mm_per_min: 5000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.33
      hard_limits: false
      pulloff_mm: 2.000
      stepstick:
        step_pin: I2SO.1
        direction_pin: I2SO.2
        disable_pin: I2SO.0

  y:
    # Y
    steps_per_mm: 80.00
    max_rate_mm_per_min: 25000.000
    acceleration_mm_per_sec2: 2500.000
    max_travel_mm: 400.000
    soft_limits: false
    homing:
      cycle: 1
[... 271 lines cut ...]
[MSG:INFO: macro0_pin gpio.4:low:pu]
[MSG:INFO: Kinematic system: CoreXY]
[MSG:INFO: Laser Ena:I2SO.16 Out:gpio.13 Freq:5000Hz Period:8191]
[MSG:INFO: Using spindle Laser]
[MSG:INFO: Connecting to STA SSID:POE-Net]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.1.15]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://OpenPnP.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
ok
```

### User Interface Software

WebUI

### What happened?

I run the axis move command and the motor jerks. First it spins, then it goes back a little, then it spins again.
This was not the case with other firmware.
[Link video](https://www.youtube.com/shorts/mR8sd86xvEg)

MitchBradley, 2023-03-22:
1. You should probably fix those indentation errors in your yaml file first.
2. I have never heard of TLC2209.  Do you mean TMC2209?
3. What exactly is "other firmware"?
4. If the drivers are TMC2209, are they set up for UART configuration or jumper-based configuration?

MitchBradley, 2023-03-22:
5. The startup messages show "FluidNC v3.0.x (noGit)" which suggests that you compiled yourself instead of installing from a precompiled release package.  We only support precompiled versions because otherwise we do not know exactly what source you are working and what else you might have done.

Oleg-Perevyshin, 2023-03-22:
2. Yes, the TMC2209 was wrong.
3. I tried marlin.
4. TMC2209 without UART. I only use Step, Dir, Enable and divide by 16
5. Now I'll do it right.

bdring, 2023-03-22:
The indent problems start at the a axis. I think there is an extra space before it and the following axes.

Oleg-Perevyshin, 2023-03-22:
<details>
  <summary>Already other errors</summary>

```cpp
$ss
[MSG:INFO: FluidNC v3.6.8]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:WARN: Ignored key axes]
[MSG:ERR: Configuration parse error on line 34: Invalid character]
[MSG:INFO: Using default configuration]
[MSG:INFO: Axes: using defaults]
[MSG:ERR: Initialization error at /stepping: I2SO bus must be configured for this stepping type]
[MSG:INFO: Configuration is invalid]
[MSG:INFO: Connecting to STA SSID:POE-Net]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.1.15]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://OpenPnP.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
ok
```

</details>

<details>
  <summary>Config</summary>

```cpp
board: MKS TinyBee
name: OpenPnP
meta: By POE Config 22.03.2023

arc_tolerance_mm: 0.002
junction_deviation_mm: 0.010
verbose_errors: false
report_inches: false
enable_parking_override_control: false
use_line_numbers: false
planner_blocks: 16

start:
  must_home: false
  deactivate_parking: false
  check_limits: true

kinematics:
  corexy:

stepping:
  engine: I2S_STATIC
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0
  segments: 12

axes:
  shared_stepper_disable_pin: NO_PIN
  shared_stepper_reset_pin: NO_PIN

axes:
  [x:|y:|z:|a:|b:|c:]
  x:
    steps_per_mm: 80.000
    max_rate_mm_per_min: 25000.000
[... 190 lines cut ...]
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN

 # USB  TX0(1)  RX0(3)
 # J3-> TX2(17) RX2(16)
 # J1-> I2SO.23 I2SO.22
 # TH1(36)        TH2(34)        THB(39)
 # HE0(I2SO.17)   HE1(I2SO.18)   HBED(I2SO.16)
 # FAN1(I2SO.19)  FAN2(I2SO.20)
 #                   EXP1                                   EXP2
 #(BEEPER) I2SO.21 | 1  2 | 13 (BTN_ENC)    (SPI MISO) 19 | 1  2 | 18 (SPI SCK)
 #    (LCD_EN)  21 | 3  4 |  4 (LCD_RS)      (BTN_EN1) 14 | 3  4 |  5 (SPI CS)
 #    (LCD_D4)   0 | 5  6   16 (LCD_D5)      (BTN_EN2) 12 | 5  6   23 (SPI MOSI)
 #    (LCD_D6)  15 | 7  8 | 17 (LCD_D7)      (SPI_DET) 34 | 7  8 | RESET
 #             GND | 9 10 | 5V                        GND | 9 10 | 3.3V
```

</details>

I don't understand anything anymore.

What am I doing wrong?

MitchBradley, 2023-03-22:
- In your most recent config file there are two axes: sections.  There should only be one
- There is an invalid line `  [x:|y:|z:|a:|b:|c:]` that should not be present.  I'm not sure where you got that, but I can guess.  Some people use the notation `[p|q|s]` to mean "choose one of p or q or s".  Perhaps there is documentation somewhere that uses that notation without explaining it.
- You said "I run the axis move command" but you did not tell us exactly which command, so we do not know which axis or axes were affected, nor do we know which speed you were trying to move
- We need to know which UI (sender, terminal, or WebUI) you used to issue the command.  Different UIs use different commands for movement like jogs.
- The speeds and accelerations that you have configured are rather high.  It is best to start slow and test different speeds until you find what the motors can handle.  Twitching motors can be caused by too much acceleration or top speeds that are too high.

Oleg-Perevyshin, 2023-03-23:
I got rid of the errors.
I did it like the wiki.
But the motors are not spinning at all.
I manage through a standard web interface.

<details>
  <summary>config.yaml</summary>

```yaml
# http://wiki.fluidnc.com/en/config/top_level_config_items
name: OpenPnP
board: MKS TinyBee
meta: 23.03.2023 by POE

# http://wiki.fluidnc.com/en/config/config_IO
i2so:
  bck_pin: gpio.25
  ws_pin: gpio.26
  data_pin: gpio.27

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

# http://wiki.fluidnc.com/en/config/sd_card
sdcard:
  cs_pin: gpio.5
  card_detect_pin: gpio.34

# http://wiki.fluidnc.com/en/config/axes
stepping:
  engine: I2S_STREAM
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  shared_stepper_reset_pin: NO_PIN
  x:
    steps_per_mm: 80.000
    max_rate_mm_per_min: 10000.000
    acceleration_mm_per_sec2: 1500.000
    max_travel_mm: 400.000
    soft_limits: false
    homing:
      cycle: 0
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 5000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.33
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 1.000
      stepstick:
        disable_pin: I2SO.0
        step_pin: I2SO.1
        direction_pin: I2SO.2

  y:
    steps_per_mm: 80.000
[... 220 lines cut ...]
[MSG:INFO: Connecting to STA SSID:POE-Net]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.1.15]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://OpenPnP.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
ok
```

</details>

I click on the arrows in the web interface and see in response:
```
$J=G91 G21 F2500 X10
ok
<Idle|MPos:0.000,0.000,0.000,0.000,0.000|FS:0,0|Pn:XYZ|WCO:0.000,0.000,0.000,0.000,0.000>
$J=G91 G21 F2500 Y10
ok
<Idle|MPos:0.000,0.000,0.000,0.000,0.000|FS:0,0|Pn:XYZ|Ov:100,100,100>
```

What could be the problem?

Oleg-Perevyshin, 2023-03-23:
Got it!
Limit switches not connected correctly.
My mistake.

Tell me how to read the analog value from the pins to which the temperature sensor is usually connected with 3D printers (GPIO 36, 39)?

MitchBradley, 2023-03-23:
Search the FluidNC wiki for "analog".  In general you should read first, ask later.

Oleg-Perevyshin, 2023-03-23:
I searched but found only output settings (M67 Analog Output)

MitchBradley, 2023-03-23:
What do you intend to do with the analog input values?

Oleg-Perevyshin, 2023-03-24:
I'm building a PCB assembly machine based on OpenPnP.
Analog vacuum sensors are connected there for control. This is an important aspect, because these values determine whether the component is captured or not.

MitchBradley, 2023-03-24:
How does OpenPnP receive the value from analog sensors?  GCode has no standard way to respond to analog values.

Oleg-Perevyshin, 2023-03-24:
A command is written in the program, for example M150. The answer is then run through a regular expression and the result is a value.

MitchBradley, 2023-03-24:
That is not standard GCode.  We try to follow the NIST standard.

Oleg-Perevyshin, 2023-03-24:
Yes, this is not a standard code, just a custom M command added.
So I can't get the vacuum value?

MitchBradley, 2023-03-24:
Not unless someone adds such a custom command.

closed 2023-03-24 by Oleg-Perevyshin

bdring, 2023-03-24:
Is that gcode command synchronous. Does it wait for the motion buffer to clear and stop?