bdring/FluidNC #846: Problem: Version 3.6.8 is not working for me

Aljo003, 2023-03-25:
### Controller Board

ESP32 WROOM 32 with TB6560 motor drivers.

### Machine Description

Laser engraver with 2 motors for Y axis and one for X axis driven with TB6560. 

### Input Circuits

```yaml
https://imgur.com/a/s3vCA2U
```

### Configuration file

```yaml
name: "ESP32 Dev Controller V4"
board: "ESP32 Dev Controller V4"

stepping:
  engine: RMT
  idle_ms: 255
  dir_delay_us: 0
  pulse_us: 4
  disable_delay_us: 0

axes:
  x:
    steps_per_mm: 800
    max_rate_mm_per_min: 2000
    acceleration_mm_per_sec2: 25
    max_travel_mm: 3500
    soft_limits: false
    homing:
      cycle: 1
      allow_single_axis: true
      positive_direction: false
      mpos_mm: 10.00
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100
    motor0: #X1
      limit_neg_pin: gpio.17:high
      limit_pos_pin: gpio.5:high
      hard_limits: true
      pulloff_mm: 0.500
      standard_stepper:
        step_pin: gpio.32
        direction_pin: gpio.33
        
  y:
    steps_per_mm: 800
    max_rate_mm_per_min: 2000
    acceleration_mm_per_sec2: 25
    max_travel_mm: 3500
    soft_limits: false
    homing:
      cycle: 2
      allow_single_axis: true
      positive_direction: false
      mpos_mm: 10.00
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100
    motor0: #Y2
      limit_neg_pin: gpio.4:high
      limit_pos_pin: gpio.16:high
      hard_limits: true
      pulloff_mm: 0.500
      standard_stepper:
        step_pin: gpio.25
        direction_pin: gpio.26
    motor1: #Y1
      limit_neg_pin: gpio.15:high
      limit_pos_pin: gpio.2:high
      hard_limits: true
      pulloff_mm: 0.500
      standard_stepper:
        step_pin: gpio.27
        direction_pin: gpio.14:low        

spi:
  miso_pin: gpio.23
  mosi_pin: gpio.22
  sck_pin: gpio.21

sdcard:
  cs_pin: gpio.19
  card_detect_pin: NO_PIN

Laser:
  pwm_hz: 5000
  output_pin: gpio.12
  enable_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  tool_num: 0
  speed_map: 0=0.000% 255=100.000%
```

### Startup Messages

```text
[MSG:INFO: FluidNC v3.6.8]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine ESP32 Dev Controller V4]
[MSG:INFO: Board ESP32 Dev Controller V4]
[MSG:INFO: SPI SCK:gpio.21 MOSI:gpio.22 MISO:gpio.23]
[MSG:INFO: SD Card cs_pin:gpio.19 detect:NO_PIN freq:0]
[MSG:INFO: Stepping:RMT Pulse:4us Dsbl Delay:0us Dir Delay:0us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (10.000,3510.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.32 Dir:gpio.33 Disable:NO_PIN]
[MSG:INFO:  X Neg Limit gpio.17]
[MSG:INFO:  X Pos Limit gpio.5]
[MSG:INFO: Axis Y (10.000,3510.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.25 Dir:gpio.26 Disable:NO_PIN]
[MSG:INFO:  Y Neg Limit gpio.4]
[MSG:INFO:  Y Pos Limit gpio.16]
[MSG:INFO:   Motor1]
[MSG:INFO:     standard_stepper Step:gpio.27 Dir:gpio.14:low Disable:NO_PIN]
[MSG:INFO:  Y2 Neg Limit gpio.15]
[MSG:INFO:  Y2 Pos Limit gpio.2]
[MSG:INFO: Axis Z (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Laser Ena:NO_PIN Out:gpio.12 Freq:5000Hz Period:8191]
[MSG:INFO: Using spindle Laser]
[MSG:INFO: Connecting to STA SSID:TP-LINK_8452]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.0.107]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
```

### User Interface Software

FluidTerm, WebUI

### What happened?

Nothing seems to work, limit swith LED is not even turning on. When using version 3.3.1 everything is working fine. No errors given when starting machine

Skorpi08, 2023-03-25:
you can add Pics by drag&drop
[image]

MitchBradley, 2023-03-25:
Your limit switches do not have any external pullups, and the internal pullups are not enabled due to no ":pu" attributes on the pins.

Aljo003, 2023-03-25:
Should I keep :high attributes or only add :pu ?

MitchBradley, 2023-03-25:
Either is okay.  :high is the default, so :pu  and :high:pu are the same

Aljo003, 2023-03-25:
Now limits switch LED is working and it's reporting in $limits but it's not triggering alarm when pressed, homing is not working too.

MitchBradley, 2023-03-25:
http://wiki.fluidnc.com/en/support/help_with_switch_problems

See especially the "Debug Mode" section

Aljo003, 2023-03-25:
Oh wait i also have wrong config file, i'm sending new one with modified switches

Aljo003, 2023-03-25:
[image]
When it hits limit switch when homing

MitchBradley, 2023-03-25:
What kind of switches?

Aljo003, 2023-03-25:
[image]
[image]

MitchBradley, 2023-03-25:
When it hits the switch, try pushing it in manually then releasing it to see what happens

Aljo003, 2023-03-25:
It's still in Homing mode but motor stop and when I hold switch same thing is spamming

Aljo003, 2023-03-25:
And also Laser is not working, SD Card is not detected

MitchBradley, 2023-03-25:
Show a photograph of your system, not a schematic.

Aljo003, 2023-03-25:
[image]
[image]

MitchBradley, 2023-03-25:
That particular SD Card adapter is known to be slow. There are some old issues complaining about it.  Try adding
```
   frequency_hz: 5000000
```
in the sdcard section of the config files.

It used to work in the old version of FluidNC because that version used an sdcard driver that always downgraded the speed.

Aljo003, 2023-03-25:
SD Card working with adding frequency_hz

MitchBradley, 2023-03-25:
For your limit switches, it is really better to have signal conditioning near the ESP32 module.  Depending on the internal pullups in a system like your, where there are long wires connected directly to ESP32 GPIOs, is a bad idea.  The signals will always be noisy. At a minimum, you should have 5K external pullups to 3.3V, with 0.1uF capacitors to ground, as close as possible to the ESP32.

Aljo003, 2023-03-25:
So 5K pullup and capacitor for each switch?

MitchBradley, 2023-03-25:
http://wiki.fluidnc.com/en/support/controller_design_guidelines#rc-filtering

MitchBradley, 2023-03-26:
Re the laser not working, your schematic shows the laser being driven by GPIO 13, which the config file has Laser: / output_pin: gpio.12

Aljo003, 2023-03-26:
looks like i changed it one time because now i have it connected to gpio 12 and i i didn't update schematics when i changed it

Aljo003, 2023-03-26:
i will make one board with resistors and capacitors for switches but i don't know if this will fix issue with homing problem if it's not even triggering hard limit alarm when i press switch.

MitchBradley, 2023-03-26:
What commands are you using to test the laser?  It will not turn on until the system is in feedrate motion state.  If you send `m3 s255` and the last movement command was G0, the laser will be off until you send G1, G2, or G3.

Aljo003, 2023-03-26:
Laser is working when i switch to G1 or other commands and then turn it on with m3. Now only limit switches are not working correctly

Aljo003, 2023-03-26:
Okay, homing for Y axis is working, Y axis is axis which have 2 motors. So there is problem with limit switches on X axis.

Aljo003, 2023-03-26:
I think i found issue with limit switches. Limit switch S5 as on schematics is not working correctly, i switched positive and negative switch and now it's working fine. I will order new switch and replace it. I will leave this open if I will have any problems with adding R/C filter. Thanks for your help and time Mitch.

Aljo003, 2023-03-26:
Negative limit switch for X axis is not working correctly, even if i switch pins, switch with other switch so there must be something wrong in code. All other switches are working fine

MitchBradley, 2023-03-26:
I don't understand that last statement.

Aljo003, 2023-03-26:
So y axis homing is working okay, but when Y axis is homing when switch is pressed this will start spamming and still in home state. 
[image]
I even tried switching pins but doesn't seem to fix it.

MitchBradley, 2023-03-26:
Something could be wrong with the wiring to the x switch.

MitchBradley, 2023-03-26:
bad ground perhaps?

MitchBradley, 2023-03-26:
Did you put RC filters with external pullups on the switches?  Until you do that, the signal quality will be so bad that I cannot help you.

Aljo003, 2023-03-26:
> Did you put RC filters with external pullups on the switches? Until you do that, the signal quality will be so bad that I cannot help you.

I didn't yet because i need to get capacitors and resistors to build RC filters. I will make that and will see if it will work then. But it's strange that will older version it's working fine. Thanks for help

MitchBradley, 2023-03-26:
The older version uses a different method for reading inputs.  The old method was very slow and could miss fast transitions.

MitchBradley, 2023-03-26:
Software improvements can often reveal hidden hardware problems.

Aljo003, 2023-03-26:
Okay, I will make RC filters first, grounds on switches are okay.

Aljo003, 2023-03-26:
I now tried putting limit switch wires as far away as i could from motor wires and machine homing was succesfull. I will add RC filters and test it then. Thanks for your help

Aljo003, 2023-03-26:
Can I also fix interference with ferrite bead as if would be easier? Or are R/C filters still better then ferrite bead?

MitchBradley, 2023-03-26:
Rc is they common solution. You will just have to experiment. I cannot predict exactly what will happen in your setup. There are many aspects off your setup that are hobbyist grade not industrial grade. Despite the fact that were have EE training and experience, we are volunteer software developers, not paid EE consultants or teachers.

Aljo003, 2023-03-28:
Okay,thanks for your help, I managed to get capacitors and resistors and will make RC filters this week and try it, is it okay if I wire all capacitors to one wire and that wire to one ground and same with resistor and then i just wire that wire between each resistor and capacitor to limit switch pin?

MitchBradley, 2023-03-28:
I repeat, we are not EE consultants.

PatrikRindlisbacher, 2023-03-28:
You can find good instructions on the net for RC filters. 
Important short lines away from the entrance. --> less than 10mm are desired.

Aljo003, 2023-04-01:
I added RC filters and now its working fine, thanks for your help

closed 2023-04-01 by Aljo003