bdring/FluidNC #847: Problem:

Geekineer, 2023-03-27:
### Controller Board

6-pack ext driver version.  Loaded FluidNC WiFI and 6_Pext_XYYZ.yaml file. 

### Machine Description

Gantry Plasma cutter (someday) XYYZ.  Today no limit switches.  Future hard switches at home and soft limits at machine max travel. 
DM542 Drivers (4).  24V supply. 

### Configuration file

```yaml
File Bart send me: 6_Pext_XYYZ
```

### Startup Messages

```text
Here is listing for $SS and $CD
$SS
[MSG:INFO: FluidNC v3.6.8]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:WARN: Cannot open configuration file:6_Pack_XYYZ]
[MSG:INFO: Using default configuration]
[MSG:INFO: Axes: using defaults]
[MSG:INFO: Machine Default (Test Drive)]
[MSG:INFO: Board None]
[MSG:INFO: Stepping:RMT Pulse:4us Dsbl Delay:0us Dir Delay:0us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Axis Y (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Axis Z (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: AP SSID DKFluidNC IP 192.168.0.242 mask 255.255.255.0 channel 3]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
ok
$CD
board: None
name: Default (Test Drive)
meta: 
stepping: 
  engine: RMT
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 0
  disable_delay_us: 0
  segments: 12
spi: 
  miso_pin: NO_PIN
  mosi_pin: NO_PIN
  sck_pin: NO_PIN
sdcard: 
  cs_pin: NO_PIN
  card_detect_pin: NO_PIN
  frequency_hz: 0
kinematics: 
  Cartesian: 
axes: 
  shared_stepper_disable_pin: NO_PIN
  shared_stepper_reset_pin: NO_PIN
  x: 
    steps_per_mm: 80.000000
    max_rate_mm_per_min: 1000.000000
    acceleration_mm_per_sec2: 25.000000
    max_travel_mm: 1000.000000
    soft_limits: false
    homing: 
      cycle: 0
      allow_single_axis: true
      positive_direction: true
      mpos_mm: 0.000000
      feed_mm_per_min: 50.000000
      seek_mm_per_min: 200.000000
      settle_ms: 250
      seek_scaler: 1.100000
      feed_scaler: 1.100000
    motor0: 
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000000
      null_motor: 
  y: 
    steps_per_mm: 80.000000
    max_rate_mm_per_min: 1000.000000
    acceleration_mm_per_sec2: 25.000000
    max_travel_mm: 1000.000000
    soft_limits: false
    homing: 
      cycle: 0
      allow_single_axis: true
      positive_direction: true
      mpos_mm: 0.000000
      feed_mm_per_min: 50.000000
      seek_mm_per_min: 200.000000
      settle_ms: 250
      seek_scaler: 1.100000
      feed_scaler: 1.100000
    motor0: 
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000000
      null_motor: 
  z: 
    steps_per_mm: 80.000000
    max_rate_mm_per_min: 1000.000000
    acceleration_mm_per_sec2: 25.000000
    max_travel_mm: 1000.000000
    soft_limits: false
    homing: 
      cycle: 0
      allow_single_axis: true
      positive_direction: true
      mpos_mm: 0.000000
      feed_mm_per_min: 50.000000
      seek_mm_per_min: 200.000000
      settle_ms: 250
      seek_scaler: 1.100000
      feed_scaler: 1.100000
    motor0: 
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000000
      null_motor: 
control: 
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN
coolant: 
  flood_pin: NO_PIN
  mist_pin: NO_PIN
  delay_ms: 0
probe: 
  pin: NO_PIN
  check_mode_start: true
macros: 
  startup_line0: 
  startup_line1: 
  macro0: 
  macro1: 
  macro2: 
  macro3: 
start: 
  must_home: true
  deactivate_parking: false
  check_limits: false
parking: 
  enable: false
  axis: Z
  target_mpos_mm: -5.000000
  rate_mm_per_min: 800.000000
  pullout_distance_mm: 5.000000
  pullout_rate_mm_per_min: 250.000000
user_outputs: 
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN
NoSpindle: 
arc_tolerance_mm: 0.002000
junction_deviation_mm: 0.010000
verbose_errors: false
report_inches: false
enable_parking_override_control: false
use_line_numbers: false
planner_blocks: 16
ok
```

### User Interface Software

WEB UI

### What happened?

Concerned with only 24V connected to 6 pack that all green LEDs near driver output stay on all the time, every time at power up, with and without ESP32 module installed. And if installed, and FluidNC running via internet, same  LEDs stay on during an X axis JOG from the WEB UI.   No DM542 drivers were connected for this test

bdring, 2023-03-27:
That config is not right for your board.

Send me an email via Tindie. Use this survey to tell me how you want the machine configured and I will send you a new config.

https://github.com/bdring/6-Pack_CNC_Controller/wiki/Programming-Survey

Geekineer, 2023-03-27:
Will do ....tnx

Geekineer, 2023-03-27:
Thanks Bart - sent via Tindie message system and copied here as well.

   -

   What is your name? *Dennis Kramer*
   -

   What is your Tindie order number? *Order #370983  6 pack w/ external
   drivers*
   -

   Motors *4 Type 17, **used on **XYYZ, Driven by DM542 drivers. *
   -

      How many motors are you using and what axes are they used for. Answer
      examples (XYZ, XYYZ, XXYYZ, etc). Max of 6 letter axes and 2 motors per
      axis. *(4) One for X (2) for Y and (1) for Z*
      -

   Switches (only specify as many as you have inputs for) *4 limit/**homing**,
   and 1 probe*
   -

      Limit switches – *Hard Switches, (4) one for X, Y, Y, and Z, used for
      homing in the ‘toward min’ direction. Switches are normally closed, one
      side tied to ground. Soft limits used for positive **(max)** end of
      travel. Travel distance (mm) 300 X, 400 Y, and 50 Z. *
      -

         List all limit switches you are using. If there are 2 on an axis
         are they separate or wired to the same input ? Answer examples (X-min,
         X-max, Y-dual (wired together), Z-max). *4 **limit **switch wired
         to separate inputs.*
         -

         If an axis uses more than one switch, which direction does it
         home? *(**all **toward min* *)*
         -

      Are you using a Z Probe? If so, if it is more than just a touch plate
      send a link to it. *Simple touch plate*
      -

      List any other switches here (Reset, Feedhold, Run, etc)
      -

   Spindle *YES – used for ARC ON of plasma cutter. No analog control just
   on/off*
   -

      What Type of Spindle and/or Laser are you using? *Uses high logic
      level to drive a SS relay for ARC ON.*
      -

   Accessories
   -

      Are there any accessories you want to control? (Vacuum, Coolant, etc)
      *No*
      -

   Unused Modules *purchased a total of 2 input and one output module –
   used as required*
   -

      Did you purchase any modules that you do not plan to use right away?
      -

   Networking.
   -

      Do you want to use Bluetooth or Wifi? *WiFi, initially AP mode then
      STA mode once things are running*
      -

   Is there anything I left out? Did you purchase any module for special
   use not covered above.

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

bdring, 2023-03-28:
I'll get that to you first thing tomorrow.

bdring, 2023-03-28:
Here is a config file that should get you started. You will still need to adjust the values for resolution, speed, etc for your machine.

[See this for uploading instructions.](http://wiki.fluidnc.com/en/installation

Config file in zip...

[6_Pext_XYYZ_Relay.zip](https://github.com/bdring/FluidNC/files/11090200/6_Pext_XYYZ_Relay.zip)

Geekineer, 2023-04-07:
Hi Bart......Going to be ordering another 6 pack, this time int driver
module version.  Do u sell or make available an stl file case for this
version.
Thanks
Dennis
* 6_Pext version working great.

bdring, 2023-04-07:
If you order a 6 Pack with a programmed ESP32 and at least one module, I ship it tested and in a case.

Geekineer, 2023-04-07:
OK. ... good to know for the future.   For now I have extra modules and
ESP32 bds,  so looking for a short term solution

Dennis

bdring, 2023-04-07:
The 3D files are available on the 6 Pack Github repo. If you make a $25 donation to the FluidNC project, I can send one with your order.

http://wiki.fluidnc.com/en/home#donations

closed 2023-04-27 by bdring