bdring/FluidNC #850: Problem: TMC driver not detected - expected 0x21 got 0x0

floco, 2023-04-05:
### Controller Board

esp32 dev kit c v4

### Machine Description

Just a diy project to move rods on a linear axix

### Input Circuits

```yaml
DIR_PIN          25 // Direction
STEP_PIN         14 // Step
SERIAL_PORT Serial2 // HardwareSerial port pins 16 & 17
DRIVER_ADDRESS 0b00 // TMC2209 Driver address according to MS1 and MS2
```

### Configuration file

```yaml
board: TMC2209 Pen
name: TMC2209 Pen
stepping:
  engine: RMT
  idle_ms: 250
  pulse_us: 2
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 80.000
    max_rate_mm_per_min: 30000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      tmc_2209:
        uart:
          txd_pin: gpio.17
          rxd_pin: gpio.16
          rts_pin: NO_PIN
          cts_pin: NO_PIN
          baud: 115200
          mode: 8N1

        addr: 0
        r_sense_ohms: 0.110
        run_amps: 1.00
        hold_amps: 0.500
        microsteps: 4
        stallguard: 30
        stallguard_debug: true
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StallGuard
        use_enable: false
        step_pin: gpio.14
        direction_pin: gpio.25
        disable_pin: NO_PIN
```

### Startup Messages

```text
[MSG:INFO: FluidNC v3.6.7]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine TMC2209 Pen]
[MSG:INFO: Board TMC2209 Pen]
[MSG:INFO: Stepping:RMT Pulse:2us Dsbl Delay:0us Dir Delay:1us Idle Delay:250ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (150.000,450.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Trinamic Stepper Uart Tx:gpio.17 Rx:gpio.16 RTS:NO_PIN Baud:115200]
[MSG:INFO:     tmc_2209 Step:gpio.14 Dir:gpio.25 Disable:NO_PIN Addr:0 R:0.110]
[MSG:INFO: Axis Y (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Axis Z (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:ERR: X Axis TMC driver not detected - expected 0x21 got 0x0]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: Connecting to STA SSID:Wifi]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.2.158]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
```

### User Interface Software

FluidTerm

### What happened?

My driver is not detected with the config file I provided, receiving "TMC driver not detected - expected 0x21 got 0x0".
Note I'm using FluidNC 3.7 (where the uart settings is still part of the motor definition).
The connections are correct as the driver works fine with another code using uart.
Using an esp32 devkit c V4 from AZdelivery, TM2209 driver from fystec v4.0, stepperonline 17HS19-2004S1

It must be a stupid mistake but I can't figure it out.

Any help would be great.
Thanks so much and thanks for this awesome software.

### Other Information

I tested the circuit (esp32, TM2209 driver from fystec v4.0, stepperonline 17HS19-2004S1) with the below code and working fine. 

`#include <Arduino.h>

#include <SpeedyStepper.h> //Simple & good stepper library, get it.

#include <TMCStepper.h>

#define DIR_PIN          25 // Direction
#define STEP_PIN         14 // Step
#define SERIAL_PORT Serial2 // HardwareSerial port pins 16 & 17
#define DRIVER_ADDRESS 0b00 // TMC2209 Driver address according to MS1 and MS2

#define R_SENSE 0.11f // Match to your driver
                      // SilentStepStick series use 0.11
                      // UltiMachine Einsy and Archim2 boards use 0.2
                      // Panucatt BSD2660 uses 0.1
                      // Watterott TMC5160 uses 0.075

bool shaft = false;  // ONLY NEEDED FOR CHANGING DIRECTION VIA UART, NO NEED FOR DIR PIN FOR THIS

TMC2209Stepper driver(&SERIAL_PORT, R_SENSE, DRIVER_ADDRESS);

SpeedyStepper stepper;

void setup() {

stepper.connectToPins(STEP_PIN, DIR_PIN); // INITIALIZE SpeedyStepper
  
SERIAL_PORT.begin(115200);      // INITIALIZE UART TMC2209
Serial.begin(115200);
delay(500);
Serial.println(F("Serial Initialized"));

  driver.begin();                // Initialize driver
                           
  driver.toff(5);                 // Enables driver in software

  driver.rms_current(600);       // Set motor RMS current
  driver.microsteps(2);            // Set microsteps to 1/2

  driver.pwm_autoscale(true);    // Needed for stealthChop
  driver.en_spreadCycle(true);   // false = StealthChop / true = SpreadCycle

  stepper.setCurrentPositionInSteps(0);                   // Set zero position
  stepper.setSpeedInStepsPerSecond(400);              //Set Speed
  stepper.setAccelerationInStepsPerSecondPerSecond(400);   //Set acceleration, smaller value for super smooth direction changing

}

void loop() {

uint16_t msread=driver.microsteps();
Serial.print(F("Read microsteps via UART to test UART receive : "));    Serial.println(msread); 
  
Serial.println(F("Move 6400 steps forward at 600ma"));
driver.rms_current(600); 
stepper.moveToPositionInSteps(6400);

Serial.println(F("Wait 3sec and turn current low so you can turn the motor shaft"));
driver.rms_current(10); 
delay(3000);

Serial.println(F("Move back to 0 position at 300ma"));
driver.rms_current(300); 
stepper.moveToPositionInSteps(0);

//MOVE MOTOR VIA UART AND CHANGE DIRECTION VIA SOFTWARE, IT RUNS AS LONG AS YOU LET IT... PROBABLY ONLY USEFUL WITH ENCODER. THE VALUE SETS ONLY THE SPEED.

driver.VACTUAL(16000); //SET SPEED OF MOTOR
delay(2000); // MOTOR MOVES 2 SEC THEN STOPS
driver.VACTUAL(0); //STOP MOTOR BY SETTING SPEED TO 0
shaft = !shaft; // REVERSE DIRECTION
driver.shaft(shaft); // SET DIRECTION

}`

bdring, 2023-04-05:
Show a schematic and photo of your controller.

Generally we do not provide support for older revisions of the firmware.

bdring, 2023-04-05:
You might have TX and RX backwards.

We can't help too much with DIY wiring. We know the firmware works. It is likely your wiring or config.

I know it "works" with the other firmware, but the firmware works for me on many different PCBs with TMC2209.

floco, 2023-04-05:
Thanks again.
I tried inverting tx and rx, but then the other code does not work, which tends to indicate the wiring is correct.
Must be something with the config like you said. I will try reducing the parameters to what is stricly required to see if I can isolate what's happening. Do I understand correctly that the ESP32 is supposed to receive 0x21 via UART to identify it is a TMC2209 but in my case it's receiving something else? Is there a way to bypass this check to see if it goes further?
I also ordered that one, to be able to compare:
http://wiki.fluidnc.com/en/hardware/3rd-party/fysetc_e4

bdring, 2023-04-05:
You inverted the tx and rx. Did FluidNC work?

floco, 2023-04-05:
nope, same message.

bdring, 2023-04-05:
There is no way to bypass the test without hacking the firmware.

Try seeing what the other firmware gets with 
```cpp
driver.version()
```

floco, 2023-04-05:
ok will check that tomorrow and report back.
Thanks and have great evening or day ;-)

floco, 2023-04-06:
well that's interesting:
```
uint8_t version=driver.version();
Serial.print(F("Version : "));    Serial.println(version); 
```
Gives:
Version : 0
I will reach to fysetc and/or the vendor to see what they have to say.

I will otherwise try to change this and see what happens:
[image]

Thanks again for your help

bdring, 2023-04-06:
Does your alternate firmware report back settings correctly if you change them?

floco, 2023-04-06:
```
uint16_t msread=driver.microsteps();
Serial.print(F("Read microsteps via UART to test UART receive : "));    Serial.println(msread); 
uint8_t version=driver.version();
Serial.print(F("Version : "));    Serial.println(version); 
uint32_t coolconf = driver.COOLCONF();
Serial.print("CoolStep Current: "); Serial.println(coolconf, BIN);
uint32_t drvStatus = driver.DRV_STATUS();
Serial.print("DRVSTATUS: "); Serial.println(drvStatus, BIN);
```
Read microsteps via UART to test UART receive : 256
Version : 0
CoolStep Current: 0
DRVSTATUS: 0

I thought the first one proved the UART was receiving, but I guess 256 is just the default value :-(
Looks like this is back to being an HW or wiring issue.
I have pin 17/TX connected to RX on the driver and 16/RX connected to TX (which on S2209 3.1 or 4.0 is supposed to be connected to RX via the resistor).
But anyway, nothing to do with FluidNC then. Sorry about that
Back to the drawing board.
Thanks again for your help

bdring, 2023-04-06:
Please donate to this project!