bdring/FluidNC #853: Problem: Servo glitches when idle_ms is enabled

MarkJB, 2023-04-06:
### Controller Board

ESP32 Plotter Controller Rev B (aka "Emerald Dingo") https://github.com/theworkisthework/ESP32_Plotter_Controller

Mostly pin compatible with [Grbl_ESP32 CNC Development Board](https://www.tindie.com/products/33366583/grbl_esp32-cnc-development-board-v41/)

### Machine Description

At this point in time, I'm just bench testing servos with the controller and no machine.

### Configuration file

```yaml
name: ESP32_Plotter_Controller_RevB_Emerald_Dingo
board: Emerald Dingo (RevB)

kinematics:
  CoreXY:

stepping:
  engine: RMT
  idle_ms: 250
  dir_delay_us: 0
  pulse_us: 3
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: gpio.2
  x:
    steps_per_mm: 100.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 2
      mpos_mm: 0.000
      positive_direction: true
      settle_ms: 250.000
      seek_mm_per_min: 2000.000
      feed_mm_per_min: 200.000
      seek_scaler: 1.100
      feed_scaler: 5.000
    motor0:
      limit_all_pin: gpio.34:low
      hard_limits: false
      tmc_2130:
        direction_pin: gpio.26
        step_pin: gpio.25
        run_amps: 0.250
        hold_amps: 0.125
        microsteps: 16
        stallguard: 16
        stallguard_debug: false
        run_mode: CoolStep
        homing_mode: CoolStep
        use_enable: false
        r_sense_ohms: 0.110
        cs_pin: gpio.17
    motor1:
      null_motor:
  y:
    steps_per_mm: 100.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 2
      mpos_mm: 0.000
      positive_direction: true
      settle_ms: 250.000
      seek_mm_per_min: 2000.000
      feed_mm_per_min: 200.000
      seek_scaler: 1.100
      feed_scaler: 5.000
    motor0:
      limit_all_pin: gpio.35:low
      hard_limits: false
      tmc_2130:
        direction_pin: gpio.33
        step_pin: gpio.32
        run_amps: 0.250
        hold_amps: 0.125
        microsteps: 16
        stallguard: 16
        stallguard_debug: false
        run_mode: CoolStep
        homing_mode: CoolStep
        use_enable: false
        r_sense_ohms: 0.110
        cs_pin: gpio.16
    motor1:
      null_motor:
  z:
    steps_per_mm: 200.000
    max_rate_mm_per_min: 555.000
    acceleration_mm_per_sec2: 500.000
    max_travel_mm: 24.000
    # KPOWER DMC809 - calculated ~180deg circumference travel = 37mm is more like 24mm linear
    soft_limits: false
    hard_limits: false
    homing:
      cycle: 1
      mpos_mm: 12.000
      positive_direction: true
      settle_ms: 250.000
      seek_mm_per_min: 2000.000
      feed_mm_per_min: 200.000
      seek_scaler: 1.100
      feed_scaler: 5.000
    motor0:
      rc_servo:
        pwm_hz: 50
        output_pin: gpio.27
        min_pulse_us: 2500
        max_pulse_us: 750
    motor1:
      null_motor:

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5

control:
  reset_pin: gpio.0:low:pu

coolant:
  delay_ms: 1000.000

probe:
  check_mode_start: false

macros:
  startup_line0:
  startup_line1:
  macro0:
  macro1:
  macro2:
  macro3:

start:
  must_home: true
  check_limits: true
  deactivate_parking: false

user_outputs:

# relay:
#   tool_num: 0
#   speed_map: 0=0.0% 1000=100.0%
#   spinup_ms: 0
#   spindown_ms: 0
#   output_pin: gpio.27
#   disable_with_s0: false
#   s0_with_disable: false
#   spinup_ms: 0
#   spindown_ms: 0
arc_tolerance_mm: 0.002
junction_deviation_mm: 0.010
verbose_errors: false
report_inches: false
enable_parking_override_control: false
use_line_numbers: false
```

### Startup Messages

```text
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:12812
load:0x40080400,len:3032
entry 0x400805e4

[MSG:INFO: FluidNC v3.6.8]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:WARN: Ignored key hard_limits]
[MSG:INFO: Machine ESP32_Plotter_Controller_RevB_Emerald_Dingo]
[MSG:INFO: Board Emerald Dingo (RevB)]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:0]
[MSG:INFO: Stepping:RMT Pulse:3us Dsbl Delay:0us Dir Delay:0us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable gpio.2]
[MSG:INFO: Axis X (-300.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2130 Step:gpio.25 Dir:gpio.26 CS:gpio.17 Disable:NO_PIN Index:-1 R:0.110]
[MSG:INFO:  X All Limit gpio.34:low]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Y (-300.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2130 Step:gpio.32 Dir:gpio.33 CS:gpio.16 Disable:NO_PIN Index:-1 R:0.110]
[MSG:INFO:  Y All Limit gpio.35:low]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Z (-12.000,12.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     rc_servo Pin:gpio.27 Pulse Len(2500,750 period:1048575)]
[MSG:INFO:   Motor1]
[MSG:INFO: X Axis driver test passed]
[MSG:INFO: Y Axis driver test passed]
[MSG:INFO: reset_pin gpio.0:low:pu]
[MSG:INFO: Kinematic system: CoreXY]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: Connecting to STA SSID:TP-Link_EAC7]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.1.114]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.6 [FluidNC v3.6.8 (wifi) '$' for help]
[MSG:INFO: '$H'|'$X' to unlock]
```

### User Interface Software

WebUI, FluidTerm

### What happened?

When stepping idle_ms is enabled, the servo randomly glitches (unexpected moves, random directions and positions).

I was testing various servos sending repeated up and down commands (G0 Z0, G0 Z24) and every once in a while (maybe one time in 10) the servo would swing to an unexpected position and stay there. Issuing a new command would reposition the servo correctly until it glitched again.

I turned off the idle (`idle_ms: 255`) and no longer see the issue.
Setting `idle_ms: 250` causes the glitch.

### Other Information

Observed the PWM signal jumping to an unexpected pulse width on an oscilloscope.

MitchBradley, 2023-04-06:
It is probably disabling the servo when the motors go idle.

bdring, 2023-04-06:
If the axis goes faster than the servo can go there will be problems.

MarkJB, 2023-04-07:
Would either of these cause the servo to swing back on itself and stop a position outside of the expected position?

Maybe its related to  `mpos_mm`? (Or me misunderstanding how that should be used).

As per above config, I've set the servo `max_travel_mm: 24.000` and homing `mpos_mm: 12.000`. This results in the servo starting at Z0 on power on and has a range of +/-12mm (this is desired for variable pressure (any Z values below 0 increase pen/brush pressure)). 

Sometimes when issuing say G0Z12, the servo will swing to that position and then swing back past Z0, stopping closer to Z-12.

I would then have to issue a new move to get it into the correct position (issuing G0Z12 again does not cause the servo to move, but G0Z11 would).

ignis32, 2026-04-09:
>>That probably would not happen with if the servo behaves as you say.

Makes sense. I've made some tests and my assumption about servo following the last known order like Hiroo Onoda did not survive reality check.

It indeed stops keeping an angle on idle and relaxes. In that case,  do not know what is happening.
I do not either think that it goes to min position due to gravity, even unpowered servo has enough resistance because of the gearbox.

>> Do you really need to disable anything, as opposed to setting idle_ms=255?

Drivers are noticably heating up even when nothing is happening, would be nice to evade that, so I would not need to turn off the plotter. 

P.S.
By the way, thank you for answering my questions, I really appreciate this open communication.

bdring, 2026-04-10:
Maybe add a new argument.

```cpp
void IRAM_ATTR Axes::set_disable(axis_t axis, bool disable, bool manual_override) {
```