bdring/FluidNC #856: Dangerous safety issue

Xplode85, 2023-04-09:
### Controller Board

xPro V5

### Machine Description

Ratrig Killerbee with RS485 controlled spindle

### Configuration file

```yaml
---
```

### Startup Messages

```text
---
```

### User Interface Software

cncjs

### What happened?

While testing safety on the CNC running FluidNC i found some serious safety issues.

If a hard limit or soft limit is triggered the stepper motors loose power creating the issue of the Z axis lowering by gravity, if this happens while the spindle is running the spindle will stay running.

Here's the hole my CNC made when i trigged the hard limit while running G code, Z axis lowered by gravity and the spindle stayed running. 

[image]

The same thing happens when doing a soft trigger.
When a hard or soft limit is triggered i would at least expect the spindle to be stopped, but prefer also all axis to stay powered with holding power.

bdring, 2023-04-09:
Missing Info

Xplode85, 2023-04-09:
I updated the original post with my configuration, had go back to the workshop.
I do unfortunately not have any message from the log, it's just the standard message about hitting hard or soft limit thou so nothing special.
Maybe there is a possibility to get more debugging info?

I'd also like to mention that i tried to get into the Fluid NC Wiki to edit some information, but wasn't able to make an account, i never got a verification email.
What i tried to fix is on the page for xPro V5, more specifically this:
[image]
gpio.4 should be TX and gpio.25 should be RX

MitchBradley, 2023-04-09:
We need the startup messages that you get by sending '$ss'

Xplode85, 2023-04-09:
I'm not sure if this is of any help, i connected to the CNC controller PC remotely, but the xPro V5 doesn't have 24v power so might be some missing stuff.
Cant look more into this tomorrow if needed.

$ss
[MSG:INFO: FluidNC v3.6.8]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine CNC_xPRO_V5_XYYZ_NC v1.2]
[MSG:INFO: Board CNC xPRO V5]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:0]
[MSG:INFO: Stepping:RMT Pulse:4us Dsbl Delay:0us Dir Delay:0us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (0.000,1300.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_5160 Step:gpio.12 Dir:gpio.14:low CS:gpio.17 Disable:NO_PIN Index:1 R:0.050]
[MSG:INFO:  X All Limit gpio.35]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Y (4.000,1284.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_5160 Step:gpio.27 Dir:gpio.26 CS:NO_PIN Disable:NO_PIN Index:2 R:0.050]
[MSG:INFO:  Y All Limit gpio.36]
[MSG:INFO:   Motor1]
[MSG:INFO:     tmc_5160 Step:gpio.33 Dir:gpio.32 CS:NO_PIN Disable:NO_PIN Index:3 R:0.050]
[MSG:INFO:  Y2 All Limit gpio.34]
[MSG:INFO: Axis Z (0.000,70.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_5160 Step:gpio.15 Dir:gpio.2 CS:NO_PIN Disable:NO_PIN Index:4 R:0.050]
[MSG:INFO:  Z All Limit gpio.39]
[MSG:INFO:   Motor1]
[MSG:ERR: X Axis TMC driver not detected - expected 0x30 got 0x0]
[MSG:ERR: Y Axis TMC driver not detected - expected 0x30 got 0x0]
[MSG:ERR: Y2 Axis TMC driver not detected - expected 0x30 got 0x0]
[MSG:ERR: Z Axis TMC driver not detected - expected 0x30 got 0x0]
[MSG:INFO: safety_door_pin gpio.16:pu]
[MSG:INFO: macro1_pin gpio.13:low:pu]
[MSG:INFO: macro2_pin gpio.0:low:pu]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: UART1 Tx:gpio.4 Rx:gpio.25 RTS:NO_PIN Baud:9600]
[MSG:INFO: Huanyang Spindle  Tx:gpio.4 Rx:gpio.25 RTS:NO_PIN Baud:9600]
[MSG:INFO: Using spindle Huanyang]
[MSG:INFO: Mist coolant gpio.21]
[MSG:INFO: Probe Pin: gpio.22:low:pu]
[MSG:INFO: Connecting to STA SSID:Wifi]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connecting...]
[MSG:INFO: Connected - IP is 10.228.34.103]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

MitchBradley, 2023-04-09:
That is the information that we need.  Do you see the ERR lines?  The problem template specifically says that you should look for them.

Xplode85, 2023-04-09:
I see the ERR messages for the stepper drivers, i don't think they are relevant right now since the controller doesn't have 24v connected to power the stepper drivers.
I will repost a new startup message tomorrow when i can test with 24v connected to the controller.

Xplode85, 2023-04-10:
Updated the first post with the startup message

bdring, 2023-04-10:
I checked the FluidNC code and compared it to standard Grbl. I think they both disable steppers on alarm. 

Most Z axes will not fall when off. This is undesirable during power failure. Designs that are susceptible to this, like the original ShapeOko 3 add a spring to deal with this.

[image]

I have been considering modifications to the motor disable logic for other reasons. Some people have expressed desires to have different behavior for different axes. For example: Some axes would disable on idle and some would not.

We could add a config item at the axis level like `idle_behavior:` or `disable_behavior:`. The values could be a emulated list like, `always_enabled`, `idle_disable` and possibly others to help people like you.

The developers would need to have a long conversation about the implications and how commands like `$MD` and `$ME`, etc. are affected.

Note: All spindles have an `off_on_alarm:` setting. Yours is likely false, because you did not have that in your config and it defaults to false. You can see your value by sending the `$Huanyang` command.

mac7988, 2023-04-10:
Just like in industrial CNC machines, the axis motors will be disabled in the event of a limit or an alarm. These machines usually have a clutch or a counterweight to prevent the Z-axis from dropping. Providing users with the option to customize this behavior could potentially introduce more safety concerns. It's worth noting that many people, myself included, often don't thoroughly read well-documented Wiki

Xplode85, 2023-04-10:
The big problem is that the spindle isn't turned off together with the steppers, maybe just add an option to automatically send stop spindle message when hitting a limit?

I know about the off_on_alarm: setting for the spindle, this would not really help much here without some kind of extra safety system in play to use the limit switches signals to trigger the alarm on the VFD. Still wouldn't help with soft limits.

The reason the spindle is dropping is probably because of the TR8x8 lead screw, a shallower angle lead screw would be better, also it's just barely dropping...sometimes it stays and sometimes it drops, i guess it's more likely to drop when the Z axis is already moving.
I have plans to replace the lead screws if i get the budget for it.

bdring, 2023-04-10:
Send `$Huanyang` and paste the reply here.

MitchBradley, 2023-04-10:
off_on_alarm applies to alarms caused by limit switches.  It sends the stop command to the spindle when FluidNC enters alarm state.

Xplode85, 2023-04-10:
Thanks guys.
I found i missunderstood how the alarm on the VFD worked i thought the VFD only had a physical way to send "alarm" and stop the spindle.

Will check the setting tomorrow, it's probably disabled so most likely this is solved :)

Xplode85, 2023-04-11:
Here's the command...

$Huanyang
/Huanyang:
  uart: 
    txd_pin: gpio.4
    rxd_pin: gpio.25
    rts_pin: NO_PIN
    cts_pin: NO_PIN
    baud: 9600
    mode: 8N1
  uart_num: -1
  modbus_id: 1
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 100
  speed_map: 0=0.000000% 0=25.000000% 6000=25.000000% 24000=100.000000%
  off_on_alarm: false

The weird thing is that i have set off_on_alarm: true in the config file and it still stays false.

closed 2023-04-11 by Xplode85