bdring/FluidNC #857: Feature: Configuration for MKS LTS 1.1

ckocyigit, 2023-04-11:
### Machine Context

I have a simple XY 400x400mm Laser engraver/cutter from makibes with the model x1 but I would guess that those are just some rebranded standard chassis etc.

### Feature Description

The machine utilizes a long term support kind of makerbases motherboards.

Makerbase MKS LTS 1.1 

I could successfully install FluidNC on it and also running it was no problem. Head movement worked however Laser and SD card did not. I've started plotting down the pinouts and connections and it seems to be very similiar to the dlc32 boards.

It would be awesome if someone, who is in the matter could adapt for this board.

### Other Approaches

Example config for this board.

### How I Can Help

I can share pictures and try to measure and plot connections on the board,

MitchBradley, 2023-04-11:
Another thing that would be awesome would be if the manufacturers that make hardware that depends on our software would contribute money to our project, to defray at least some of the costs that we incur in supporting yet another hardware variant every week or so.  But that is not likely to happen, so the next most awesome thing would be if more users would donate.

ckocyigit, 2023-04-12:
@dorsi90 Thank you very much

I've uploaded it and it seems to be working for the laser.
Unfortunately SDCard is not being recognized. I will check if the GPIO's corrospond correctly.
(I've also formatted as suggested by FluidNC)

You should MR this file into the example folder for future users 😄

ckocyigit, 2023-04-12:
I've tried a different sd card.
It seems to be something different. Log says: [MSG:ERR: sdmmc_card_init failed code 0x107] on either sdcard.

Yes I know that, its currently outside of the frame. Do you have discord/email for further messaging?

dorsi90, 2023-04-12:
@ckocyigit yes you can write me to info@tronica.it
Are sd cards formatted as fat32?

MitchBradley, 2023-04-12:
You can try adding ` frequency_hz: 1000000` to the `sdcard:` section, indented the same as the other items under sdcard:  That might help.

You can also try reformatting the SD Card using a small allocation unit size.  Large allocation units might require larger amounts of RAM than the ESP32 has.

Some MicroSD cards simply do not support SPI mode at all.  For those the error is often 0x108 .  It can be hard to tell in advance which cards support SPI mode - but older, smaller cards in the 4GB range are often good.  Newer, larger cards typically have lots of advanced features and can drop support for legacy modes like SPI.

oldschool91, 2023-06-21:
Hello, my TTS 55 10w burner is also equipped with an MKS LTS v1 card, and I would like to be able to control my air assit with the SCL output as in this tutorial "Trees Totem S / Air Assist Control - #9 by Trek2120 " , but the commands are not active in the firmware is it possible to modify the firmware? If yes, how?
I contacted MKS who referred me to TTS and TTS told me that they did not have the firmware...
Thanks for your help
Christopher

bdring, 2023-06-21:
Most people add air assist via coolant commands.

Please read the wiki to learn how to add this to your config. It is quite easy and does not require a compile.

oldschool91, 2023-06-21:
I did as in the wiki, but it does not work on my card, hence my question the firmware should not allow it

oldschool91, 2023-06-21:
ok not to do anything stupid or I create a new post? I am a little lost

oldschool91, 2023-06-21:
I just figured out how to open a post I won't bother you here anymore.
Sorry

Lusui8787, 2023-08-08:
purchased a MAKIBES X1 laser engraver. Connected with Laser GRBL program. But I don't understand where to get the GCODE configuration file for this laser. And therefore the laser head itself moves, but the laser itself does not appear. Help how to fix it. Thank you sincerely.

[image]

MitchBradley, 2023-08-08:
The signon message says "SimpleLaser 1.1h".  This GitHub repo is for FluidNC, not SimpleLaser.  Ask the people who sold you the machine.

Mtjkrjcr, 2023-08-15:
Hello, could someone send me a picture of the motherboard?

ckocyigit, 2023-12-28:
> Hello, could someone send me a picture of the motherboard?

[image]

[image]

denizzzka, 2024-07-21:
Interested in it too: it installed by manufacturer Flyingbear into Laserman engravers

Also, here is display connector:
[image]

Display "Makerbase" connected by two rainbow wires: 
[image]

denizzzka, 2024-07-21:
Also, they use Z axis to obtain info from the fire (PIR?) sensor: 
[image]

If fire burns on table sensor stops system. Maybe this is important, I don't know maybe this is common practice

closed 2024-12-12 by bdring

denizzzka, 2024-12-18:
@ckocyigit can you publish your binary for this controller? Thank you!