bdring/FluidNC #858: Problem: Problem getting the configuration in Openbuilds Control v1.0.364

Itsmetoo9, 2023-04-12:
### Controller Board

MKS TinyBee

### Machine Description

Not linked with the machine

### Configuration file

```yaml
board: MKS TinyBee V1.0 XXYYZ
name:

kinematics:
  Cartesian:

i2so:
  bck_pin: gpio.25
  data_pin: gpio.27
  ws_pin: gpio.26

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  # uses TH2 IO34 active low - MAKE SURE jumper J2 is set to SDDET!!!
  card_detect_pin: gpio.34:low
  # frequency_hz: 10000000

stepping:
  engine: I2S_STATIC
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 2

probe:
  pin: gpio.14:low:pu
  #check_mode_start: true

axes:
  x:
    steps_per_mm: 160
    max_rate_mm_per_min: 3000.000
    acceleration_mm_per_sec2: 80.000
    max_travel_mm: 770.000
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 1500.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.33:high:pd
      hard_limits: true
      pulloff_mm: 14.000
      stepstick:
        step_pin: I2SO.1
        direction_pin: I2SO.2
        disable_pin: I2SO.0

    # use E0 driver for 2nd X axis motor
    motor1:
      hard_limits: true
      pulloff_mm: 4.000
      stepstick:
        step_pin: I2SO.10
        direction_pin: I2SO.11
        disable_pin: I2SO.9

  y:
    steps_per_mm: 160
    max_rate_mm_per_min: 3000.000
    acceleration_mm_per_sec2: 70.000
    max_travel_mm: 770.000
    soft_limits: true
    homing:
      cycle: 3
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 2000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.32:high:pd
      hard_limits: false
      pulloff_mm: 30.000
      stepstick:
        step_pin: I2SO.4
        direction_pin: I2SO.5
        disable_pin: I2SO.3

    # use E1 driver for 2nd Y axis motor
    motor1:
      limit_neg_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 30.000
      stepstick:
        step_pin: I2SO.13
        direction_pin: I2SO.14
        disable_pin: I2SO.12

  z:
    steps_per_mm: 400.000
    max_rate_mm_per_min: 2000.000
    acceleration_mm_per_sec2: 60.000
    max_travel_mm: 140.000
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 140.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 500.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_pos_pin: gpio.22:high:pd
      hard_limits: true
      pulloff_mm: 3.000
      stepstick:
        step_pin: I2SO.7
        direction_pin: I2SO.8
        disable_pin: I2SO.6

control:
  safety_door_pin: NO_PIN
  # on MT_DET connector
  reset_pin: gpio.35:low
  # on TH1 connector
  feed_hold_pin: gpio.36:low
  # on TB connector
  cycle_start_pin: gpio.39:low
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

coolant:
  # Heated Bed Terminal Block
  flood_pin: i2so.16
  # HE0 Terminal Block
  mist_pin: i2so.17
  delay_ms: 0

# spindle PWM signal
PWM:
  pwm_hz: 5000
  # on EXP1 IO15 connector.
  # note IO15 will give some short pulses on boot, which may activate spindle
  # use IO17 on EXP1 to avoid
  output_pin: gpio.15:high
  direction_pin: NO_PIN
  enable_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  tool_num: 0
  spinup_ms: 0
  spindown_ms: 0
  speed_map: 0=0.000% 24000=100.000%

#Laser:
#  pwm_hz: 5000
#  # on 3D Touch connector (has pdwn + PWM)
#  output_pin: gpio.2:high:pd
#  s0_with_disable: true
#  tool_num: 1
#  speed_map: 0=0.000% 1000=100.000%

start:
  must_home: false
```

### Startup Messages

```text
Nothing specific
```

### User Interface Software

OpenBuilds Control

### What happened?

When you go on the Grbl Settings Tab or FluidNC Settings tab, the displayed information is interrupted by <Idle...> messages, rending the configuration incorrect:

axes: 
  shared_stepper_disable_pin: NO_PIN
  shared_stepper_reset_pin: NO_PIN
  x: 
    steps_per_mm: 160.000000
    max_rate_mm_per_min: 8000.000000
    acceleration_mm_per_sec2: 80.000000
    max_travel_mm: 810.000000
    soft_limits: true
    homing: 
      cycle: 2
      allow_single_axis: true
      positive_direction: false
      mpos_mm: 0.000000
      feed_mm_per_min: 300.000000
      seek_mm_per_min: 1500.000000
      settle_ms: 500
      seek_scaler: 1.100000
      feed_scaler: 1.100000
    motor0: 
      limit_neg_pin: gpio.33:pd
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 4.000000
      stepstick: 
        step_pin: I2SO.1
        direction_pin: I2SO.2
        disable_pin: I2SO.0
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: NO_PIN
        reset_pin: NO_PIN
    motor1: 
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 4.000000
      stepstick: 
        step_pin: I2SO.10
        direction_pin: I2SO.11
        disable_pin: I2SO.9
        ms1_pin: NO_PIN
**<Idle|MPos:20.000,0.000,0.000|FS:0,0>**
        ms2_pin: NO_PIN
        ms3_pin: NO_PIN
        reset_pin: NO_PIN

### Other Information

No idea how to fix this...

bdring, 2023-04-12:
Something is requesting status during the reading of the config. Status is an immediate command.

Those commands have the highest priority in the system. We do not want to bloke them.

petervanderwalt, 2023-04-12:
Thanks @bdring added a note to disable query loop during config dump  https://github.com/OpenBuilds/OpenBuilds-CONTROL/issues/283#issuecomment-1505532537 

We still have a bunch of todo's in https://github.com/OpenBuilds/OpenBuilds-CONTROL/issues/283 (If you look it over and see any of our plans being the wrong thing / wrong way to do it - happy to hear about it! ) - trying to support FluidNC with the minimum of additional work (hehe) so basically need to parse the config to get some values we used to get out of grbl/grblHAL's $$,   and then a new config editor to upload yaml (xmodem - love it) and then our machine profiles to be converted to yaml (planning to build the yaml on the fly off our existing grbl profiles)

MitchBradley, 2023-04-12:
FYI, FluidNC has a new feature that can help with status reporting. http://wiki.fluidnc.com/en/support/interface/automatic_reporting .  A sender can turn on auto-reporting and then will not need to issue explicit ? queries - nor $G's.

petervanderwalt, 2023-04-12:
> 

Thanks - in time we may tweak CONTROL to cater to firmware-specific features, at the moment lack of time means the more we can reuse all the code we already did for grbl, the better - grblHAL slot in nicely there with minimal effort (: