bdring/FluidNC #861: Problem: Wrong resistor lables on the board

Saschko2000, 2023-04-14:
### Controller Board

FluidNC Pen/Laser CNC Controller (SPI)

### Machine Description

https://oshwlab.com/bdring/tmc2130-2-axis

### Configuration file

```yaml
doesn't matter
```

### Startup Messages

```text
doesn't matter
```

### What happened?

The lables for 2 resitor pairs are switched.
- R17 / R19
- R18 / R20

### Other Information

The lables for 2 resitor pairs are switched.
- R17 / R19
- R18 / R20

bdring, 2023-04-14:
Fixed, The robot that builds these does not car :-)

closed 2023-04-14 by bdring

Saschko2000, 2023-04-14:
Thanks for the fast fix :-)

I ordered 5 boards from PCBWay, and they are all placed wrong :-(
And they are really tiny to hand solder ;-)

bdring, 2023-04-14:
The 1 click order from JLC always worked and probably cheaper.