bdring/FluidNC #862: v3.6.8 SD Card not work.

Teleskopik, 2023-04-16:
### Controller Board

self-made

### Machine Description

3-axis milling machine

### Configuration file

```yaml
name: "ESP32 Dev Controller V4"
board: "ESP32 Dev Controller V4"

stepping:
  engine: RMT
  idle_ms: 250
  dir_delay_us: 10
  pulse_us: 5
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  
  x:
    steps_per_mm: 200
    max_rate_mm_per_min: 5000
    acceleration_mm_per_sec2: 500
    max_travel_mm: 1000
    homing:
      cycle: 2
      mpos_mm: 10
      positive_direction: false
    
    motor0:
      limit_all_pin: NO_PIN
      stepstick:
        direction_pin: gpio.33
        step_pin: gpio.32
    motor1:
      null_motor:

  y:
    steps_per_mm: 200
    max_rate_mm_per_min: 5000
    acceleration_mm_per_sec2: 500
    max_travel_mm: 700
    homing:
      cycle: 2
      mpos_mm: 10
      positive_direction: false

    motor0:
      limit_all_pin: NO_PIN
      stepstick:
        direction_pin: gpio.26
        step_pin: gpio.25
    motor1:
      null_motor:

  z:
    steps_per_mm: 800
    max_rate_mm_per_min: 3000
    acceleration_mm_per_sec2: 500
    max_travel_mm: 200
    homing:
      cycle: 1
      mpos_mm: 10
      positive_direction: true

    motor0:
      limit_all_pin: NO_PIN
      stepstick:
        direction_pin: gpio.14
        step_pin: gpio.27
    motor1:
      null_motor:

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN
  
coolant:
  flood_pin: NO_PIN
  mist_pin:  NO_PIN

        
probe:
  pin: gpio.34:low:pu

  
  
PWM:
  pwm_hz: 5000
  output_pin: gpio.4
  enable_pin: gpio.22
  direction_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 0
  speed_map: 0=0% 10000=100%
```

### Startup Messages

```text
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:12812
load:0x40080400,len:3032
entry 0x400805e4

[MSG:INFO: FluidNC v3.6.8]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:3axis_v4.yaml]
[MSG:WARN: gpio.34:low does not support :pu attribute]
[MSG:INFO: Machine ESP32 Dev Controller V4]
[MSG:INFO: Board ESP32 Dev Controller V4]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:2000000]
hz = 2000000
[MSG:INFO: Stepping:RMT Pulse:5us Dsbl Delay:0us Dir Delay:10us Idle Delay:250ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (10.000,1010.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.32 Dir:gpio.33 Disable:NO_PIN]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Y (10.000,710.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.25 Dir:gpio.26 Disable:NO_PIN]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Z (-190.000,10.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.27 Dir:gpio.14 Disable:NO_PIN]
[MSG:INFO:   Motor1]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: PWM Spindle Ena:gpio.22 Out:gpio.4 Dir:NO_PIN Freq:5000Hz Period:8191]
[MSG:INFO: Using spindle PWM]
[MSG:INFO: Probe Pin: gpio.34:low]
[MSG:INFO: Connecting to STA SSID:Serg]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.0.34]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.6 [FluidNC v3.6.8 (wifi) '$' for help]
[MSG:INFO: '$H'|'$X' to unlock]
```

### User Interface Software

WebUI version: github.com/MitchBradley/ESP3D-WEBUI@a90d79b

### What happened?

No SD card
[MSG:ERR: mount_to_vfs failed code 0xffffffff]

### Other Information

When downloading the FluidNC-3.4.0 firmware, the SD card works.

MitchBradley, 2023-04-16:
Try a lower value for frequency_hz per http://wiki.fluidnc.com/en/config/sd_card#sdcard-section

mac7988, 2023-04-17:
Is there an external SD CARD PCB known to work and not create this issue?

Teleskopik, 2023-04-17:
[image]
When downloading the FluidNC-3.4.0 firmware, the SD card works.

When downloading the FluidNC-3.6.8 firmware, No SD card.
reduced the frequency to 100,000 khz

[MSG:INFO: Machine ESP32 Dev Controller V4]
[MSG:INFO: Board ESP32 Dev Controller V4]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:100000]
hz = 100000

bdring, 2023-04-17:
3.6.8 works with 100,000 ?

Teleskopik, 2023-04-17:
No.

MitchBradley, 2023-04-17:
Is it an MMC card, not an SD card?  The library that we use now only works with SD and does not support MMC on SPI.

MitchBradley, 2023-04-17:
2 GB cards can have problems too, because they use an internal block size that is different from all other card sizes.

Teleskopik, 2023-04-19:
It worked with the new SD card.

closed 2023-04-19 by bdring