bdring/FluidNC #863: Problem:

netechsys, 2023-04-17:
### Controller Board

6-pack

### Machine Description

Cartesian YYXZ, 2 Spindles (10V T0, Laser T1)

### Configuration file

```yaml
board: 6-pack
name: 6Pack Design
meta: 
stepping:
  engine: I2S_stream
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 6
  disable_delay_us: 5
  segments: 12

axes:
  shared_stepper_disable_pin: NO_PIN
  shared_stepper_reset_pin: NO_PIN
  x:
    steps_per_mm: 400.785
    max_rate_mm_per_min: 6000.000
    acceleration_mm_per_sec2: 500.000
    max_travel_mm: 1200.000
    soft_limits: false
    homing:
      cycle: 2
      allow_single_axis: true
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 1000.000
      seek_mm_per_min: 1000.000
      settle_ms: 250
      seek_scaler: 1.100
      feed_scaler: 5.000

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.33
      hard_limits: true
      pulloff_mm: 5.000
      stepstick:
        step_pin: I2SO.2:low
        direction_pin: I2SO.1:low
        disable_pin: I2SO.0
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: NO_PIN
        reset_pin: NO_PIN

    motor1:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      null_motor:

  y:
    steps_per_mm: 160.275
    max_rate_mm_per_min: 6000.000
    acceleration_mm_per_sec2: 500.000
    max_travel_mm: 1350.000
    soft_limits: false
    homing:
      cycle: 2
      allow_single_axis: true
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 1000.000
      seek_mm_per_min: 1000.000
      settle_ms: 250
      seek_scaler: 1.100
      feed_scaler: 5.000

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.32:low
      hard_limits: true
      pulloff_mm: 5.000
      stepstick:
        step_pin: I2SO.5:low
        direction_pin: I2SO.4:low
        disable_pin: I2SO.7
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: I2SO.6
        reset_pin: NO_PIN

    motor1:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.34:low
      hard_limits: false
      pulloff_mm: 4.7
      stepstick:
        step_pin: I2SO.10:low
        direction_pin: I2SO.9:low
        disable_pin: I2SO.8
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: I2SO.11
        reset_pin: NO_PIN

  z:
    steps_per_mm: 199.000
    max_rate_mm_per_min: 1500.000
    acceleration_mm_per_sec2: 500.000
    max_travel_mm: 200.000
    soft_limits: false
    homing:
      cycle: 1
      allow_single_axis: true
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 3000.000
      seek_mm_per_min: 3000.000
      settle_ms: 250
      seek_scaler: 1.100
      feed_scaler: 5.000

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.35
      hard_limits: true
      pulloff_mm: 5.000
      stepstick:
        step_pin: I2SO.13
        direction_pin: I2SO.12
        disable_pin: I2SO.15
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: I2SO.14
        reset_pin: NO_PIN

    motor1:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      null_motor:

kinematics:
  Cartesian:

i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN
  frequency_hz: 0
  
control:
  safety_door_pin: gpio.36:low
  reset_pin: NO_PIN
  feed_hold_pin: gpio.25
  cycle_start_pin: gpio.39
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

coolant:
  flood_pin: gpio.13
  mist_pin: gpio.15
  delay_ms: 1000

probe:
  pin: gpio.2:pu
  check_mode_start: false

macros:
  startup_line0: 
  startup_line1: 
  macro0: 
  macro1: 
  macro2: 
  macro3: 

start:
  must_home: true
  deactivate_parking: false
  check_limits: true

parking:
  enable: false
  axis: Z
  target_mpos_mm: -5.000
  rate_mm_per_min: 800.000
  pullout_distance_mm: 5.000
  pullout_rate_mm_per_min: 250.000

user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: gpio.12
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN

arc_tolerance_mm: 0.002
junction_deviation_mm: 0.010
verbose_errors: false
report_inches: false
enable_parking_override_control: false
use_line_numbers: false
planner_blocks: 16
10V:
  forward_pin: gpio.4
  reverse_pin: gpio.16
  pwm_hz: 5000
  direction_pin: NO_PIN
  output_pin: gpio.26
  enable_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 6000
  spindown_ms: 6000
  tool_num: 0
  speed_map: 0=0.000% 24000=100.000%
  off_on_alarm: true
Laser:
  pwm_hz: 5000
  output_pin: gpio.14
  enable_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  tool_num: 1
  speed_map: 0=0.000% 100=100.000%
  off_on_alarm: true
```

### Startup Messages

```text
Grbl 3.6 [FluidNC v3.6.8 (wifi) '$' for help]
[MSG:INFO: '$H'|'$X' to unlock]
[MSG:INFO: FluidNC v3.6.8]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:config-cgs2.yaml]
[MSG:INFO: Machine CGS 6Pack Design]
[MSG:INFO: Board 6-pack]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:0]
[MSG:INFO: Stepping:I2S_stream Pulse:4us Dsbl Delay:5us Dir Delay:6us Idle Delay:255ms]
[MSG:INFO: User Digital Output:0 on Pin:gpio.12]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (0.000,1200.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.2:low Dir:I2SO.1:low Disable:I2SO.0]
[MSG:INFO:  X All Limit gpio.33]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Y (0.000,1350.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.5:low Dir:I2SO.4:low Disable:I2SO.7]
[MSG:INFO:  Y All Limit gpio.32:low]
[MSG:INFO:   Motor1]
[MSG:INFO:     stepstick Step:I2SO.10:low Dir:I2SO.9:low Disable:I2SO.8]
[MSG:INFO:  Y2 All Limit gpio.34:low]
[MSG:INFO: Axis Z (-200.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.13 Dir:I2SO.12 Disable:I2SO.15]
[MSG:INFO:  Z All Limit gpio.35]
[MSG:INFO:   Motor1]
[MSG:INFO: safety_door_pin gpio.36:low]
[MSG:INFO: feed_hold_pin gpio.25]
[MSG:INFO: cycle_start_pin gpio.39]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: 10V Spindle Ena:NO_PIN Out:gpio.26 Dir:NO_PIN Fwd:gpio.4 Rev:gpio.16 Freq:5000Hz Period:8191]
[MSG:INFO: Laser Ena:NO_PIN Out:gpio.14 Freq:5000Hz Period:8191]
[MSG:INFO: Using spindle 10V]
[MSG:INFO: Flood coolant gpio.13]
[MSG:INFO: Mist coolant gpio.15]
[MSG:INFO: Probe Pin: gpio.2:pu]
[MSG:INFO: Connecting to STA SSID:Home-FN]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 10.250.254.83]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
ok
```

### User Interface Software

WebUI / OpenBuilds Control / Lightburn

### What happened?

If I configure the system with the above configuration, the Laser will not turn on.  The PWM signal is not sent to the Laser.  If I change the configuration to PWM with the same configurations, the Laser powers up as expected.  I am just adding the Laser to the system, so this is the first time I am trying to run with both the 10V spindle and the Laser/PWM spindle.  Changing back and forth is working.

BlueOrangeLive, 2023-04-17:
The laser only works if:
They will only operate during a G1, G2, or G3 moves. They will not operate during G0, Jog, Homing etc.
If you need it to operate in those modes, use a PWM spindle.
     They turn off when in idle or doing a rapid move.
     M3 is constant power and M4 is dynamic power mode (scales linearly with speed during accel/decel)

http://wiki.fluidnc.com/en/config/config_spindles

netechsys, 2023-04-17:
Thanks. I tried running a job from Lightburn, and the movement worked, but the laser never turned on. I was connecting to the controller over http, not serial. I also exported the g-code from lightburn, and tried to run it from Openbuilds controller over a serial connection, and same results. 

I think the config is correct, since it worked for pwm, but needed to make sure I didn't miss something.

BlueOrangeLive, 2023-04-17:
Using multiple spindles

You can define as many as your hardware will support. They will act totally independently. You must use separate I/O pins for each one. Simply add each spindle definition to the config file. Each must have a unique tool_num:

--->>> You switch between tools with the tool change gcode (M6). M6 T2 would switch to the spindle that covers tool number 2. 

-- >> your Laser is M6 T1 <<--  in your yaml File.

If the spindles are offset, you will need to deal with that.

    The simplest way to to rezero the new spindle.
    You could use a separate coordinate system like G54 for one spindle and G55 for another.
    You could create a little macro that zeros the machine based on the known offset, like move to zero, move the offset amount and rezero.

bdring, 2023-04-22:
Can this issue be closed?

closed 2023-04-27 by bdring