bdring/FluidNC #865: Feature: Add tangential knife / oscillating knife support

jejmule, 2023-04-18:
### Machine Context

I have a Chinese CNC die-cutter (2000 x 1000) equipped with tangential knife unit and a diode laser. The machine has 3 axis : XY and C to rotate the knife. The knife is moved up and down with a basic pneumatic stage, the cutting depth is managed by the tool himself.

Currently, I have to switch between two different types of controller depending on my tool : laser or knife.
- For tangential knife : Edding CNC provide a good piece of software to control tangential knife : you can send XY g-code to the machine and the C Axis is automatically generated based on some parameters. This is great, but there is a limitation since it requires a motorized Z axis to move the cutter. I made a custom arduino to read Z axis information and depending on height to switch on/off control vane to move the knife up/down. I have also modified an inkscape plugin to generate XYC gcode for my cutter
- For laser cutting : my version of Eding CNC do not have the feature to change laser intensity while cutting (G1,G2,G3).... I have installed GRBL on arduino, it works great but do not manage tangential knife as I would like. I am using lightburn that only generate XY gcode and doesn't support C axis.

My dream : A single controller to manage both tangential knife and laser that I can manage with simple XY gcode

### Feature Description

Add an axis configuration to manage tangential knife.

Axis could be configured as a tangential knife. In this case, the C axis orientation while follow the XY cutting direction vector. If the angle between two consecutive vectors is larger than max_angle, the knife is lift, rotate, lift down and so on.
Here is a set of parameters define by Edding CNC to manage knife : https://docs.edingcnc.com/settings/interpreter#ptrQW

Edding CNC parameters are a good start, but something is missing : the option to choose between a motorized Z axis or a pneumatic one controled on a GPIO pin

### Other Approaches

...

### How I Can Help

I have programming knowledge that I am using professionally and for my hobby : python, C++, Arduino, starting some ESP32 projects

I do not know at all about the internal architecture of FLUID NC, but I will be pleased to have a try and work on this implementation. I will need a hand to know where to look at in the code and where to do the modification.
It looks like a kind of post-processor to generate Caxis and Z commands based on XY trajectory.

MitchBradley, 2023-04-18:
It might be possible to create a kinematics module that derives a C axis value based on the trajectory.  Look in the Kinematics subdirectory to see how the various existing kinematic systems compute motor positions from axis moves.

jejmule, 2023-04-19:
Thank you for your input. Indeed, kinematics seems to be the right place.
I will fork and play with it :)
I am new to fluid NC is there a developer documentation?

MitchBradley, 2023-04-19:
The only docs are what is on the wiki.

jejmule, 2023-04-25:
I am working on it and facing an issue to know the current motor position : 

bool TangentialKnife::cartesian_to_motors(float* target, plan_line_data_t* pl_data, float* position) {
        //calculate direction of motion in XY plane
        float angle = atan2(target[1] - position[1], target[0] - position[0]) * 180 / M_PI;
        //get current angle of knife
        //???
}

COREXY kinematics use transform_cartesian_to_motors with current position and target to retrieve current motor positions. But it doesn't work in case of tangential knife since the knife angle is determined by direction of motion and not positions.

What trick or function while you used to determine C axis position/angle

Thanks

bdring, 2023-04-25:
Look at the real time status reporting to see how to get the current position.

It is unlikely you want the current position in most cases for kinematics. Gcode is processed in advance of motion and placed into a buffer. The current position many have no relation to the move you are working on.

Gcodes basically say to go to a point. They do not say go from one point to the new point. If you need to know the previous point, you must track it yourself. After each move is processed, save the target as the previous location for the next move. The feed rate compensation in non linear kinematics uses this method.

jejmule, 2023-04-25:
@bdring thanks for your input. Indeed, I realized that after posting my question. I am keeping current knife angle as a private attributes for my kinematics.

My strategy is to have a variable for the current knife_angle and a flag to know if the variable is defined for first execution of cartesian_to_motors. Or does the C axis position is 0 after execution of init?

Another question about the length of position and target array. Minimal configuration for a tangential knife is a XYZC machine, it means 6 axis are defined with AB as null axis. Is it correct that target and position lengths are 6?

I hope, I do not bother you with my basic questions since I am discovering the code.
Thanks for help and congrats for this great project ;)

bdring, 2023-04-25:
Use MAX_N_AXIS to define the size of any position array.

jejmule, 2023-04-25:
Another question : 

My GCODE will only contain XY position, no Z and C. Is it correct that target and position arraysw will only have XY values (in am not talking about array length) since it generated when parsing GCODE?

I think, I am almost good with my kinematics. Curious to have a test on my machine. Now, I have to move to the hardware side to connect the MKS board to my machine. I will let you know

bdring, 2023-04-25:
You need to copy the value on unused axes.

Althafzahra, 2023-11-05:
Hallo, Have you finished this project? 
I plan to make a Tangential knife machine

bdring, 2024-12-27:
Closing. This is not a bug or enhancement that the developers would add.

closed 2024-12-27 by bdring