bdring/FluidNC #866: Problem: 'Run'-, 'Idle'-Messages brack/slowdown execution GCode, pushing EndStop solves issue

qrstu, 2023-04-19:
### Controller Board

MKS DLC32

### Machine Description

I'm using my SculpfunLaser also as SolderPast despenser. Added Z-Axis

### Configuration file

```yaml
$cd
board: None
name: MKS DLC32 PastDespencer
meta: (12/02/2023) mroe
stepping: 
  engine: I2S_static
  idle_ms: 25
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0
  segments: 12
i2so: 
  bck_pin: gpio.16
  data_pin: gpio.21
  ws_pin: gpio.17
i2c0: 
  sda_pin: gpio.0
  scl_pin: gpio.4
  frequency: 100000
spi: 
  miso_pin: gpio.12
  mosi_pin: gpio.13
  sck_pin: gpio.14
sdcard: 
  cs_pin: gpio.15
  card_detect_pin: NO_PIN
  frequency_hz: 0
kinematics: 
  Cartesian: 
axes: 
  shared_stepper_disable_pin: I2SO.0
  shared_stepper_reset_pin: NO_PIN
  x: 
    steps_per_mm: 80.000000
    max_rate_mm_per_min: 5000.000000
    acceleration_mm_per_sec2: 250.000000
    max_travel_mm: 410.000000
    soft_limits: false
    homing: 
      cycle: 1
      allow_single_axis: true
      positive_direction: false
      mpos_mm: 0.000000
      feed_mm_per_min: 75.000000
      seek_mm_per_min: 1000.000000
      settle_ms: 500
      seek_scaler: 1.100000
      feed_scaler: 1.100000
    motor0: 
      limit_neg_pin: gpio.36:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 10.000000
      stepstick: 
        step_pin: I2SO.1
        direction_pin: I2SO.2
        disable_pin: NO_PIN
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: NO_PIN
        reset_pin: NO_PIN
  y: 
    steps_per_mm: 80.000000
    max_rate_mm_per_min: 5000.000000
    acceleration_mm_per_sec2: 250.000000
    max_travel_mm: 400.000000
    soft_limits: false
    homing: 
      cycle: 1
      allow_single_axis: true
      positive_direction: false
      mpos_mm: 0.000000
      feed_mm_per_min: 75.000000
      seek_mm_per_min: 1000.000000
      settle_ms: 500
      seek_scaler: 1.100000
      feed_scaler: 1.100000
    motor0: 
      limit_neg_pin: gpio.35:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 10.000000
      stepstick: 
        step_pin: I2SO.5
        direction_pin: I2SO.6
        disable_pin: NO_PIN
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: NO_PIN
        reset_pin: NO_PIN
  z: 
    steps_per_mm: 1600.000000
    max_rate_mm_per_min: 500.000000
    acceleration_mm_per_sec2: 500.000000
    max_travel_mm: 300.000000
    soft_limits: false
    homing: 
      cycle: 1
      allow_single_axis: true
      positive_direction: true
      mpos_mm: 0.000000
      feed_mm_per_min: 100.000000
      seek_mm_per_min: 800.000000
      settle_ms: 500
      seek_scaler: 1.100000
      feed_scaler: 1.100000
    motor0: 
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.34:low
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 10.000000
      stepstick: 
        step_pin: I2SO.3
        direction_pin: I2SO.4
        disable_pin: NO_PIN
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: NO_PIN
        reset_pin: NO_PIN
control: 
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN
coolant: 
  flood_pin: gpio.25
  mist_pin: NO_PIN
  delay_ms: 0
probe: 
  pin: NO_PIN
  check_mode_start: true
macros: 
  startup_line0: 
  startup_line1: 
  macro0: 
  macro1: 
  macro2: 
  macro3: 
start: 
  must_home: false
  deactivate_parking: false
  check_limits: false
parking: 
  enable: false
  axis: Z
  target_mpos_mm: -5.000000
  rate_mm_per_min: 800.000000
  pullout_distance_mm: 5.000000
  pullout_rate_mm_per_min: 250.000000
user_outputs: 
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN
oled: 
  i2c_num: 0
  i2c_address: 60
  width: 128
  height: 64
  radio_delay_ms: 1000
Relay: 
  direction_pin: NO_PIN
  output_pin: gpio.32
  enable_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 1
  speed_map: 0=0.000000% 1=100.000000%
  off_on_alarm: false
arc_tolerance_mm: 0.002000
junction_deviation_mm: 0.010000
verbose_errors: false
report_inches: false
enable_parking_override_control: false
use_line_numbers: false
planner_blocks: 16
```

### Startup Messages

```text
[MSG:INFO: FluidNC v3.6.8]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:2023-03-03_PastDespener.yaml]
[MSG:INFO: Machine MKS DLC32 PastDespencer]
[MSG:INFO: Board None]
[MSG:INFO: I2SO BCK:gpio.16 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.14 MOSI:gpio.13 MISO:gpio.12]
[MSG:INFO: SD Card cs_pin:gpio.15 detect:NO_PIN freq:0]
[MSG:INFO: I2C SDA: gpio.0, SCL: gpio.4, Freq: 100000, Bus #: 0]
[MSG:INFO: OLED I2C address:0x3c width: 128 height: 64]
[MSG:INFO: Stepping:I2S_static Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:25ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable I2SO.0]
[MSG:INFO: Axis X (0.000,410.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.1 Dir:I2SO.2 Disable:NO_PIN]
[MSG:INFO:  X Neg Limit gpio.36:low]
[MSG:INFO: Axis Y (0.000,400.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.5 Dir:I2SO.6 Disable:NO_PIN]
[MSG:INFO:  Y Neg Limit gpio.35:low]
[MSG:INFO: Axis Z (-300.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.3 Dir:I2SO.4 Disable:NO_PIN]
[MSG:INFO:  Z Pos Limit gpio.34:low]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Relay Spindle Ena:NO_PIN Out:gpio.32 Dir:NO_PIN]
[MSG:INFO: Using spindle Relay]
[MSG:INFO: Flood coolant gpio.25]
[MSG:INFO: Connecting to STA SSID:RHOME]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connecting...]
[MSG:INFO: Connecting....]
[MSG:INFO: Connected - IP is 192.168.0.128]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
ok
```

### User Interface Software

WebUI,Despenser.exe (https://github.com/MakeMagazinDE/Solder-Paste-Dispenser)

### What happened?

- Homing works fine --> no issues
- when I run GCode generated by despenser.exe from Gerber-File (gpt) I get following issues:
G0 X14.14 Y0.63 ok 
G0 Z2 ok 
M3 S1000 ok 
G4 P0.1900 ok 
M5 ok 
G0 Z-0.05 ok 
G1 Z0.1 F200 ok 
G0 Z2 ok
**<Run|MPos:0.000,0.000,1.553|FS:500,0>
<Idle|MPos:0.000,0.000,2.000|FS:0,0>
<Idle|MPos:0.000,0.000,2.000|FS:0,0>
<Idle|MPos:0.000,0.000,2.000|FS:0,0>**
G0 X14.14 Y2.53 ok 
G0 Z2 ok 
M3 S1000 ok 
G4 P0.1900 ok 
M5 ok 
G0 Z-0.05 ok 
G1 Z0.1 F200 ok 
G0 Z2 ok 
**<Run|MPos:0.000,1.900,1.562|FS:500,>
<Idle|MPos:0.000,1.900,2.000|FS:0,0>
<Idle|MPos:0.000,1.900,2.000|FS:0,0>
<Idle|MPos:0.000,1.900,2.000|FS:0,0>
<Idle|MPos:0.000,1.900,2.000|FS:0,0>
<Idle|MPos:0.000,1.900,2.000|FS:0,0>**
G0 X16.34 Y1.58 ok 
G0 Z2 ok 
M3 S1000 ok 
G4 P0.1900 ok 
M5 ok 
G0 Z-0.05 ok 
G1 Z0.1 F200 ok 
G0 Z2 ok 
**<Run|MPos:2.200,0.950,1.551|FS:500,0>**
G0 X0.00 Y0.00 ok 
G0 Z2 ok 
M3 S1000 ok 
G4 P0.2580 ok 
M5 ok 
G0 Z-0.05 ok 
G1 Z0.1 F200 ok 
G0 Z2 ok 
G0 X0.00 Y1.89 ok 
G0 Z2 ok 
M3 S1000 ok 
G4 P0.2580 ok 
M5 ok 
G0 Z-0.05 ok 
G1 Z0.1 F200 ok 
G0 Z2 ok 
G0 X0.00 Y8.25 ok 
G0 Z2 ok 
M3 S1000 ok 
G4 P0.2580 ok 
M5 ok 
G0 Z-0.05 ok 
G1 Z0.1 F200 ok 
G0 Z2 ok 
G0 X0.00 Y10.15 ok 
G0 Z2 ok 
M3 S1000 ok 
G4 P0.2580 ok 
M5 ok 
G0 Z-0.05 ok 
G1 Z0.1 F200 ok 
G0 Z2 ok 
G0 X13.21 Y12.57 ok 
G0 Z2 ok 
M3 S1000 ok 
G4 P0.2580 ok 
M5 ok 
G0 Z-0.05 ok 
G1 Z0.1 F200 ok 
G0 Z2 ok 
G0 X13.21 Y14.47 ok 
G0 Z2 ok 
M3 S1000 ok 
G4 P0.2580 ok 
M5 ok 
G0 Z-0.05 ok 
G1 Z0.1 F200 ok 
G0 Z2 ok 
G0 X20.32 Y16.89 ok 
G0 Z2 ok 
M3 S1000 ok 
G4 P0.2580 ok 
M5 ok 
G0 Z-0.05 ok 
G1 Z0.1 F200 ok 
G0 Z2 ok 
**<Run|MPos:6.175,16.263,1.562|FS:500,0>
<Idle|MPos:6.175,16.263,2.000|FS:0,0>
<Idle|MPos:6.175,16.263,2.000|FS:0,0>
<Idle|MPos:6.175,16.263,2.000|FS:0,0>
<Idle|MPos:6.175,16.263,2.000|FS:0,0>
<Idle|MPos:6.175,16.263,2.000|FS:0,0>
<Idle|MPos:6.175,16.263,2.000|FS:0,0>**
G0 X20.32 Y18.79 ok 
G0 Z2 ok 
M3 S1000 ok 
G4 P0.2580 ok 
M5 ok 
G0 Z-0.05 ok 
G1 Z0.1 F200 ok 
G0 Z2 ok 
**<Run|MPos:6.175,18.163,1.021|FS:500,0>
<Idle|MPos:6.175,18.163,2.000|FS:0,0>
<Idle|MPos:6.175,18.163,2.000|FS:0,0>
<Idle|MPos:6.175,18.163,2.000|FS:0,0>
<Idle|MPos:6.175,18.163,2.000|FS:0,0>
<Idle|MPos:6.175,18.163,2.000|FS:0,0>**

GCode runes with brackes. Each Bracke takes about 2-5 seconds. **Marked Bold**

If I push any of the endstops during the run of the Gcode, maschine doesn't generate and 'Run'-, 'Idle'-Messages anymore.
and speed of execution is as expected. No stops occure.
I have no glue why that is happening. I recognized this by accident that pushing endstop solves the issue ;-)
Is it a bug or miss configuration ?
Regards
qrstu

qrstu, 2023-04-20:
@bdring 
If it's disabled by default, then I'm not using auto reporting.
The application despenser.exe generates GRBL 0.9 code. Unfortuantly I can't see the debug messages in WebUI when I'm connected with the despenser-App to the MKS DLC32.
Does that mean I have to activate auto reporting, I'm going to test.
For me it's strange that Gcode is running as expected during one of the endstops has been triggered.
I don't receive any <Run|MPos:6.175,18.163,1.021|FS:500,0>,<Idle|MPos:6.175,18.163,2.000|FS:0,0> Messages.
Regards
qrstu

bdring, 2023-04-20:
Auto reporting is off by default.  FluidNC will only send messages when requested in normal mode. 

Your application is probably hiding the responses. Does it have a verbose mode that shows all responses?

You do not have hard limits enabled, so touching the limit switches will not generate an alarm. The pin state is not shown in the responses, so that is a problem. Have you tested the switch responses using FluidTerm?

http://wiki.fluidnc.com/en/support/help_with_switch_problems

MitchBradley, 2023-04-20:
Auto reporting is per-sender, so one user interface can turn it on but another will leave it off.  Very few - perhaps no - third-party senders know about auto-reporting yet, so it is unlikely that the despenser app uses auto reporting.

WebUI does use auto-reporting by default, so you will see automatic status reports when  you are using WebUI - but WebUI's use of auto reports should not affect the despenser app.

Another debugging technique for switch problems is to send $message/level=debug .  If  you do that then [MSG:DEBUG ...] messages will be sent to the sender console when switch states change.

qrstu, 2023-04-20:
@bdring
The switch (Endstop) responses I can see in fluidterm. They are all ok.
But I don't have a verbose mode in my application to see what will be sent. 
As long I don't have the possibility to log the hole communication between FluidNC and an Application I can't figure out what's going wrong :-( (Telnet, WebUI, Fluidterm) ex. App Connected by COM9 and Logging by Telnet Port23
--> **The workaround for me is to push on "Endstop" then everything is working as expected.**
**Or probably a future request for Logging that the hole communication on a specific COM-Port can be logged by Telnet/Fluidterm/WebGUI.**
Thank you for your help.
I think you can close the issues
Regards qrstu

bdring, 2023-04-20:
There are some comm port spy programs you can search the Internet for. They allow you to see the comm port communications.

closed 2023-04-20 by bdring

qrstu, 2023-04-21:
@bdring 
Could monitor traffic with Serial Analizer :-)
--> App is sending --> ? --> In undefineded sequences after sending gcode which produces breaks. 
Pushed Endstop suppress sending "?"

MitchBradley, 2023-04-21:
I looked at the source code for the Dispenser app.  It will not work with FluidNC because it depends on the ancient Grbl 0.9 status report format.  Grbl 1.1, released in 2016, has a slightly different status report.  FluidNC status reports are compatible with Grbl 1.1.  In addition to the status report format incompatibility, I saw several aspects of the Dispenser code that look suspicious.  I think that there are a lot of valid situations that will make that code stop working.  If you need to use that app, you will either have to use an old controller that runs Grbl 0.9, or convince the app's author to make it support the Grbl 1.1 protocol.

qrstu, 2023-04-22:
@MitchBradley 
Thank you for your review.
I will stay on FluidNC and let it run with EndStop pressed, then it's running for me. Not nice but working :-)
I know that the Jogging in the App isn't compatible with GRBL 1.1