bdring/FluidNC #872: Problem: MKS DLC32 V2.1 cannot open Config after wrong flash Firmware

Jet-A1, 2023-04-24:
### Controller Board

MKS DLC32 V2.1

### Machine Description

Sculpfun S10 modified. 2 Belt driven Axes X, Y. 10W Diode Laser Modul TTL driven, fixed Focus. No Z Axes. 

### Configuration file

```yaml
# Basic configuration file for X/Y Laser using GT2 belts 1.8ºNEMA motors
# at 16 microSteps and 16 Tooth pulleys

board: MKS-DLC32 V2.1
name: 3dpBurner3
meta: villamany 30/Jan/2023

kinematics:
  Cartesian:

stepping:
  engine: I2S_STATIC
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0
axes:
  shared_stepper_disable_pin: I2SO.0
  x:
    steps_per_mm: 100
    max_rate_mm_per_min: 16000
    acceleration_mm_per_sec2: 1500
    max_travel_mm: 895
    soft_limits: false
    homing:
      cycle: 0
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 500.000
      seek_mm_per_min: 16000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.36:low
      hard_limits: false
      pulloff_mm: 2.000
      stepstick:
        step_pin: I2SO.1
        direction_pin: I2SO.2:low

  y:
    steps_per_mm: 100
    max_rate_mm_per_min: 16000
    acceleration_mm_per_sec2: 1500
    max_travel_mm: 880
    soft_limits: false
    homing:
      cycle: 0
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 500.000
      seek_mm_per_min: 16000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.35:low
      hard_limits: false
      pulloff_mm: 2.000
      stepstick:
        step_pin: I2SO.5
        direction_pin: I2SO.6:high
 
  z:
    steps_per_mm: 80
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 500.000
    max_travel_mm: 80.000
    soft_limits: true
    homing:
      cycle: 0
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 200.000
      seek_mm_per_min: 500.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.34
      hard_limits: false
      pulloff_mm: 1.000
      stepstick:
        step_pin: I2SO.3
        direction_pin: I2SO.4
        
i2so:
  bck_pin: gpio.16
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.12
  mosi_pin: gpio.13
  sck_pin: gpio.14

sdcard:
  cs_pin: gpio.15
  card_detect_pin: NO_PIN
#  card_detect_pin: gpio.39
  
control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

macros:
  startup_line0:
  startup_line1:
  macro0:
  macro1:
  macro2:
  macro3:

coolant:
  flood_pin: NO_PIN
  mist_pin: NO_PIN
  delay_ms: 0

#probe:
#  pin: gpio.22
#  check_mode_start: true

Laser:
  pwm_hz: 5000
  output_pin: gpio.32
  disable_with_s0: false
  s0_with_disable: false
  tool_num: 0
  speed_map: 0=0.000% 1000=100.000%
#PWM:
#  direction_pin: NO_PIN
#  spinup_ms: 1000
#  spindown_ms: 1000
  
user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  #Beeper connector. Command: "M62 P0" turn beeper On. 
  #"G4 P0.5" wait for 0.5s. "M63 P0" turn beeper OFF
  digital0_pin: I2SO.7
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN

start:
  must_home: false

#MKS DLC32 v2.1 pinMap

# X- gpio.36
# SD_DET gpio.39
# Z- gpio.34
# Y- gpio.35
# SPINDLE PWM gpio.32
# LCD_RS gpio.33
# LCD_CS_INV gpio.25
# LCD_TOUCH_CS_INV gpio.26
# LCD_RST_INV gpio.27
# SD_SCK gpio.14
# SD_DO gpio.12
# SD_DI gpio.13
# SD_CS gpio.15
# I2C_SDA gpio.0
# I2C_SCL gpio.4
# I2S_BCK gpio.16
# I2S_WS gpio.17
# LCD_EN_INV gpio.5
# LCD_SCK gpio.18
# LCD_MISO gpio.19
# I2S_DATA gpio.21
# RXD0 gpio.3
# TXD0 gpio.1
# Probe gpio.22
# LCD_MOSI gpio.23

# XYZ_EN I2SO.0
# X_STEP I2SO.1
# X_DIR I2SO.2
# Z_STEP I2SO.3
# Z_DIR I2SO.4
# Y_STEP I2SO.5
# Y_DIR I2SO.6
# BEEPER I2SO.7
```

### Startup Messages

```text
$ss
[MSG:INFO: FluidNC v3.6.8]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:WARN: Cannot open configuration file:MKS_DLC32_V21_laser.yaml]
[MSG:INFO: Using default configuration]
[MSG:INFO: Axes: using defaults]
[MSG:INFO: Machine Default (Test Drive)]
[MSG:INFO: Board None]
[MSG:INFO: Stepping:RMT Pulse:4us Dsbl Delay:0us Dir Delay:0us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Axis Y (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Axis Z (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: STA SSID is not set]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
ok
```

### User Interface Software

Lightburn, GRBL

### What happened?

The first time insatlling Fluidnc i did after erase.bat the install-fs.bat, witch broke up with Error.

I did today the following:

-Start Controller with 12v wait 3 sec. than connect USB. Power Supply is an 750W PC Power Supply. Computer is Windows 64 22H2 all Updates done.
-Then start Erase.bat from fluidnc-v3.6.8-win64 Screenshot
-Switch Power off and disconnect USB, wait 10 sec.
-Switch Power on wait 3 sec. connect USB
-Start install-wifi.bat from fluidnc-v3.6.8-win64 Screenshot
-Get Warn MSG could not open Configuration

### Other Information

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
![Screenshot (15)](https://user-images.githubusercontent.com/22130877/233910181-3e8e3169-8acc-4873-8
[image]
966-8a5fac211198.png)
[image]
![Screenshot (18)](https://user-images.githubusercontent.com/22130877/233910226-629e3aaf-58a7-4925-a499-c02da8
[image]
76043
[image]
2.png)
![Screenshot (21)](https://user-images.githubusercontent.com/22130877/233910269-6155d465-c759-46ef-a529-ff7e53a6
[image]
b01d.png)
![Screenshot (23)](https://user-images.githubusercontent.com/22130877/233910301-29be768f-73ef-421e-a204-0e3fd3
[image]
a29674.png)
[image]
[image]
[image]

MitchBradley, 2023-04-24:
I do not see any indications above of either install-fs.bat - which will create a filesystem for the config file to occupy - or an upload of the config file to that filesystem.  Please follow these instructions http://wiki.fluidnc.com/en/installation , especially the items about install-fs and uploading the config file.

Jet-A1, 2023-04-24:
[image]
[image]

Ok i now have Web Connection, so far so good, when i understand right now i have to send a Configuration File to get the Machine run, right ?

bdring, 2023-04-24:
Send `$LocalFS/List` to see all files on the local file system.

Cut and paste the response here.

Jet-A1, 2023-04-24:
Hello,
this is what the WebUI Console answered:

$LocalFS/List
[FILE: preferences.json|SIZE:769]
[FILE: favicon.ico|SIZE:1150]
[FILE: MKS_DLC32_v21_laser.yaml|SIZE:3758]
[FILE: index.html.gz|SIZE:115883]
[/spiffs/ Free:48.53 KB Used:120.84 KB Total:169.38 KB]
ok

bdring, 2023-04-24:
Send `$Config/Filename=MKS_DLC32_v21_laser.yaml` then reboot

Cut and paste the command if you can. You need to get it exactly correct including the letter case.

Jet-A1, 2023-04-25:
Oh my God, ok case sensitive, i understand, test this the next Days. Thanks for your Help. 

BTW i see you sell Laser Boards, but they are out of Stock. Nice Hardware you sell :-) Do you have a German Distributor ? The Laser Pen looks nice with TMC2009, does that one have an Output for Air Assist ? 

Greetings

bdring, 2023-04-25:
Can we close this issue?

Jet-A1, 2023-04-27:
Yes you can close the Issue, again, Thank you very much for your Help.

closed 2023-04-27 by MitchBradley