bdring/FluidNC #878: ATC Base Feature

Rich235, 2023-04-27:
### Machine Context

I have built a high precision 4 axis CNC metal mill and am porting FluidNC to it, replacing Mach3.
I'm also building an ISO20 spindle able to tool change.
I'd like to add a tool change over second UART (or via two spare ESP32 pins) via M6 Tx.
May I please ask you to share the atc.cpp file you show in your ATC video please here: https://www.youtube.com/watch?v=3ikLcC5NidU&ab_channel=BartonDring
That would allow me to understand more of your code base and ATC and probing integration approaches and probably save me a lot of time. I will of course share my outputs if viable.

### Feature Description

ATC base code over UART or two hard pins in FluidNC. People can then build on that for their own ATC approach

### Other Approaches

MTC - manual tool change, that is also problematic with current FluidNC, that I also wish to address to add functionality using UGS. I see some other senders interpret M6 and allow manual tool change and jogging to set Z zero. I can't see that in UGS.

### How I Can Help

Please just share the atc.cpp file, shared visually here: https://www.youtube.com/watch?v=3ikLcC5NidU&ab_channel=BartonDring

Thank you so much for all your excellent hard work on bringing CNC to many. I started with Mach3 in 2004 and it was all a lot harder than it is now thanks to people like you.

bdring, 2023-04-27:
I think this is it.

https://github.com/bdring/FluidNC/blob/ATC_Experiments/FluidNC/src/Spindles/KressATC.cpp

Rich235, 2023-04-29:
Thank you, yes, very helpful and impressive approach. I've got that into FluidNC and sorted the functions elsewhere so it now almost compiles but fails at the very end with a Collect2.exe error from this line in KressATC.cpp:  `motor_steps_to_mpos(saved_mpos, motor_steps);` with: " undefined reference to `motor_steps' "

Do I need to explicitly scope motor_steps using ‘::’? I've tried an extern definition of motor_steps in this file, but then get:

" undefined reference to `Spindles::motor_steps' "

You didn't need anything added with your GRBL version of KressATC so perhaps I'm missing something else about how motor_steps gets seen as already defined?

If you can point me in the right direction with this rather fundamental point, I'd be very grateful. Thank you very much.

Rich235, 2023-04-30:
OK, fixed using `get_motor_steps()` in place of `motor_steps`. Now compiles and links, but only if I comment out all the handling for retrieving the tool positions from the config yaml file.

Adding `virtual void item(const char* name, std::vector<float>& value)= 0;` in the HandlerBase.h file requires mods to the Generator.cpp and Generator.h files, which are OK, but also to the Completer.h file which are not OK. 

When I add `void item(const char* name, std::vector<float>& value) override { item(name); }` to the Completer.h file as you had in your GRBL version, I get errors from MachineConfig.cpp "pure virtual function "Configuration::HandlerBase::item(const char *name, std::vector<float, std::allocator<float>> &value)" has no overrider" 

In MachineConfig.cpp, this error is given for the functions `Configuration::ParserHandler handler(parser);` , `Configuration::AfterParse afterParse;` and `Configuration::Validator validator;`

I can't understand this as the overrider is there. Have you some guidance for this pls? Thanks.

MitchBradley, 2023-04-30:
It's hard to debug this from a description.  Publish a branch somewhere so we can look at the actual code.

Rich235, 2023-04-30:
OK thank you, will do, with some notes. In the meantime, I've put in to the KressATC.cpp file tool position coordinate values rather than YAML sourced values. Run the code, setup the KressATC spindle, but does nothing on T1 M6. So, understanding how to connect it up would also be much appreciated. How does it connect with or replace `void WEAK_LINK user_tool_change(uint32_t new_tool)`  in GCode.cpp? Thanks again.

MitchBradley, 2023-04-30:
A WEAK_LINK definition is a definition that is used if the symbol is not otherwise defined, so you just have to define user_tool_change() in some file.

Rich235, 2023-04-30:
Ah yes OK. I will investigate. Thanks.

Rich235, 2023-05-01:
Can you see my fork: [https://github.com/Rich235/FluidNC](url) ?
I've got basic tool change working now in FluidNC, thanks. BUT as above, the handler functions are not working. I've uncommented them so if you try and compile this you should see the issue. Line 54 in HandlerBase.h is needed for the floats for the tool positions from yaml into the KressATC.cpp file. Generator is OK, but Completer does not like it (or some other unrelated issue). Thanks for your help.

Rich235, 2023-05-01:
The yaml file I'm using is NewMill.yaml in the example_configs dir.

MitchBradley, 2023-05-01:
It appears to me that you neglected to pick up the various changes to configuration files per https://github.com/bdring/FluidNC/commit/c1484450f13cd0b4207ad4421f0e27a2ec8e75a0

Rich235, 2023-05-02:
Thanks very much for that comparison, I didn't know how to do that. All updated and working. I can continue with my development.

closed 2024-12-11 by bdring