bdring/FluidNC #883: Problem:  Installing and configuring z probe MKS DLC32 2.1

Renaultje, 2023-04-29:
### Controller Board

MKS DLC32 2.1

### Machine Description

3 axes cnc router with sanyo denki stepper motor and TMC2209 drivers

### Input Circuits

```yaml
z-probe connected to ground and signal
```

### Configuration file

```yaml
board: MKS-DLC32 V2.1
name: CNC Router
meta: (29.04.2023) by ME

kinematics:
  Cartesian:

stepping:
  engine: I2S_STATIC
  idle_ms: 10
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0
axes:
  shared_stepper_disable_pin: I2SO.0
  x:
    steps_per_mm: 265.657
    max_rate_mm_per_min: 500.000
    acceleration_mm_per_sec2: 150.000
    max_travel_mm: 325.000
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 500.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.36
      hard_limits: false
      pulloff_mm: 2.000
      stepstick:
        step_pin: I2SO.1
        direction_pin: I2SO.2

  y:
    steps_per_mm: 265.657
    max_rate_mm_per_min: 500.000
    acceleration_mm_per_sec2: 150.000
    max_travel_mm: 220.000
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 500.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.35
      hard_limits: false
      pulloff_mm: 2.000
      stepstick:
        step_pin: I2SO.5
        direction_pin: I2SO.6:low

  z:
    steps_per_mm: 265.657
    max_rate_mm_per_min: 500.000
    acceleration_mm_per_sec2: 350.000
    max_travel_mm: 80.000
    soft_limits: true
    homing:
      cycle: 0
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 1000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.34
      hard_limits: false
      pulloff_mm: 1.000
      stepstick:
        step_pin: I2SO.3
        direction_pin: I2SO.4

i2so:
  bck_pin: gpio.16
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.12
  mosi_pin: gpio.13
  sck_pin: gpio.14

sdcard:
  cs_pin: gpio.15
  card_detect_pin: NO_PIN

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: gpio.33:low:pu
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

macros:
  startup_line0:
  startup_line1:
  macro0: $SD/Run=lasertest.gcode
  macro1: $SD/Run=home.gcode
  macro2:
  macro3:

coolant:
  flood_pin: NO_PIN
  mist_pin: NO_PIN
  delay_ms: 0

probe:
  pin: gpio.22
  check_mode_start: true

Laser:
  pwm_hz: 5000
  #L on Beeper / IN on TTL
  output_pin: gpio.32
  enable_pin: I2SO.7
  disable_with_s0: false
  s0_with_disable: false
  tool_num: 0
  speed_map: 0=0.000% 0=12.500% 1700=100.000%
# 135=0mA 270=5mA 400=10mA 700=16mA
user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN

start:
  must_home: false

# 5,18,19,22,23,25,26,27,32,33,39,I2SO.7
# SDA 0 / SCL 4
```

### Startup Messages

```text
Startup is normal
```

### User Interface Software

WebUI

### What happened?

Installed the fluidNC 3.7.0 on MKS DLC32 2.1. It is working now but now i cannot setup the z probe correctly. It will probe when touching the probe but when i release it, it stops. This should be the other way around. So when it goes down and touches the probe it should stop. I cannot find the correct settings in the web ui.

I have found the probe settings for thickness.

Could anyone point me in the right way please...???

Kind regards
Renaultje

### Other Information

thanks in advance

bdring, 2023-04-29:
Send `?` to get the status of the probe when it is not touching anything.

If it says it is touching, invert the active state of the pin. The wiki should tell you how to do that.

bdring, 2023-04-29:
http://wiki.fluidnc.com/en/config/probe#setup

Renaultje, 2023-04-29:
THNX for the quick response, but i am a beginner with this stuff. i don't understand how this works, this is not a setting in the web UI? How can i set this?

MitchBradley, 2023-04-29:
The settings are not in the WebUI; WebUI is just one of many user interface programs that can talk to FluidNC.  The relevant settings are in the config file.  You might need to add :low to this line from your config file:
```
probe:
  pin: gpio.22
```
to make it read `pin: gpio.22:low`

You will need to edit the config file with a text editor on a host computer, then upload the new copy to FluidNC.  Search the wiki for "upload" to find documentation about that.

Renaultje, 2023-04-29:
Okay THNX this works, but now the thickness isn't right. When i give the thickness of the plate (wich is 19.22mm) it should zero to the bottom of the plate? after probing?

The thickness of the plate is set in webUI????

bdring, 2023-04-29:
Are you off by the direction? Try the same value multiplied by -1.

<http://wiki.fluidnc.com/en/config/probe#optional-p-parameter>

MitchBradley, 2023-04-30:
I have been looking at the WebUI probing code.  WebUI was written by a third party as a different project, so there are aspects of it that we have not inspected carefully, including probing.  According to my reading of the code, WebUI issues these commands:
```
G38.2 G91 Z-<Max travel> F<Feed rate>
G53 G0 Znn
G10 L20 P0 Z<Plate thickness>
G90
```
Breaking that down - the G38.2 command initiates probing in the -Z direction for the specified maximum distance at the specified feed rate.  The `G53 G0 Znn` command returns to the spot where the probe touch was first detected, in case there was a little extra motion afterwards - it is not an explicit backoff, but rather "return to touch point".  The G10 ... command then sets the Z coordinate of the current work coordinate system to the declared plate thickness.  G90 returns to absolute mode.

Renaultje, 2023-05-01:
Okay thanks for the reply, I'm trying to figure this out. I wil continu in the weekend and i will get back to it.

Again many thanks

commit 90d362d3 (2023-04-30): Added probe retract and use FluidNC G38.2 P parameter - Fixes #883
closed 2023-06-08 by bdring