bdring/FluidNC #887: How do I  edit the Config file for XYYZ homing?

jappiemike, 2023-05-02:
### Controller Board

6-pack External drives

### Machine Description

Gantry router with ball screws on all axis and dual steppers and drivers for Y axis. Dual proximity siteched on all axis.

### Configuration file

```yaml
board: 6 Pack
name: 6 Pack External XYZ 10V
meta: 2022-08-20 B. Dring
start:
  must_home: false
  deactivate_parking: false
  check_limits: true
stepping:
  engine: I2S_STATIC
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0
axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 160
    max_rate_mm_per_min: 3000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 490
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 10.000
      feed_mm_per_min: 250.000
      seek_mm_per_min: 1500
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100
    motor0:
      limit_neg_pin: gpio.34:low
      limit_pos_pin: gpio.35:low
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 5.000
      # Terminal Block 1
      standard_stepper:
        step_pin: I2SO.2
        direction_pin: I2SO.1: high
        disable_pin: I2SO.0
  y:
    steps_per_mm: 160
    max_rate_mm_per_min: 3000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 455
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: True
      mpos_mm: 10.000
      feed_mm_per_min: 250.000
      seek_mm_per_min: 1500
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100
    motor0:
      limit_neg_pin: gpio.32:low:pu
      limit_pos_pin: gpio.33:low:pu
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 5.000
      # Terminal Block 2
      standard_stepper:
        step_pin: I2SO.5
        direction_pin: I2SO.4:high
        disable_pin: I2SO.7
  z:
    steps_per_mm: 320
    max_rate_mm_per_min: 800.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 90
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 5.000
      feed_mm_per_min: 150.000
      seek_mm_per_min: 500.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100
    motor0:
      limit_neg_pin: gpio.36:low
      limit_pos_pin: gpio.39:low
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 5.000
      # Terminal Block 3
      standard_stepper:
        step_pin: I2SO.10
        direction_pin: I2SO.9:low
        disable_pin: I2SO.8
probe:
  pin: gpio.26:low:pd
  check_mode_start: true
control:
  safety_door_pin: gpio.16:low:pd
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN
i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17
spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18
sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5
10V:
  forward_pin: NO_PIN
  reverse_pin: NO_PIN
  pwm_hz: 3000
  output_pin: gpio.14
  enable_pin: NO_PIN
  direction_pin: NO_PIN
  disable_with_s0: true
  s0_with_disable: true
  spinup_ms: 10000
  spindown_ms: 10000
  tool_num: 0
  speed_map: 0=0.000% 24000=100.000%
  off_on_alarm: true
```

### Startup Messages

```text
There are just the normal messages no errors or warnings
```

### User Interface Software

Webui or UGS, I've not decided

### What happened?

I would like to know what to add to the config file to enable dual Y axis homing so that the machine squares itself every time It is homed. The 2 examples in the samples folder are not really clear.
Thanks
Mike

bdring, 2023-05-02:
Please see the wiki

http://wiki.fluidnc.com/

and the examples.

https://github.com/bdring/FluidNC/tree/main/example_configs

MitchBradley, 2023-05-02:
If you have dual steppers and drivers on Y, you need both motor0 and motor1 sections under the y: axis section, per the examples.  Each such motor section needs a separate limit switch for squaring.

jappiemike, 2023-05-02:
I did look at the examples but there was no clear guidance for example BlackBox_X32_XYYZ-DOOR.yaml has 2 Y motors but appears to have no limit switches where as 6_pack_TMC5160_XYYZ.yaml appears to use dual switched to 1 pin. I'm just looking for some clarification. Not being rude but I have to go to work now so it'll be a while before I reply. Thanks for the answers so far.

jappiemike, 2023-05-05:
OK, thanks for the answers, once I've fitted the extra proximity switch I'll change the config to enable it.
Cheers
Mike

closed 2023-05-05 by jappiemike