bdring/FluidNC #890: Problem: Error21

small66666, 2023-05-09:
### Controller Board

MKS DLC32 2.1

### Machine Description

3-осевой лазерный станок ЧПУ с  драйверами TMC2209

### Configuration file

```yaml
$0=7 (Step pulse time)
$1=255 (Step idle delay)
$2=0 (Step pulse invert)
$3=0 (Step direction invert)
$4=0 (Invert step enable pin)
$5=1 (Invert limit pins)
$6=0 (Invert probe pin)
$10=1 (Status report options)
$11=0.010 (Junction deviation)
$12=0.002 (Arc tolerance)
$13=0 (Report in inches)
$20=1 (Soft limits enable)
$21=1 (Hard limits enable)
$22=1 (Homing cycle enable)
$23=3 (Homing direction invert)
$24=300.000 (Homing locate feed rate)
$25=1000.000 (Homing search seek rate)
$26=25.000 (Homing switch debounce delay)
$27=2.000 (Homing switch pull-off distance)
$28=1000.000 (Spindle frequency)
$30=1000.000 (Maximum spindle speed)
$31=0.000 (Minimum spindle speed)
$100=1600.000 (X-axis travel resolution)
$101=1600.000 (Y-axis travel resolution)
$102=80.000 (Z-axis travel resolution)
$110=2100.000 (X-axis maximum rate)
$111=2100.000 (Y-axis maximum rate)
$112=6000.000 (Z-axis maximum rate)
$120=100.000 (X-axis acceleration)
$121=100.000 (Y-axis acceleration)
$122=500.000 (Z-axis acceleration)
$130=410.000 (X-axis maximum travel)
$131=282.000 (Y-axis maximum travel)
$132=80.000 (Z-axis maximum travel)
```

### Startup Messages

```text
Ожидание соединения...
ok
[PRODUCER:MKS DLC32]
[V2.30(8M.H35.20221223)]
[VER:1.1h.2022122301:]
[OPT:VMPH,63,500]
Target buffer size found
[MSG:Using machine:MKS DLC32]
[MSG:Mode=AP:SSDI=ESPWIFI_16431:IP=192.168.4.1:MAC=C4-DE-E2-37-2F-41]
ok
```

### User Interface Software

LightBurn

### What happened?

Если использовать Wi-Fi соединение и в настройках программы ( Режим передачи - Буферизованный ) то
Layer 1
error:21
В блоке обнаружено более одной команды g-кода из одной модальной группы.
На или около линии 28:
Задание остановлено
Поток завершен в 0:00
При подключении через USB то нет этой ошибки.

Babaika1987, 2023-07-14:
Сколько нужно пожертвование для экрана

bdring, 2024-12-12:
Closing old issues

closed 2024-12-12 by bdring