bdring/FluidNC #897: Problem:

April1004, 2023-05-16:
### Controller Board

ESP32-WROOM-32

### Machine Description

ESP32-WROOM-32 with FluidNC on it. Connected via USB to a Linx laptop. Not connected to any other hardware than jumper wires.

### Configuration file

```yaml
control:
  safety_door_pin: NO_PIN
  feed_hold_pin: NO_PIN
  reset_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: gpio.5:low:pu
  macro1_pin: gpio.18:low:pu
  macro2_pin: gpio.19:low:pu
  macro3_pin: gpio.21:low:pu

macros:
  startup_line0:
  startup_line1:
  macro0:
  macro1:f>
  macro2:#f<
  macro3:
```

### Startup Messages

```text
MSG:INFO: FluidNC v3.6.5]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine Eriks cnc]
[MSG:INFO: Board Eriks cnc]
[MSG:INFO: SPI not defined]
[MSG:INFO: No SD Card CS Pin]
[MSG:INFO: See http://wiki.fluidnc.com/en/config/sd_card#sdfallbackcs-access-sd-without-a-config-file]
[MSG:INFO: Stepping:RMT Pulse:10us Dsbl Delay:0us Dir Delay:10us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (10.000,1010.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.12:low Dir:gpio.26:low Disable:NO_PIN]
[MSG:INFO:  X All Limit gpio.0:pu]
[MSG:INFO: Axis Y (10.000,1010.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.14:low Dir:gpio.25:low Disable:NO_PIN]
[MSG:INFO:  Y All Limit gpio.2:pu]
[MSG:INFO: Axis Z (-990.000,10.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.27:low Dir:gpio.33:low Disable:NO_PIN]
[MSG:INFO: macro0_pin gpio.5:low:pu]
[MSG:INFO: macro1_pin gpio.18:low:pu]
[MSG:INFO: macro2_pin gpio.19:low:pu]
[MSG:INFO: macro3_pin gpio.21:low:pu]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: Probe Pin: gpio.4:low:pu]
[MSG:INFO: WiFi is disabled]
ok
```

### User Interface Software

FluidTerm

### What happened?

First of all:
Thank you for this awesome FluidNC-Project :).

I would like to adjust the feedrate while running a job with hardware buttons.

I shorted the configured macro pins to GND for a second. One after another.

I do that while there is a Gcode command running. For example: G1X1000

On shorting a macro-pin to GND I get:
_[MSG:ERR: Macro can only be used in idle state]_
No matter which macro pin I use.

The Feedrate is not overridden as indicated by:
<Run|MPos:-179.161,0.000,0.000|FS:100,0|Pn:XY|Ov:100,100,100>

How is a real-time command macro supposed to look in the config file?
It would be nice to have an example in http://wiki.fluidnc.com/config/macros#config-example

Or am I facing another problem here?

Many thanks in advance

bdring, 2023-05-16:
This is a complicated problem to solve

We are working on major changes that address what is allowed and who is in control. That might simplify this issue.

It will take sevel weeks to see where this goes. Please consider donating to support this work.

April1004, 2023-05-16:
So, did I understand the following correctly?
The feature "real-time command macro feature" as described in http://wiki.fluidnc.com/config/macros#using-realtime-commands-in-macros is not yet properly implemented.

Best Regards Erik

April1004, 2023-05-16:
Okay, I found out, that my gcode sender UGS (universal gcode sender) allows to override feed rates during running jobs. That works for me.

Thanks,
Erik

MitchBradley, 2023-05-16:
The inclusion of realtime characters in macros was probably a mistake.  It was requested by a user who wanted to issue coolant toggles from macros.  I added a general solution without thinking through all of the problems it causes.  See https://github.com/bdring/FluidNC/pull/774 for a discussion of an alternate approach to coolant control.

Since there is often a shortage of GPIO inputs due to needing them for limits, we have not spent of lot of time improving the usefulness of GPIOs as UI controls.  Instead, we encourage people to make "pendants" out of separate microprocessors, than can send UI commands over, for example, websockets.  You can do that very cheaply with, for example, an ESP32-S2 that costs about $2, running a simple python program.  The #displays_pendants Discord channel has discussions of what people are doing in that area.

closed 2024-12-12 by bdring