bdring/FluidNC #898: Unable to connect with Lightburn GRBL (1.1f or later)

Murugesh-Hobbyist, 2023-05-20:
### Controller Board

ESP32 Dev Module
[image]

### Machine Description

Neema-23 with DM556T

### Configuration file

```yaml
name: "ESP32 Dev Controller V4"
board: "ESP32 Dev Controller V4"

stepping:
  engine: RMT
  idle_ms: 0
  dir_delay_us: 1
  pulse_us: 4
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: gpio.13:high

  x:
    steps_per_mm: 800
    max_rate_mm_per_min: 2000
    acceleration_mm_per_sec2: 25
    max_travel_mm: 220
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 10
      seek_mm_per_min: 500
      feed_mm_per_min: 300
      settle_ms: 250
      seek_scaler: 1.5
      feed_scaler: 1.5
    
    motor0:
      limit_all_pin: gpio.17:low:pu
      pulloff_mm: 10.000
      stepstick:
        direction_pin: gpio.14
        step_pin: gpio.12
    motor1:
      null_motor:

  y:
    steps_per_mm: 800
    max_rate_mm_per_min: 2000
    acceleration_mm_per_sec2: 25
    max_travel_mm: 220
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 10
      seek_mm_per_min: 500
      feed_mm_per_min: 300
      settle_ms: 250
      seek_scaler: 1.5
      feed_scaler: 1.5

    motor0:
      limit_all_pin: gpio.4:low:pu
      pulloff_mm: 10.000
      stepstick:
        direction_pin: gpio.15
        step_pin: gpio.26
    motor1:
      null_motor:

  z:
    steps_per_mm: 80
    max_rate_mm_per_min: 2000
    acceleration_mm_per_sec2: 200
    max_travel_mm: 220
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 10
      seek_mm_per_min: 500
      feed_mm_per_min: 300
      settle_ms: 250
      seek_scaler: 1.5
      feed_scaler: 1.5

    motor0:
      limit_all_pin: gpio.16:low:pu
      pulloff_mm: 10.000
      stepstick:
        direction_pin: gpio.33
        step_pin: gpio.27
    motor1:
      null_motor:

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN

coolant:
  flood_pin: gpio.25
  mist_pin:  gpio.21

        
probe:
  pin: gpio.32:low:pu

PWM:
  pwm_hz: 5000
  output_pin: gpio.2
  enable_pin: gpio.22
  direction_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 0
  speed_map: 0=0% 255=100%
```

### Startup Messages

```text
Waiting for connection...
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
Grbl 3.7 [FluidNC v3.7.0 (wifi) '$' for help]
[MSG:INFO: '$H'|'$X' to unlock]
[MSG:DBG:  X All Limit 0]
[MSG:DBG:  Z All Limit 0]
[MSG:DBG:  Y All Limit 0]
G0
G0
G0
G0
G0
G0
G0
G0
G0
G0
G0
G0
```

### User Interface Software

Light Burn

### What happened?

Hi Team,

in doument -> [Github_issue.docx](https://github.com/bdring/FluidNC/files/11522307/Github_issue.docx)

I have installed Fluid NC in ESP32 board pictured below.
[image]

Uploaded the mentioned yaml file.
In Light Burn, we have 3 options of GRBL.
[image]

Among them, GRBL-M3 (1.1e or earlier) works fine.
Below are the screen shots:
[image]

[image]

But as am trying to build laser machine, I want to achieve a laser mode functionality of M4 Dynamic Laser Power Mode which is achievable in GRBL (1.1f or later). 
[https://docs.lightburnsoftware.com/CommonGrblSetups.html#grbl-flavors]

(https://docs.lightburnsoftware.c
[image]
om/CommonGrblSetups.html%23grbl-flavors)
 

https://github.com/gnea/grbl/blob/master/doc/markdown/laser_mode.md
[image]

So I tried using the GRBL (1.1f or later).
[image]

But am unable to connect with the Fluid NC. Loads firmware and start generating G0 command irrespective of any inputs.
[image]

Please help me to connect with GRBL (1.1f or later) in Light Burn with this ESP32 - Fluid NC.

bdring, 2023-05-20:
Have you tried a custom start message?

http://wiki.fluidnc.com/en/features/grbl_compatibility#custom-start-message

Murugesh-Hobbyist, 2023-05-20:
[image]

This is my current custom start message.
Please let me know what to try in that field.

bdring, 2023-05-20:
How about the one shown on the wiki

Start/Message=Grbl 1.1g [\H]

Please consider donating to this project!

Murugesh-Hobbyist, 2023-05-20:
Tried now Bart, still same problem.
[image]

[image]

Even no responce if comand is entered.

bdring, 2023-05-20:
Show me the console with the command being sent.

Cut and paste the text here please

Murugesh-Hobbyist, 2023-05-20:
I tried with basic command like,
$$
$HX
$HY
$HZ
G0 X20
but no response. It keeps on scrolling with G0.

[image]

bdring, 2023-05-20:
Sorry, but I cannot afford to buy Lightburn. I know other people have used it with great success.

What do the lightburn people say?

Murugesh-Hobbyist, 2023-05-20:
Opened an issue with lightburn but no response yet.
If possible can you please check with 30-Days-trial version of light burn?
I'm affride whether my board doesn't supports GRBL (1.1f or later).

bdring, 2023-05-20:
Are you planning on donating to our project?

MitchBradley, 2023-05-21:
I tried with LightBurn 1.2.04, controller type GRBL, and could not duplicate your problem.  I used FluidNC 3.7.0 with $start/message=Grbl 1.1g [\H].  Everything appeared to work, and I never saw any instance of repeated G0 scrolling.

I also tried with controller type GRBL-M3, with identical results.

FluidNC definitely support Grbl 1.1f + M4 laser mode.  I use it on my laser engraver.

MitchBradley, 2023-05-21:
I also tried with the standard startup message `Grbl \V [FluidNC (\R) \H]` and that worked too.

Murugesh-Hobbyist, 2023-05-22:
Hi Bart/Mitch Appreciate your response. Currently im studying. believe i can sponsor when i initiate a income with it.
Hi Mitch, I tried using a new board of same type. now it is loading correctly but the thing is, it is not taking the command M4.
I'm getting,

M4
[MSG:DBG: Bad GCode: M4]
error:20
Unsupported or invalid g-code command found in block.

M4 is the reason that i was moved from GRBL 1.1earlier. Please help me to resolve it.
Whatever the drawing we generate GCODE via Lightburn GRBL (1.1f or later), is having M4 command.

Please have the screenshot and console result.

[image]

Waiting for connection...
Waiting for connection...
[VER:3.7 FluidNC v3.7.0:]
[OPT:MPHS]
[MSG: Machine: ESP32 Dev Controller V4]
[MSG: Mode=AP:SSID=FluidNC:IP=192.168.0.1:MAC=B8-D6-1A-42-AC-ED]
ok
Starting stream
$10=1
Stream completed in 0:00
M4
error:9
G-code locked out during alarm or jog state.
Starting stream
[MSG:DBG: Homing Cycle X]
[MSG:DBG: Homing nextPhase FastApproach]
[MSG:DBG: Starting from 0.000,0.000,0.000]
[MSG:DBG: Planned move to -330.000,0.000,0.000 @ 500.000]
[MSG:DBG:  X All Limit 1]
[MSG:DBG: Homing limited X]
[MSG:DBG: Homing nextPhase Pulloff0]
[MSG:DBG: Starting from -23.433,0.000,0.000]
[MSG:DBG: Planned move to -13.433,0.000,0.000 @ 300.000]
[MSG:DBG:  X All Limit 0]
[MSG:DBG:  X All Limit 1]
[MSG:DBG:  X All Limit 0]
[MSG:DBG: CycleStop Pulloff0]
[MSG:DBG: Homing nextPhase SlowApproach]
[MSG:DBG: Starting from -13.432,0.000,0.000]
[MSG:DBG: Planned move to -28.432,0.000,0.000 @ 300.000]
[MSG:DBG:  X All Limit 1]
[MSG:DBG: Homing limited X]
[MSG:DBG: Homing nextPhase Pulloff1]
[MSG:DBG: Starting from -15.460,0.000,0.000]
[MSG:DBG: Planned move to -5.460,0.000,0.000 @ 300.000]
[MSG:DBG:  X All Limit 0]
[MSG:DBG:  X All Limit 1]
[MSG:DBG:  X All Limit 0]
[MSG:DBG:  X All Limit 1]
[MSG:DBG:  X All Limit 0]
[MSG:DBG: CycleStop Pulloff1]
[MSG:DBG: Homing nextPhase Pulloff2]
[MSG:DBG: mpos was -5.460,0.000,0.000]
[MSG:DBG: mpos becomes 10.000,0.000,0.000]
[MSG:DBG: mpos transformed 10.000,0.000,0.000]
[MSG:DBG: Homing done]
Stream completed in 0:08
M4
[MSG:DBG: Bad GCode: M4]
error:20
Unsupported or invalid g-code command found in block.

bdring, 2023-05-22:
You need to define the spindle as a laser type, not PWM. M4 only works with lasers.

Avataar120, 2023-05-22:
Did you activate DTR in Lightburn device settings?

Murugesh-Hobbyist, 2023-05-22:
Bart, can you please correct my config file with respect to laser? I couldn't get it.

[Laserx.txt](https://github.com/bdring/FluidNC/files/11526066/Laserx.txt)

Yes Avataar, Enabled them already. Below is my LightBurn device settings.

[image]

bdring, 2023-05-22:
I can do it later this week. I am quite busy now. The wiki has all the info if you want to do it yourself.

Murugesh-Hobbyist, 2023-05-22:
Sure Brat. I will check for myself and update you.

Avataar120, 2023-05-22:
I don't see anything wrong in your config file.
I can confirm that connection with Lightburn is working like a charm on the controller I have developed for CO2 laser machines (https://lasercutting.avataar120.com/en/2023/04/14/plug-and-play-fluidnc-k40-shield-for-lightburn/)
I sometimes got G0 answers when ESP32 is rebooting in loop (some bad chips have stability issues)

Did you try with Fluidterm ?
If working well with Fluidterm, try to change the ESP32 for another one

Murugesh-Hobbyist, 2023-05-22:
No Avataar. As Brat suggested, below config solved M4 command problem.
Laser:
  pwm_hz: 5000
  output_pin: gpio.2
  enable_pin: gpio.22
  disable_with_s0: false
  s0_with_disable: true
  tool_num: 0
  speed_map: 0=0.000% 255=100.000%
  off_on_alarm: true

While im trying to engrave an image, I'm getting error for [MSG:DBG: Bad GCode: G1X0.2F6000S72.7:73]
error:1
G-code words consist of a letter and a value. Letter was not found.
On or near line 8:
Job halted

Below are the errors in console, engrave image and settings.

[image]

Avataar120, 2023-05-22:
So this is another issue 😀
Did you select GRBL when installing Lightburn ?
Lightburn is generating GCODE according to this initial selection.

You can have a look to my post in order to setup Lightburn correctly for FLUIDNC :
https://lasercutting.avataar120.com/en/2022/09/03/installing-lightburn-and-setting-it-up-in-3-steps/

Follow the instructions step by step send you'll have everything setup including M4 command

Murugesh-Hobbyist, 2023-05-22:
Yeah Avataar 😅 its another issue.
Just now gone through your link. Everything is fine except constant power mode.
I was tunned off the constant power mode as i refered to LightBurn forum.

[image]

Bellow is my current Layer parameters

[image]

So I'm confused here. whats the purpose of constant power mode 🤔

Avataar120, 2023-05-22:
You can change to non constant power mode, no problem.
But you'll have to fine tune the pwm/% map

Murugesh-Hobbyist, 2023-05-22:
Alright Avataar, As i couldn't identify the error, I reinstalled the LightBurn and setup the same again.
Now the motor spins with laser enable disable without stoping at burn points.
Seems like M4 Dynamic Laser Power Mode is achieved.
Will install this in the machine and update. Thanks.

robZeilinga, 2023-08-23:
@Murugesh-Hobbyist : did you solve your Problem?

Murugesh-Hobbyist, 2023-08-24:
I'm glad you're tracking me.
After few rework, It's resolved.

Thanks all for the support. 
Closing the issue.

closed 2023-08-24 by Murugesh-Hobbyist