bdring/FluidNC #901: Problem: Large GCode file run from SD card makes FluidNC Hang/Crash

jayde-d, 2023-05-21:
### Controller Board

MKS DLC32
FluidNC V. 3.7.0

### Machine Description

OpenBuilds Acra System with Servo Z Axis.

### Configuration file

```yaml
board: MKS-DLC32
name: K40 MOD
meta: 2022-12-27 by Tong

kinematics:
  Cartesian:

stepping:
  engine: I2S_STATIC
#Static only, Stream Produces a second "ghost line" when doing engraving/Filling
  idle_ms: 254
  pulse_us: 6
  dir_delay_us: 10
  disable_delay_us: 0
axes:
  shared_stepper_disable_pin: I2SO.0
  x:
    steps_per_mm: 112.500
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 1000.000
    max_travel_mm: 313.000
    soft_limits: false
    homing:
      cycle: -1
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 6000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.36
      hard_limits: false
      pulloff_mm: 1.000
      stepstick:
        step_pin: I2SO.1
        direction_pin: I2SO.2:high

  y:
    steps_per_mm: 112.500
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 1000.000
    max_travel_mm: 230.000
    soft_limits: false
    homing:
      cycle: -1
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 6000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.35
      hard_limits: false
      pulloff_mm: 1.000
      stepstick:
        step_pin: I2SO.5
        direction_pin: I2SO.6:high

  z:
    steps_per_mm: 100.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 5.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 5.000

    motor0:
      rc_servo:
        pwm_hz: 55
        output_pin: gpio.22
        min_pulse_us: 1000
        max_pulse_us: 1250
        

i2so:
  bck_pin: gpio.16
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.12
  mosi_pin: gpio.13
  sck_pin: gpio.14

sdcard:
  cs_pin: gpio.15
  card_detect_pin: gpio.39
  frequency_hz: 10000000

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: gpio.33:low:pu
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

macros:
  startup_line0:
  startup_line1:
  macro0: G90&G53G0Z5.0
  macro1: 
  macro2:
  macro3:

coolant:
  flood_pin: NO_PIN
  mist_pin: NO_PIN
  delay_ms: 0

probe:
  pin: NO_PIN
  check_mode_start: false

Laser:
  pwm_hz: 5000
#For software PWM control on K40, IN on TTL connection, G next to IN to G on TTL. No need for Enable
  output_pin: gpio.32
  enable_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: false
  tool_num: 0
  speed_map: 0=7.500% 2200=100.000%
# 165=1mA (not enough to fire), 880=9mA 2200=16mA
# Set your own MAX and Minimum,
# Change max until desired MAX mA on gauge
# Change min until laser just before laser fires.

user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN

start:
  must_home: false

# 5,18,19,22,23,25,26,27,32,33,39,I2SO.7
# SDA 0 / SCL 4
```

### Startup Messages

```text
before the [MSG:INFO: FluidNC v3.7.0] I get this:

E (601) esp_core_dump_flash: Noÿ¼Kòüdump partition found!

Happened after I wiped and reloaded FluidNC from my PC to upgrade from 3.6.8
```

### User Interface Software

WebUI, FluidTerm

### What happened?

When I run a large file (230kb gcode) from my SD Card, after a certain amount of commands, the entire board freezes up. WebUI thinks that it's lost connection, and FluidTerm is unresponsive to anything besides quitting. Attached the file that causes the issue.
[patch_chao.gcode.txt](https://github.com/bdring/FluidNC/files/11524991/patch_chao.gcode.txt)

angelmoi, 2023-07-08:
which firmware version are you using??

I have tested with gcode > 1 megabyte and  it works perfectly

angelmoi, 2023-07-08:
I am using 3.7.1 firmware  version

jayde-d, 2023-07-08:
Sorry for the late reply. i don't know what happened when i was running
this then (i was working on a very tight time restriction) but it stopped
happening and coulsnt be reproduced. i ran a like 2 or 3 MB file that took
several hours when i fixed the problem by reflashing FluidNC onto the ESP
module and it went away.

angelmoi, 2023-07-08:
Ah, ok you mean everything is ok right now?

Angel

closed 2023-07-08 by MitchBradley