bdring/FluidNC #902: Problem:

txg5230, 2023-05-22:
### Controller Board

4-pack

### Machine Description

[16:23:40.277]收←◆E (602) esp_core_dump_flash: No\0槵K晛dump partition found!
E (602) esp_core_dump_flash: No core dump partition found!

[16:23:40.324]收←◆
[MSG:INFO: FluidNC v3.7.0]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:4595.yaml]
[MSG:WARN: Ignored key frequency_hz]
[MSG:INFO: Machine 6 Pack StepStick XYZABC]
[MSG:INFO: Board 6 Pack]
[MSG:INFO: I2SO BCK:gpio.17 WS:gpio.16 DATA:gpio.4]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:8000000]
[MSG:INFO: Stepping:I2S_stream Pulse:10us Dsbl Delay:0us Dir Delay:1us Idle Delay:250ms]
[MSG:INFO: Axis count 4]
[MSG:INFO: Axis X (150.000,1400.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.6 Dir:I2SO.5 Disable:NO_PIN]
[MSG:INFO:  X Neg Limit gpio.14:low:pu]
[MSG:INFO: Axis Y (150.000,2600.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.4 Dir:I2SO.3 Disable:NO_PIN]
[MSG:INFO:  Y Neg Limit gpio.12:low:pu]
[MSG:INFO: Axis Z (-150.000,150.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.2 Dir:I2SO.1 Disable:NO_PIN]
[MSG:INFO:  Z Neg Limit gpio.13:low:pu]
[MSG:INFO: Axis A (150.000,1110.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.0 Dir:I2SO.7 Disable:NO_PIN]
[MSG:INFO:  A Neg Limit gpio.15:low:pu]
[MSG:INFO: safety_door_pin gpio.33:low:pu]
[MSG:INFO: feed_hold_pin gpio.25:low:pu]
[MSG:INFO: cycle_start_pin gpio.26:low:pu]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: PWM Spindle Ena:gpio.32 Out:gpio.2 Dir:NO_PIN Freq:5000Hz Period:8191]
[MSG:INFO: Using spindle PWM]
[MSG:INFO: Probe Pin: gpio.27:low:pu]

[16:23:40.603]收←◆[MSG:INFO: Connecting to STA SSID:yygg]
[MSG:INFO: Connecting.]

[16:23:42.603]收←◆[MSG:INFO: Connecting..]

[16:23:44.603]收←◆[MSG:INFO: Connecting...]

[16:23:46.605]收←◆[MSG:INFO: Connected - IP is 192
[16:23:46.636]收←◆.168.2.2]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.7 [FluidNC v3.7.0 (wifi) '$' for help]

### Input Circuits

```yaml
[16:23:40.277]收←◆E (602) esp_core_dump_flash: No\0槵K晛dump partition found!
E (602) esp_core_dump_flash: No core dump partition found!

[16:23:40.324]收←◆
[... 283 lines cut ...]
[MSG:INFO: PWM Spindle Ena:gpio.32 Out:gpio.2 Dir:NO_PIN Freq:5000Hz Period:8191]
[MSG:INFO: Using spindle PWM]
[MSG:INFO: Probe Pin: gpio.27:low:pu]

[16:23:40.603]收←◆[MSG:INFO: Connecting to STA SSID:yygg]
[MSG:INFO: Connecting.]

[16:23:42.603]收←◆[MSG:INFO: Connecting..]

[16:23:44.603]收←◆[MSG:INFO: Connecting...]

[16:23:46.605]收←◆[MSG:INFO: Connected - IP is 192
[16:23:46.636]收←◆.168.2.2]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.7 [FluidNC v3.7.0 (wifi) '$' for help]
```

### What happened?

Send 85, cancel the position loss, as a result of jogging occasionally stop and normal should be slow down

bdring, 2023-05-22:
I don't understand your problem. Please explain with much more details. Show the problem in FluidTerm if you can.

There is one error in your config file. It should not be creating any errors, but it is best to fix it.

```yaml
sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5
frequency_hz: 1000000
```

should be 

```yaml
sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5
  frequency_hz: 1000000
```

txg5230, 2023-05-23:
Send 85, stop immediately. occasional

bdring, 2023-05-23:
0x85 stop jog

txg5230, 2023-05-24:
High speed motor power outage immediately, normal should slow down. occasional

bdring, 2023-05-24:
Show the commands you are sending. I do not understand what you are saying.

Show a picture of your controller and machine.

What is your native language and country?

txg5230, 2023-05-28:
中国，慢跑速度F15000，发送85，正常应该减速停止。但是偶尔会立刻停止导致坐标丢失。就想直接给机器断电一样

closed 2024-12-19 by bdring