bdring/FluidNC #904: Problem: Can't upload file gcode on SD card by Wi-fi

DamRos00, 2023-05-26:
### Controller Board

MKSDLC32V2.1

### Machine Description

A simple CNC machine, cartesian 

### Configuration file

```yaml
board: MKS DLC32 2.0
name: machine
meta: (12/2/2021) B. Dring for mks
stepping:
  engine: I2S_STREAM
  idle_ms: 25
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: i2so.0
  x:
    steps_per_mm: 80.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 500.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.36
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm:1.000
      stepstick:
        step_pin:  i2so.1
        direction_pin: I2SO.2

  y:
    steps_per_mm: 80.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 500.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 3
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.35
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm:1.000
      stepstick:
        step_pin: I2SO.5
        direction_pin: I2SO.6

  z:
    steps_per_mm: 80.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 500.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.34
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm:1.000
      stepstick:
        step_pin: I2SO.3
        direction_pin: I2SO.4

i2so:
  bck_pin: gpio.16
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.12
  mosi_pin: gpio.13
  sck_pin: gpio.14

sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.15

probe:
  pin: gpio.22
  check_mode_start: true

start:
  must_home: false

PWM:
  pwm_hz: 5000
  output_pin: gpio.32
  direction_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 1000
  spindown_ms: 1000
  tool_num: 0
  speed_map: 0=0.000% 100=100.000%
```

### Startup Messages

```text
[MSG:INFO: FluidNC v3.6.7]

[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]

[MSG:INFO: Local filesystem type is spiffs]

[MSG:INFO: Configuration file:MKS_DLC32_21_XYZ.yaml]

[MSG:INFO: Machine K40 Laser]

[MSG:INFO: Board MKS-DLC32 V2.1]

[MSG:INFO: I2SO BCK:gpio.16 WS:gpio.17 DATA:gpio.21]

[MSG:INFO: SPI SCK:gpio.14 MOSI:gpio.13 MISO:gpio.12]

[MSG:INFO: SD Card cs_pin:gpio.15 detect:NO_PIN]

[MSG:INFO: Stepping:I2S_static Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:0ms]

[MSG:INFO: Axis count 3]

[MSG:INFO: Shared stepper disable I2SO.0]

[MSG:INFO: Axis X (0.000,325.000)]

[MSG:INFO:   Motor0]

[MSG:INFO:     stepstick Step:I2SO.1 Dir:I2SO.2 Disable:NO_PIN]

[MSG:INFO:  X Neg Limit gpio.36]

[MSG:INFO: Axis Y (0.000,220.000)]

[MSG:INFO:   Motor0]

[MSG:INFO:     stepstick Step:I2SO.5 Dir:I2SO.6:low Disable:NO_PIN]

[MSG:INFO:  Y Neg Limit gpio.35]

[MSG:INFO: Axis Z (0.000,80.000)]

[MSG:INFO:   Motor0]

[MSG:INFO:     stepstick Step:I2SO.3 Dir:I2SO.4 Disable:NO_PIN]

[MSG:INFO:  Z Neg Limit gpio.34]

[MSG:INFO: macro0_pin gpio.33:low:pu]

[MSG:INFO: Kinematic system: Cartesian]

[MSG:INFO: Laser Ena:I2SO.7 Out:gpio.32 Freq:5000Hz Period:8191]

[MSG:INFO: Using spindle Laser]

[MSG:INFO: Probe Pin: gpio.22]

[MSG:INFO: STA SSID is not set]

[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]

[MSG:INFO: AP started]

[MSG:INFO: WiFi on]

[MSG:INFO: Captive Portal Started]

[MSG:INFO: HTTP started on port 80]

[MSG:INFO: Telnet started on port 23]
```

### User Interface Software

ESP3D for FluidNC

### What happened?

I have a mobile application, for Android and iOS, written in Javascript using React Native framework and I am trying to upload my Gcode in the SD card of the machine using formData and blob as in the code below, where in the gcode variable are saved all the commands to be sent to the machine according to the correct syntax. I connect with the machine via WebSocket protocol on port 81:
 
`var target = "http://" + address + "/upload"
var formData = new FormData()
var blob = new Blob([gcode], {type: "text/xml"})
formData.append("path", "/");
formData.append( "/" + "file.gcode" + "S", blob.size)
formData.append("myfile[]", blob, "/" + "file.gcode")`

But when I try to send the formdata to the URL in the target variable, where address is the IP of the machine, using the React Native API, like axios, I get a positive response like this below:

`{"data":{"files":[{"name":"System Volume Information","shortname":"System Volume Information","size":"-1","datetime":""}],"path":"/","total":"970.38 MB","used":"1 B","occupation":"1","mode":"direct","status":"Ok"},"status":200,"headers":{"cache-control":"no-cache","connection":"close","content-length":"205","content-type":"application/json"},"config":{"transitional":{"silentJSONParsing":true,"forcedJSONParsing":true,"clarifyTimeoutError":false},"adapter":["xhr","http"],"transformRequest":[null],"transformResponse":[null],"timeout":0,"xsrfCookieName":"XSRF-TOKEN","xsrfHeaderName":"X-XSRF-TOKEN","maxContentLength":-1,"maxBodyLength":-1,"env":{},"headers":{"Accept":"application/json, text/plain, */*","Content-Type":"multipart/form-data"},"method":"post","url":"http://192.168.3.144/upload","data":{"_parts":[["path","/"],["/file.gcodeS",64594],["myfile[]",{"_data":{"blobId":"efcff72e-5f17-4000-ad9b-ac11bb788a5d","offset":0,"size":64594,"type":"text/xml","__collector":{}}}]]}},"request":{"UNSENT":0,"OPENED":1,"HEADERS_RECEIVED":2,"LOADING":3,"DONE":4,"readyState":4,"status":200,"timeout":0,"withCredentials":true,"upload":{},"_aborted":false,"_hasError":false,"_method":"POST","_perfKey":"network_XMLHttpRequest_http://192.168.3.144/upload","_response":"{\"files\":[{\"name\":\"System Volume Information\",\"shortname\":\"System Volume Information\",\"size\":\"-1\",\"datetime\":\"\"}],\"path\":\"/\",\"total\":\"970.38 MB\",\"used\":\"1 B\",\"occupation\":\"1\",\"mode\":\"direct\",\"status\":\"Ok\"}","_url":"http://192.168.3.144/upload","_timedOut":false,"_trackingName":"unknown","_incrementalEvents":false,"_performanceLogger":{"_timespans":{"network_XMLHttpRequest_http://192.168.3.133:19000/logs":{"startTime":4103458652.916666,"endTime":4103458694.946416,"totalTime":42.02974987030029},"network_XMLHttpRequest_http://192.168.3.144":{"startTime":4103460689.125666,"endTime":4103460763.1425,"totalTime":74.01683378219604},"network_XMLHttpRequest_http://192.168.3.146":{"startTime":4103460689.349041},"network_XMLHttpRequest_http://192.168.3.133:19000/symbolicate":{"startTime":4103465701.732666,"endTime":4103465807.081458,"totalTime":105.34879207611084},"network_XMLHttpRequest":{"startTime":4103467314.014083,"endTime":4103467810.741416,"totalTime":496.72733306884766},"network_XMLHttpRequest":{"startTime":4103470254.807833,"endTime":4103470343.675,"totalTime":88.8671669960022},"network_XMLHttpRequest":{"startTime":4103470357.364625,"endTime":4103470360.714666,"totalTime":3.350040912628174},"network_XMLHttpRequest":{"startTime":4103471028.474208,"endTime":4103471141.830958,"totalTime":113.35675001144409},"network_XMLHttpRequest":{"startTime":4103471143.41425,"endTime":4103471239.577083,"totalTime":96.16283321380615},"network_XMLHttpRequest_http://192.168.3.144/upload":{"startTime":4103471318.551208,"endTime":4103471376.575333,"totalTime":58.02412509918213}},"_extras":{},"_points":{"initializeCore_start":4103458448.035125,"initializeCore_end":4103458472.059458},"_pointExtras":{},"_closed":false},"responseHeaders":{"Content-Type":"application/json","Content-Length":"205","Cache-Control":"no-cache","Connection":"close"},"_requestId":null,"_headers":{"accept":"application/json, text/plain, */*","content-type":"multipart/form-data"},"_responseType":"","_sent":true,"_lowerCaseResponseHeaders":{"content-type":"application/json","content-length":"205","cache-control":"no-cache","connection":"close"},"_subscriptions":[],"responseURL":"http://192.168.3.144/upload"}}`

I have deleted some links from the response that are useless. The problem is that the file was not uploaded. My question then is: as there are there commands to get the file started inside the SD card, such as $SD/RUN=/ from Wi-fi, are there commands to perform the upload? Or I am mistaken something in the call with axios? Because that Javascript script from web application works and uploads the file. Thank you very much for the help!

MitchBradley, 2023-05-26:
File upload is done via POST to port 80 with a url like https://<ip>/upload .  You can use Developer Tools in Chrome to see what the exchange looks like when WebUI does it.

Alex-CodeLab, 2023-05-27:
probably blocking because CORS

DamRos00, 2023-05-29:
I tried this code here to perform the upload operation, where the target variable is the URL "http://" + machine address + "upload" and the formData variable is the file I want to upload:

`axios({
      method: "post",
      url: target,
      data: formData,
      headers: { "Content-Type": "multipart/form-data" },
    })
      .then(function (response) {
        //handle success
        console.log("SUCCESS: " + JSON.stringify(response));
      })
      .catch(function (response) {
        //handle error
        console.log("ERROR: " + JSON.stringify(response));
      });`

And testing it from the React Native browser emulator, running on http://localhost:19006, the file is correctly uploaded despite the Cors policy error is fired, as the screenshots below show.

[image]
**CORS error**

[image]
**Formdata I am sending**

[image]
**File uploaded**

 Whereas, testing this same code from the mobile emulator, in this case for iOS, running on exp://192.168.3.133:19000, the then callback is called, which means the call was successful, whose response you can see on my first message, but the file was not uploaded. How is this possible?

Alex-CodeLab, 2023-05-29:
"which means the call was successful,"

the webserver is very limited, and sometimes gives incorrect return values, so that can not be trusted.
(for example, `200/ok` when it should return `404/notfound` )

MitchBradley, 2023-05-29:
Did you try the developer tools thing I suggested?  Can you run you program on Linux or some other system that will let you capture packets with Wireshark or some other sniffer?

GavinLi2016, 2024-08-03:
I meet the same problem in Andriod Phone. I use XMLHttpRequest object in webview to send file. It is ok for small file,but failed for large file (about 7M ). and it is ok for PC chrome.