bdring/FluidNC #905: CNC lathe and milling machine

Mixa778, 2023-05-31:
### Machine Context

CNC lathe and milling machine

### Feature Description

Threading capabilities with G-code G33 threading CNC lathe

    Longitudinal carving.
    Cross thread.
    Tapered thread.
    Single and multi-start threads with constant pitch.
    Variable pitch and taper angle due to chain of thread-cutting blocks.

Drilling cycle G73, G74, G76, G80, G81, G82, G83, G84, G85, G86, G87, G88, G89, G98, G99
What is a 'plunge drilling cycle'?

You will often hear the term 'plunge pitch'. It is the practice of drilling a short distance (plunge step), stepping back some distance, and then going down to take another step. This is necessary in order to remove chips from the hole.

Recutting chips is always bad for tool life. Another purpose of plunge drilling is to improve chip evacuation. There is only enough room in the hole for the chips that enter the flutes of the twist drill, and the deeper the hole, the more difficult it is to remove the chips from the hole. Plunge drilling or 'deep drilling' also helps to remove chips in two ways.

First, even a very short pitch with minimal retraction helps break chips, resulting in shorter chips. Shorter chips are much easier to remove.

Second, if the drill enters the hole a significant distance, it helps to reduce the distance the drill helix has to carry the chips.

One thing that is important to avoid when deep drilling is to allow coolant or air/mist to wash the chips back into the hole. For this reason, with optimal core drilling cycles, the twist drill will not completely withdraw from the hole.

Another thing to keep in mind is that most manufacturers do not recommend this drilling cycle for carbide drills. This increases the chipping tendency of the brittle carbide.

There are a few rules of thumb as to when you need to start using a drill-by-drill cycle rather than just diving straight down. Most tool manufacturers suggest starting when the hole is 4 diameters deep.
Drilling cycle G code

As you can see, cycles can be divided according to their purpose - drilling, boring or tapping, whether they are plunge cycles, how they retract, and anything special that happens at the bottom of the hole. For example, the delay helps to ensure a smooth hole bottom and remove chips from the bottom of the hole. Chips trapped between the tip of the drill and the bottom of the hole when running the drill for the next step will greatly increase tool wear, especially with materials such as stainless steel.
Changing retraction with G codes G98 and G99

G codes G98 and G99 are used to change the retraction mode of drilling canned cycles. If G98 is in effect (given before the cycle, for example G99 shown above), the retraction is returned to the original Z height. obstacles such as clamps or other parts of the part.
Multiple holes until G80 cancels cycle

As already mentioned, these drilling cycles are modal. This means that you can simply change a few XY coordinates after running the loop and the machine will happily loop at each location. Use G80 to cancel the cycle. After executing G80, the machine returns to G00 mode.

In the example above, we get 3 holes before G80 cancels the canned cycle.
The G95 command is used to specify the feed in mm/rev (inch/revolution) format.

If the CNC unit is set to inches, the feedrate is expressed in inches per revolution, when using millimeters as the unit, the feedrate is expressed in millimeters per revolution.

### Other Approaches

I would like to have full-fledged commands for writing programs manually, and if there is such an opportunity, it will be possible to use it in the school curriculum and hobbies in order to learn how to program a machine, this is very useful for developing thinking and imagination.

### How I Can Help

Financially ready to help.

bdring, 2023-05-31:
We currently don't support encoders, so adding lathe features that require that would be a big project. It is also hard to develop such things when the primary developers don't have the expensive hardware required to test these features. People interested in lathe features is a tiny minority of our users. There are not a lot of good, low cost, DIY lathe options. 

The drilling features are typically handled by CAM software these days. Most people design parts in 3D parametric software and go directly to CAM. While it would be a bit time consuming to add drilling features, it is probably not too complex or likely to impact the stability of the firmware. We would be happy to consider a PR.

Bottom line: I am not ruing out either of these suggestions, but they are not a priority at this time.

jmis-soliditysoftware, 2023-06-23:
This is implemented in GRBL-L-Mega.  Here's their documentation on how it works:

https://github.com/MetalWorkerTools/grbl-L-Mega/wiki/How-it-works

MitchBradley, 2023-06-23:
Sounds like an argument for using GRBL-L-Mega for that use case.  We do not have the time or resources to cover every possible CNC usage.  We are already tapped-out trying to support the range of use cases that we do cover.

HuubBuis, 2023-07-26:
I developed the threading implementation for the  [Grbl-L](https://github.com/MetalWorkerTools/grbl-L]grbl-L) and [Grbl-L-Mega](https://github.com/MetalWorkerTools/grbl-L-Mega) versions. Doing it for the ESP32 processor is also on my ToDo list. I am not sure if it is feasible,  but I will give it a try. Currently my time is occupied by a few other projects but I hope to get started end next year using FluidNC as base.

DedArchi, 2023-09-13:
At least G81/G80 for drilling PCB.