bdring/FluidNC #906: Feature: Soft UART via a single bidirectional GPIO pin

geoffsim, 2023-05-31:
### Machine Context

TMC22XX UART Support

### Feature Description

The implementation of a soft UART operating via a single bidirectional GPIO pin would simplify the support of TMC22XX stepper drivers. Both Klipper (and Marlin) implement this type of UART.

### Other Approaches

Currently, each UART requires a TX and RX pin wired via a 1K resistor to the single bi-directional PDN_UART pin on the stepper driver. Wiring multiple drivers together is possible by using different addresses via MS1/MS2. 

Adding a bi-directional feature will remove the need for the complex wiring, and resolve the issue with TMC2209's (an on-board pull-down resistor causes problem when multiple drivers are wired in parallel).

### How I Can Help

I can test any changes on an MKS TinyBee, a WROOM ESP32 system, and a D1 R32 system.

bdring, 2023-05-31:
Currently we are working on other features. We can add that to the list of things we might consider in the future. 

It is unlikely to be a high priority, because we already have a few options for people using TMC2209 drivers.

closed 2024-02-03 by bdring

040medien, 2024-02-18:
@bdring I noticed you closed this as completed. Will this be part of an upcoming release?