bdring/FluidNC #910: Problem: When running Gcode all moves xyz continue even if the RS485 spindle is reported as unresponsive(e.g. if VFD switched off or not connected). When connected the spindle speed is controllable

JohnCMBC, 2023-06-03:
### Controller Board

MKS DLC32 V2.1

### Machine Description

Gantry router, XYYZ, external drivers DQ542MA ,Huanyang spindle - RS485 controlled using generic 485 board connected to display headers on DLC, XYZ proximity endstops and probe via 12/3.3v opto-isolator

### Configuration file

```yaml
board: MKS-DLC32 V2.1
name: Johns CNC
meta: (17-05-2023) by AJM

verbose_errors: true

kinematics:
  Cartesian:

stepping:
  engine: I2S_STATIC
  idle_ms: 250
  pulse_us: 5
  dir_delay_us: 5
  disable_delay_us: 0
axes:
  shared_stepper_disable_pin: I2SO.0
  
  x:
    steps_per_mm: 800
    max_rate_mm_per_min: 2900
    acceleration_mm_per_sec2: 500.000
    max_travel_mm: 750.000
    soft_limits: true
    homing:
      feed_mm_per_min: 100
      seek_mm_per_min: 2000
      cycle: 2
      mpos_mm: 0.000
      positive_direction: false
    
    motor0:
      limit_neg_pin: gpio.36:low
      pulloff_mm: 2.000
      stepstick:
        step_pin: I2SO.1
        direction_pin: I2SO.2
    motor1:
      null_motor:

  y:
    steps_per_mm: 800
    max_rate_mm_per_min: 2900.000
    acceleration_mm_per_sec2: 500.000
    max_travel_mm: 750.000
    soft_limits: true
    homing:
      feed_mm_per_min: 100
      seek_mm_per_min: 2000
      cycle: 2
      mpos_mm: 0.000
      positive_direction: false
      
    motor0:
      limit_neg_pin: gpio.35:low
      pulloff_mm: 2.000
      stepstick:
        step_pin: I2SO.5
        direction_pin: I2SO.6
    motor1:
      null_motor:

  z:
    steps_per_mm: 800
    max_rate_mm_per_min: 2900.000
    acceleration_mm_per_sec2: 500.000
    max_travel_mm: 150.000
    soft_limits: true
    homing:
      feed_mm_per_min: 100
      seek_mm_per_min: 2000
      cycle: 2
      mpos_mm: 0
      positive_direction: true

    motor0:
      limit_neg_pin: gpio.34:low
      pulloff_mm: 2.000
      stepstick:
        step_pin: I2SO.3
        direction_pin: I2SO.4
    motor1:
      null_motor:
      
i2so:
  bck_pin: gpio.16
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.12
  mosi_pin: gpio.13
  sck_pin: gpio.14

sdcard:
  cs_pin: gpio.15
  card_detect_pin: NO_PIN
  frequency_hz: 1000000

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  
macros:
  startup_line0:
  startup_line1:
  macro0: 
  macro1:
  macro2:
  macro3:
 
coolant:
  flood_pin: NO_PIN
  mist_pin: NO_PIN
  delay_ms: 0

probe:
  pin: gpio.22
  check_mode_start: true
  
Huanyang:
  spinup_ms:10000
  spindown_ms:10000
  uart:
    txd_pin: gpio.25
    rxd_pin: gpio.33:pd
    rts_pin: gpio.26
    baud: 9600
    mode: 8N1
  modbus_id: 1
  tool_num: 0
  off_on_alarm: true
  speed_map: 0=0% 0=25% 6000=25% 24000=100%

start:
  must_home: false
```

### Startup Messages

```text
$SS
[MSG:INFO: FluidNC v3.7.0]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:MKS_DLC32_21_XYZ.yaml]
[MSG:DBG: Running after-parse tasks]
[MSG:DBG: Checking configuration]
[MSG:INFO: Machine Johns CNC]
[MSG:INFO: Board MKS-DLC32 V2.1]
[MSG:INFO: I2SO BCK:gpio.16 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.14 MOSI:gpio.13 MISO:gpio.12]
[MSG:INFO: SD Card cs_pin:gpio.15 detect:NO_PIN freq:1000000]
[MSG:INFO: Stepping:I2S_static Pulse:5us Dsbl Delay:0us Dir Delay:5us Idle Delay:250ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable I2SO.0]
[MSG:INFO: Axis X (0.000,750.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.1 Dir:I2SO.2 Disable:NO_PIN]
[MSG:INFO:  X Neg Limit gpio.36:low]
[MSG:DBG:  X Neg Limit 0]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Y (0.000,750.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.5 Dir:I2SO.6 Disable:NO_PIN]
[MSG:INFO:  Y Neg Limit gpio.35:low]
[MSG:DBG:  Y Neg Limit 1]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Z (-150.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.3 Dir:I2SO.4 Disable:NO_PIN]
[MSG:INFO:  Z Neg Limit gpio.34:low]
[MSG:DBG:  Z Neg Limit 1]
[MSG:INFO:   Motor1]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: UART1 Tx:gpio.25 Rx:gpio.33:pd RTS:gpio.26 Baud:9600]
[MSG:INFO: Huanyang Spindle  Tx:gpio.25 Rx:gpio.33:pd RTS:gpio.26 Baud:9600]
[MSG:INFO: Using spindle Huanyang]
[MSG:INFO: Probe Pin: gpio.22]
[MSG:INFO: Connecting to STA SSID:BlueDragon]
[MSG:INFO: Connecting.]
[MSG:INFO: Huanyang PD005,PD011 Freq range (25,400) Hz (1500,24000) RPM]
[MSG:INFO: Huanyang PD144 Rated RPM @ 50Hz:3000]
[MSG:INFO: Huanyang PD143 Poles:2]
[MSG:INFO: Huanyang PD014 Accel:10.000]
[MSG:INFO: Huanyang PD015 Decel:10.000]
[MSG:DBG: vfd_cmd_task mode:1]
[MSG:INFO: Connecting..]
[MSG:INFO: Connecting...]
[MSG:INFO: Connected - IP is 192.168.0.231]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
ok

and after FluidNC connects:

[MSG:DBG: path /index.html CT text/html]
[MSG:DBG: WebSocket 0 from 192.168.0.202 uri /]
[MSG:INFO: Channel auto report interval set to 50 ms]
[MSG:DBG: path /preferences.json CT text/html]
```

### User Interface Software

FluidNC, OpenBuildsCONTROL, ugsplatform, Candle

### What happened?

Loaded Gcode to any of the user interfaces and after setting up work co-ords, ran the file (tried more than 1).
Expected the router to move to the start location then to the safe distance from the stock, run the spindle up to speed, then proceed to cutting...
What happens is that the job continues (after G4 P10 delay) even if the VFD is unresponsive or disconnected which would result in broken endmills (I haven't actually gone this far - am still testing but have had a similar problem on the same router mechanics controlled by LinuxCNC  which was sorted by configuration/program changes). Please note I only have the hardware for one axis during testing which moves and homes correctly for X,Y and Z when configured for same.
I don't know if this is a config/Gcode/Fluid problem.

Gcode loaded:
(Post-Processor : OpenbuildsFusion360PostGrbl.cps V1.0.32)
(Units = mm)

(Drawing name : pcb trace modded v4)
(Program Name : 1001)

(1 Operation :)
(1 : 2D Contour1)
(  Tool 2: Flat End Mill 2 Flutes, Diam = 0.4mm, Len = 12.00mm)
(  Spindle : RPM = 7277)
(  Machining time : 5 min 32 sec)

G90 G94 G17
G21

(Operation 1 of 1 : 2D Contour1)
G54

G53 G0 Z-10
G0 X46.414 Y27.627
S7277 M3
G4 P10.
G0 X46.414 Y27.627 Z15 F584
Z5
G1 Z1 F195
Z-0.05
X46.416 Y27.646 F584
Y27.648
X46.419 Y27.662
...................................................

MitchBradley, 2023-06-03:
Can you show us the serial console log messages showing that the spindle was unresponsive?  Pick a program that fails and a specific sender and show the full log from that, starting at reboot and continuing through the failure point.  The fact that it fails with multiple senders and multiple gcode programs is interesting, but in order to debug it we need a specific case with all of the information.

JohnCMBC, 2023-06-03:
**Power on DLC32 controller
Run fluidterm and connect to COM3**

Select a COM port
0: COM1 (\Device\Serial0)
1: COM3 (\Device\Serial2)
Choice: 1
FluidNC v1.2.0 using COM3
Exit: Ctrl-C, Ctrl-Q or Ctrl-], Clear screen: CTRL-W
Upload: Ctrl-U, Reset ESP32: Ctrl-R, Send Override: Ctrl-O
ß

**Reset (Ctrl-R)**

Resetting MCU
et.R╒╣☺ 8 &è▓☻é0:▓&º╙SH¿
«.↨é┬┼ ¿⌡]§ION_╥Q¬¶UëI╜╜t:0°L,BÜA%}↓AST▀¶*M!_BO╧U⌐Hh╜╣ÖÑgsi≡'éb☻→A%]P:0°YYCí,δ}ærv:┬┴0¼\E╔┘:0xé,d▀û┘Θ0x00,¼.·æ╔v:░┴0,ï}æ╔┘:0°é,w}æ╔┘:0xé
φïòΘDIO¼→▒╜ìk dZùèj
loßûé┬═fffé30¼½╣Θ1184C!δ+æ:0°‼é║┬000ë+╣Θ132▓j
Φ♂é☻óàÑ▒ü12 ≥δ½ü4
l+æΘ░╤0░é40Γleεº&Æ┬
eï╔σü0x4░┬05YSH°E (601) esp_core_dump_flash: Noÿ¼Kòüdump partition found!
E (602) esp_core_dump_flash: No core dump partition found!

[MSG:INFO: FluidNC v3.7.0]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:MKS_DLC32_21_XYZ.yaml]
[MSG:DBG: Running after-parse tasks]
[MSG:DBG: Checking configuration]
[MSG:INFO: Machine Johns CNC]
[MSG:INFO: Board MKS-DLC32 V2.1]
[MSG:INFO: I2SO BCK:gpio.16 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.14 MOSI:gpio.13 MISO:gpio.12]
[MSG:INFO: SD Card cs_pin:gpio.15 detect:NO_PIN freq:1000000]
[MSG:INFO: Stepping:I2S_static Pulse:5us Dsbl Delay:0us Dir Delay:5us Idle Delay:250ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable I2SO.0]
[MSG:INFO: Axis X (0.000,750.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.1 Dir:I2SO.2 Disable:NO_PIN]
[MSG:INFO:  X Neg Limit gpio.36:low]
[MSG:DBG:  X Neg Limit 0]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Y (0.000,750.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.5 Dir:I2SO.6 Disable:NO_PIN]
[MSG:INFO:  Y Neg Limit gpio.35:low]
[MSG:DBG:  Y Neg Limit 1]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Z (-150.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.3 Dir:I2SO.4 Disable:NO_PIN]
[MSG:INFO:  Z Neg Limit gpio.34:low]
[MSG:DBG:  Z Neg Limit 1]
[MSG:INFO:   Motor1]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: UART1 Tx:gpio.25 Rx:gpio.33:pd RTS:gpio.26 Baud:9600]
[MSG:INFO: Huanyang Spindle  Tx:gpio.25 Rx:gpio.33:pd RTS:gpio.26 Baud:9600]
[MSG:INFO: Using spindle Huanyang]
[MSG:INFO: Probe Pin: gpio.22]
[MSG:INFO: Connecting to STA SSID:BlueDragon]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.0.231]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
[... 94 lines cut ...]
[GC:G3 G54 G17 G21 G90 G94 M3 M9 T3 F960 S10000]
[GC:G1 G54 G17 G21 G90 G94 M3 M9 T3 F960 S10000]
[GC:G2 G54 G19 G21 G90 G94 M3 M9 T3 F960 S10000]
[GC:G1 G54 G19 G21 G90 G94 M3 M9 T3 F960 S10000]
[GC:G1 G54 G17 G21 G90 G94 M3 M9 T3 F960 S10000]
[GC:G1 G54 G17 G21 G91 G94 M3 M9 T3 F960 S10000]
<Run|MPos:58.333,8.500,-31.000|FS:960,10000|Pn:YZ|SD:92.04,/sd/SquareTest_1_T3.nc>
<Run|MPos:42.331,8.500,-31.000|FS:960,10000|Pn:YZ|SD:92.04,/sd/SquareTest_1_T3.nc>
[GC:G1 G54 G17 G21 G90 G94 M3 M9 T3 F960 S10000]
[GC:G1 G54 G17 G21 G91 G94 M3 M9 T3 F960 S10000]
[GC:G1 G54 G17 G21 G90 G94 M3 M9 T3 F960 S10000]
<Run|MPos:34.700,7.600,-23.288|FS:960,10000|Pn:YZ|SD:98.83,/sd/SquareTest_1_T3.nc>
<Run|MPos:31.369,6.870,0.000|FS:2969,10000|Pn:YZ|SD:98.83,/sd/SquareTest_1_T3.nc>
[MSG:DBG: VFD setState:0 SpindleSpeed:10000]
[MSG:DBG: RPM:10000 mapped to device units:4]
[MSG:INFO: VFD Queue Full]
[MSG:INFO: VFD Queue Full]
<Idle|MPos:0.000,0.000,0.000|FS:0,10000|Pn:YZ|SD:98.83,/sd/SquareTest_1_T3.nc>
[GC:G1 G54 G17 G21 G90 G94 M5 M9 T3 F960 S10000]
[MSG:DBG: VFD setState:0 SpindleSpeed:0]
[MSG:DBG: RPM:0 mapped to device units:4]
[MSG:INFO: VFD Queue Full]
[MSG:INFO: Program End]
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:YZ|WCO:0.000,0.000,-30.000|SD:100.00,/sd/SquareTest_1_T3.nc>
[MSG: /sd/SquareTest_1_T3.nc file job succeeded]

JohnCMBC, 2023-06-05:
Hi Mitch thanks for your prompt reply. I believe I have carried out the tasks required and have put then on GitHub. This was my first post and I inadvertently closed and opened the thread a couple of times. I hope this hasn't screwed things up.
For the responses I created a simple "square" cutout in Fusion and generated the code using the grbl post proc.
Cheers
John >

kevin0730, 2023-06-28:
I have had the same issue recently.  I will send the log messages when i get a chance to get them.

kevin0730, 2023-07-21:
I have just been reading through the FluidNC Wiki about the RS485 VFD and see the warning to make sure that the VFD is fully powered up before powering the ESP32.  In my case I had the VFD off when I started the GCode file to run my part.  I will do some testing to see if that is what caused the malfunction.  However, I would have expected and alarm because there was no RS485 communication.