bdring/FluidNC #913: Problem:

Bill83046, 2023-06-06:
### Controller Board

6-pack

### Machine Description

Ultimate Bee 4. 3 nm stepper motors with feedback using Normaly open 5vc 4mm detection proximity sensors
VFD spindle

### Configuration file

```yaml
board: 6 Pack
name: 6 Pack External XYYY Huanyang
meta: 2023-03-21 B. Dring for William Roberts

stepping:
  engine: I2S_STREAM
  idle_ms: 250
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 100.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.33
      limit_pos_pin: gpio.32
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.2
        direction_pin: I2SO.1
        disable_pin: I2SO.0

  y:
    steps_per_mm: 100.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 3
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.35
      limit_pos_pin: gpio.34
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.5
        direction_pin: I2SO.4
        disable_pin: I2SO.7
    
    motor1:
      limit_neg_pin: gpio.2
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.10
        direction_pin: I2SO.9
        disable_pin: I2SO.8

  z:
    steps_per_mm: 100.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.25
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.13
        direction_pin: I2SO.12
        disable_pin: I2SO.15
        
probe:
  pin: NO_PIN
  check_mode_start: true
  
        
i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5

start:
  must_home: false

Huanyang:
  uart:
    txd_pin: gpio.14
    rxd_pin: gpio.15
    rts_pin: gpio.13
    baud: 9600
    mode: 8N1
  modbus_id: 1
  tool_num: 10
  speed_map: 0=0% 0=25% 6000=25% 24000=100%
```

### Startup Messages

```text
[MSG:INFO: FluidNC v3.6.8]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:WARN: Cannot open configuration file:test.yaml]
[MSG:INFO: Using default configuration]
[MSG:INFO: Axes: using defaults]
[MSG:INFO: Machine Default (Test Drive)]
[MSG:INFO: Board None]
[MSG:INFO: Stepping:RMT Pulse:4us Dsbl Delay:0us Dir Delay:0us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Axis Y (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Axis Z (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: STA SSID is not set]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.6 [FluidNC v3.6.8 (wifi) '$' for help]
[MSG:INFO: '$H'|'$X' to unlock]
E (22807) sdmmc_io: sdmmc_io_reset: unexpected return: 0x102
[MSG:ERR: sdmmc_card_init failed code 0x102]
```

### User Interface Software

WEB Eui

### What happened?

O stepper motors wired , no VFD wired
wire 2 proximity sensors for limit switches to check functionality (5vc nO 4mm detection)
Both changed state using WEB Eui on the console when activated no problems that I could see.

Out of the blue no sd card. read where the firmware was updated to take care of this with the change in Freq.

MitchBradley, 2023-06-06:
"Out of the blue no sd card. read where the firmware was updated to take care of this with the change in Freq."

Could you please explain this more carefully.  That sentence is hard to understand.

bdring, 2023-06-06:
There is no SD card defined. The config defaulted.

Bill83046, 2023-06-07:
Mitch
Everything was working properly I got an error sd card failed to mount( sorry I did not get a screenshot of that ) tried rebooting a couple of times same problem. Searching I found where you addressed the problem with SD cards by adding this (frequency_hz: 9000000) Also found a firmware that updated this. So I do not understand how this happened my fault could very well be as I am new to this software. The SD card had to have been defined at some point as It was surely working. I know that I did not delete sd card configuration lines knowingly.  
Is this the SD card define lines?
sd card:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5
  frequency_hz: 9000000
Sorry to be a pain
Bill Roberts

bdring, 2023-06-07:
If you look at the startup messages, it has a warning.

```
[MSG:WARN: Cannot open configuration file:test.yaml]
```

This means it cannot find your config file.

See this wiki section.

http://wiki.fluidnc.com/en/support/troubleshooting_config_files#cannot-open-configuration-file

Bill83046, 2023-06-07:
Thank you Bart and Mitch for your help. It was me,  not understanding the
file system and where files belong. After reading through some documents on
FluidNC I was able to correct my errors and put files where they needed to
be. As I said I am new the Fluid NC and it is a learning curve for me.
Going forward I am sure that I will make a few more mistakes, hopefully
fixable. The system is now up and running with the updated firmware.
Bill Roberts

closed 2023-06-07 by MitchBradley