bdring/FluidNC #914: Problem:

basementmedia2, 2023-06-08:
### Controller Board

VEVOR 2500 mW CNC 3018 Pro-M  Machine, don't know Controller Board-Modell

[image]

### Machine Description

VEVOR 2500 mW CNC 3018 Pro-M 

### Configuration file

```yaml
-
```

### Startup Messages

```text
-
```

### User Interface Software

-

### What happened?

I am not able to flash the board with ./install-wifi.sh

Getting this error:

./install-wifi.sh
esptool.py --chip esp32 --baud 230400 dump_mem 0x3ff5a018 4 SecurityFuses.bin
esptool.py v3.1
Found 2 serial ports
Serial port /dev/ttyUSB0
Connecting........_____....._____....._____....._____....._____....._____....._____
/dev/ttyUSB0 failed to connect: Failed to connect to ESP32: Timed out waiting for packet header
Serial port /dev/ttyS0
Connecting........_____....._____....._____....._____....._____....._____....._____
/dev/ttyS0 failed to connect: Failed to connect to ESP32: Timed out waiting for packet header

A fatal error occurred: Could not connect to an Espressif device on any of the 2 available serial ports.
esptool failed

But /dev/ttyUSB0 is found by Candle2 and also by Arduino
In Arduino  can execute $$ command and get the GRBL-Response.

Is there something more i can try to get the firmware flashed? Maybe another baud rate?
Or do i just need another controller board?

Skorpi08, 2023-06-08:
Dein ernst? Laut deinem Profil bist du programmierer aber kriegst es nicht hin zu lesen dass FluidNC nur auf ESP32 Controllern läuft was dein Board nicht hat?!

basementmedia2, 2023-06-08:
Ich war der festen Überzeugung dass das ein ESP32 ist. Da ist mir wohl eine andere Charakter-Eigenschaft in die Quere gekommen :-) 

Sorry für den vorschnellen Post
Dann besorg ich mir ein anderes Board und der Käs is gefressen

Danke dir

closed 2023-06-08 by bdring